verilator/verilator #4787: Allow setting a path prefix for files accessed by $readmem and $writemem

anders-code, 2023-12-30:
If Verilog calls $readmemh("my_relative_file.mem"), it is opened relative to the current working directory of the sim.

Sometimes, especially since verilated code may be used as a library, this may not always be desired.

For example, an app or testbench may wish to locate .mem files using an environment variable or relative to the binary. Even if the filename in verilog is absolute, the user may want to redirect it to a chroot() or something of that kind.

I've added path prefix strings and accessors to the Context structure for readmem and writemem. These strings default to empty and are simply prepended to the filename given in Verilog. This lets the user manipulate the location that .mem files are found in a flexible manner.

wsnyder, 2023-12-31:
1. What do other simulators provide to enable a similar feature.
2. What prevents putting the appropriate resolution code in Verilog?
3. Why $readmem/writemem only - there are a lot of other file accessing routines e.g. $fopen etc.

anders-code, 2024-01-01:
Wow those are some great questions!

1. Sadly most simulators are often used embedded in front-end IDEs that run the sim in some kind baroque and deeply nested directory structure. So the support forums are littered with people asking why $readmem didn't find their file. Most of the time the solution given is to use a relative path, perhaps placing the file in a parent or sibling directory.
    - This works OK, especially if the directory structure is fixed.
    - It's subtle in that the file is found relative to the current working directory the sim is run from. But in these cases, the IDE often runs the sim relative to a known place in the directory structure, so it appears to work as expected.
    - A better answer which I've seen proposed is to use a parameter or `define to pass in the filename or path.
    - This line of thinking is where I started from. But Verilator is more like a synthesis engine than a a simulator in this way. So parameters are resolved at "compile" time, not runtime. But also, unlike a synthesis engine, the actual $readmem happens at runtime on the target. So I concluded that Verilator is somewhat unique in this regard.
    - Verilator is amenable to be used or included as a library in a larger C++ program in a very flexible way whereas most other simulators' executable is a single-use sim driver program.
    - I also checked the standard 1800-2017 and it is just under-specified. It says nothing at all about what the "file" argument means or how it should be interpreted. Forward slashes? Back? Drive letters? POSIX? It's good to not be overspecified, but means $readmem is not going to be portable.
    - to conclude: I know of no other simulator that has this capability, it's neither required or excluded by the standard, but there are reasons that I believe Verilator is unique.

2. To be honest, I though about the parameter/'define and as mentioned above saw that wouldn't work. In my case, the actual prefix I'm using is `get_exec_dir()+"../data"` where get_exec_dir() gets the directory the binary is run from (of course this is system specific and coded in C/C++). Neither of these are possible to do in Verilog directly (or at least in a pleasant way).
  - Another alternative would be `getenv("DATA_DIR")` or some logic with fallbacks
  - I just today learned about $value$plusargs, that could actually work
  - DPI? I'm not sure whether the other sims I use support that enough? Maybe split it to a Ventilator-specific .v file?
  - philosophically I looked at it as Verilog world, high-level: "give me the init for microcode ROM named foo.mem" and simulator runtime's job to treat that as a filename (maybe) and go find it.
  - Possibly, as I'm learning Verilator, I might should just read the file in C/C++ and then smash it into the memory using public_flat_rw or DPI instead of using $readmem at all. But then Vivado?
  - In fact, my original implementation was to do the following once at init. But it felt racy and hacky. So hacky that I spent a couple hours learning the innards of Verilator (which is time well spent):
    ``` C
     orig_dir = getcwd();
     chdir(data_dir);
     xyz->eval();
     chdir(orig_dir)
     ```

3. I considered $fopen and thought that "if you're doing this in Verilog, you are directly expecting POSIX-like behavior and specifically desire low-level control". $dumpfile? maybe? I waffled on $writemem as well and whether to have two prefixes or just one. In the end the fact that they shared a AST node made me lean toward "just readmem/writemem".

I should explain that the application I'm working on is not a classic simulation driver program. In such a case, I could just run the tool from a known path, etc. What I'm doing is a CPU in verilog and integrating it with an existing system simulator that is written in C and has opinions about how it initializes and runs, etc. Verilator is positively perfect for this application. I also do standard verilog test benches using Vivado on the same code-base.

I'm not hard over on any of this, btw. I can definely update the patch, doc, etc. whatever you suggest. And I appreciate this project very much.

wsnyder, 2024-01-02:
Since this is non-standard and also seems likely to cause portability and support issues, I'd suggest instead doing it either in Verilog, or with a "chdir" in beginning of your C++ main().

>Another alternative would be getenv("DATA_DIR") or some logic with fallbacks

Yes, you can probably DPI import "string getenv(string var)" directly into any SystemC simulator.

>I just today learned about $value$plusargs, that could actually work

Also reasonable.

>DPI? I'm not sure whether the other sims I use support that enough? Maybe split it to a Ventilator-specific .v file?

Any recent simulator should.

Closing as likely no change here but feel free to followup.

closed 2024-01-02 by wsnyder

anders-code, 2024-01-02:
I actually concur.  While I don't necessarily completely agree that there's any reason it would "cause portability" or "support issues" and I also don't like the "chdir()" approach because it's a very invasive thing to change an entire applications pwd, there is a couple advantages.

One major advantage (to me) of other approaches discussed is that I won't be dependent on a special recent development version of Verilator and that it can be done in a (more) portable and standard way. I can't see any drawback

Here's what I ended up doing which worked out nicely:

``` verilog
// parameter MEMORY_INIT_FILE is a string with "filename.mem"
initial begin
    string prefix = ""; // no +READMEM_PREFIX=..., prefix is empty and no change
    $value$plusargs("READMEM_PREFIX=%s", prefix); 
    $readmemh({prefix, MEMORY_INIT_FILE}, mem);
end
```

Then in C/C++ you construct a string like:
``` c
snprintf(/*... */, "+READMEM_PREFIX=%s/../data/", exec_dir);
Verilated::commandArgs(1, argv); // or append to real argv, or ..., etc.
```