verilator/verilator #4789: Trouble compiling Verilator

jordankrim, 2023-12-31:
I did this from the head of the verilator tree in Linux:
./configure
make
At the end of the compile was this error:
help2man: can't get `--help' info from bin/verilator_gantt
make: *** [Makefile:195: verilator_gantt.1] Error 13

wsnyder, 2023-12-31:
From the shell run "bin/verilator_gantt --help" and see what error is happening and debug why.

jordankrim, 2023-12-31:
(/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env) [jkrim@doritos bin]$ ./verilator_gantt --help
usage: verilator_gantt [-h] [--debug] [--no-vcd] [--vcd VCD] filename

Create Gantt chart of multi-threaded execution.

Verilator_gantt creates a visual representation to help analyze Verilator
#xmultithreaded simulation performance, by showing when each macro-task
#xstarts and ends, and showing when each thread is busy or idle.

For documentation see
https://verilator.org/guide/latest/exe_verilator_gantt.html

positional arguments:
  filename    input profile_exec.dat filename to process

optional arguments:
  -h, --help  show this help message and exit
  --debug     enable debug
  --no-vcd    disable creating vcd
  --vcd VCD   filename for vcd outpue

Copyright 2018-2023 by Wilson Snyder. This program is free software; you
can redistribute it and/or modify it under the terms of either the GNU
Lesser General Public License Version 3 or the Perl Artistic License
Version 2.0.

SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

I don't see an error.  The --help just prints above.

wsnyder, 2023-12-31:
Try the exact command that make executes

          help2man --no-info --no-discard-stderr --version-string=- bin/verilator_gantt -o verilator_gantt.1

jordankrim, 2024-01-01:
(/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env) [jkrim@doritos verilator-master]$ help2man --no-info --no-discard-stderr --version-string=- bin/verilator_profcfunc -o verilator_profcfunc.1
help2man: can't get `--help' info from bin/verilator_profcfunc

But, when I did a chmod +x  bin/verilator_profcfunc then I got this: help2man --no-info --no-discard-stderr --version-string=- bin/verilator_profcfunc -o verilator_profcfunc.1
(/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env) [jkrim@doritos verilator-master]$ 

So I tried to do a make (again) and got this:
Build complete!

Now type 'make test' to test.

So I tried to do a make test and got this:
test_regress/t/t_a1_first_cc.pl
Died at test_regress/t/t_a1_first_cc.pl line 2.
make: *** [Makefile:170: smoke-test] Error 13

wsnyder, 2024-01-01:
That fille should have been executable when you got it from git.  You did something strange to break the permissions. Remove your Verilator directory  and start over from scratch, following the instructions please.

jordankrim, 2024-01-02:
OK.  I deleted everything and downloaded the zip file (verilator-master.zip) and unzipped it to a directory call verilator-master.
Ran autoconf from that dir and ran ./configure and then make and then make test.  Buts seems like the same results for the make test as before, although this time I did not change any permissions on any files.
I also added the line to my ~/.gdbinit:

add-auto-load-safe-path /home/jkrim/verilator-master/test_regress/.gdbinit

The make test still fails with this output:

------------------------------------------------------------
making verilator in src
make -C src 
make[1]: Entering directory '/home/jkrim/verilator-master/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/jkrim/verilator-master"
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_dbg'
make[2]: Nothing to be done for 'serial_vlcov'.
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/jkrim/verilator-master"
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_dbg'
make -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_opt'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_opt'
make -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
make[2]: Entering directory '/home/jkrim/verilator-master/src/obj_opt'
      Compile flags:  g++ -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/jkrim/verilator-master"
make[2]: Leaving directory '/home/jkrim/verilator-master/src/obj_opt'
make[1]: Leaving directory '/home/jkrim/verilator-master/src'
test_regress/t/t_a1_first_cc.pl
======================================================================
dist/t_a1_first_cc: ==================================================
-Skip: dist/t_a1_first_cc: scenario 'dist' not enabled for test
dist/t_a1_first_cc: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_a1_first_cc: ==================================================
	perl ../bin/verilator --debug --debugi 0 --gdbbt --no-dump-tree -V
No stack.
BFD: /home/jkrim/verilator-master/bin/verilator_bin_dbg: unable to initialize decompress status for section .debug_aranges
BFD: /home/jkrim/verilator-master/bin/verilator_bin_dbg: unable to initialize decompress status for section .debug_aranges
"/home/jkrim/verilator-master/bin/verilator_bin_dbg": not in executable format: file format not recognized
No executable file specified.
Use the "file" or "exec-file" command.
No stack.
%Error: export VERILATOR_ROOT=/home/jkrim/verilator-master
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize gdb /home/jkrim/verilator-master/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --debugi 0 --gdbbt --no-dump-tree -V" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize gdb /home/jkrim/verilator-master/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --debug --debugi 0 --gdbbt --no-dump-tree -V" -ex 'set width 0' -ex 'bt' -ex 'quit'
Use of uninitialized value $param{"logfile"} in concatenation (.) or string at ./driver.pl line 1698.
%Warning: vlt/t_a1_first_cc: Exec of perl failed: 

vlt/t_a1_first_cc: %Error: Exec of perl failed: 
vlt/t_a1_first_cc: FAILED: Exec of perl failed: 
==SUMMARY: Passed 0  Failed 1  Time 0:00
==SUMMARY: Passed 0  Failed 1  Time 0:00

======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of perl failed: 
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:00
make: *** [Makefile:170: smoke-test] Error 10

wsnyder, 2024-01-02:
What OS are you using?  There's something pretty strange in your setup.

The install instructions do not say anything about a .zip, so I worry you might not be following the proper instructions, they are here:  https://verilator.org/guide/latest/install.html

jordankrim, 2024-01-02:
Red Hat Enterprise Linux release 8.9 (Ootpa)
I thought one could just use the .zip file vs. having to use the clone of the one on Github.

wsnyder, 2024-01-02:
I don't know what's causing the BFD issues, seems some toolchain problem that I don't know what Verilator could do to cause.  We don't do any development way back on gcc 8 which you are using, so you'll need to debug yourself.

Alternatively, I would note that Redhat 8 is end of life in 4 months, so your IT department hopefully have some more modern releases you can try instead.

jordankrim, 2024-01-03:
I have 2 Verilator repositories:

1. ~/verilator (Verilator 5.019 devel rev v5.018-44-gb8417abee)
2. ~/verilator-master (Verilator 5.021 devel rev v5.020-3-g3adb31f53)

I did a make clean on both repos.  I then compiled each: autoconf, ./configure, gmake, and make test
For #1 it works perfectly, passing all the tests.
For #2 it fails as I have shown before.

So, I find it bewildering that 5.019 works and 5.021 does not.  Thoughts?

wsnyder, 2024-01-03:
You can use git bisect to determine the exact change.  I suspect it will be something that changed related to gcc/linker that is hitting a bug in your older gcc 8.

jordankrim, 2024-01-03:
I see a flag -gz being used in the compile of the failing one (verilator-master) vs. the good one (verilator).  Maybe I can change that (how?) and see if it works?

wsnyder, 2024-01-03:
If you are saying in 5.019 that there was a comment like

# AS PER VERILATOR FORUM, REMOVE COMPRESSED FILE AS CAUSING PROBLEMS ???

There never was any such comment, that was something you must have done locally.

Anyhow glad you found the fix.  This indicates a bug in gcc/ld as the documentation says -gz should be ignored if non functional.  As this distro will be EOL in 5 months and you are the only one reporting it (so far), I don't see it worth improving configure to workaround this issue.  Please just remove the -gz in the next few versions.

closed 2024-01-03 by wsnyder

Alpha-Girl, 2024-01-18:
> If you are saying in 5.019 that there was a comment like
>
> `# AS PER VERILATOR FORUM, REMOVE COMPRESSED FILE AS CAUSING PROBLEMS ???`
>
> There never was any such comment, that was something you must [...]

I face the same problem:
```
for p in verilator.1 verilator_coverage.1 verilator_gantt.1 verilator_profcfunc.1 ; do \
  /usr/bin/install -c -m 644 $p /home/hyx/verilator/share/man/man1/$p; \
done
/usr/bin/install: cannot stat 'verilator_gantt.1': No such file or directory
/usr/bin/install: cannot stat 'verilator_profcfunc.1': No such file or directory
make: *** [Makefile:248: installman] Error 1
```

wsnyder, 2024-01-18:
@Alpha-Girl you have a different error message, please file a new issue showing the complete log file starting at downloading the kit, and describe what OS distro you are using.