verilator/verilator #4790: $time() Returns Wrong Value

polmacanceart, 2023-12-31:
When $time() is called midway between a timeunit, it should round to the nearest time unit.

Example:

```
/* Working through the $time example from IEEE Std 1364-2005
 * Section 17.7.1
 * The example uses a 10ns timeunit with a 1ns timeprecision
 * For 16ns $time should return 2
 * For 32ns $time should return 3
 */

module div_test ();
   timeunit 10ns;
   timeprecision 1ns;

   longint should_be_2, should_be_3;
   
   initial
     begin : initial_blk1
	should_be_2 = 0;
	should_be_3 = 0;
	#(16ns);
	$display("$time=%d, $realtime=%g", $time(), $realtime());
	should_be_2 = $time();
	#(16ns);
	$display("$time=%d, $realtime=%g", $time(), $realtime());
	should_be_3 = $time();
	#(16ns);
	$finish(1);
     end

   initial
     begin : initial_blk2
	#(100ns);
	$display("Error: We should not get here");
	$finish(1);
     end
	
   final
     begin : last_blk
	if (should_be_2 != 2) $display("Error: should_be_2 = %0d", 
				       should_be_2);
	if (should_be_3 != 3) $display("Error: should_be_3 = %0d",
				       should_be_3);
	$display("Info: should_be_2 = %0d", should_be_2);
	$display("Info: should_be_3 = %0d", should_be_3);
	
    end
   
endmodule

```
Driven from SystemC as follows:

```
#include "Vdiv_test.h"
#include <systemc.h>

int sc_main(int argc, char ** argv) {
  sc_set_time_resolution(1.0, SC_NS);
  Verilated::commandArgs(argc, argv);
  Vdiv_test* top=new Vdiv_test{"div_test"};
  
  while (!Verilated::gotFinish()) {
    sc_start(1, SC_NS);
  }
  top->final();
  delete top;
  return 0;
}

```

Compiled as follows:

```
verilator --sc --exe --timing sc_main.cpp div_test.sv -CFLAGS -std=c++20
make -j -C obj_dir/ -f Vdiv_test.mk Vdiv_test
./obj_dir/Vdiv_test

```

Gives:

```
        SystemC 2.3.3-Accellera --- Jan 20 2023 20:04:28
        Copyright (c) 1996-2018 by all Contributors,
        ALL RIGHTS RESERVED
$time=                   1, $realtime=1.6
$time=                   3, $realtime=3.2
- div_test.sv:18: Verilog $finish
Error: should_be_2 = 1
Info: should_be_2 = 1
Info: should_be_3 = 3

```

Should give:

```
        SystemC 2.3.3-Accellera --- Jan 20 2023 20:04:28
        Copyright (c) 1996-2018 by all Contributors,
        ALL RIGHTS RESERVED
$time=                   2, $realtime=1.6
$time=                   3, $realtime=3.2
- div_test.sv:18: Verilog $finish
Info: should_be_2 = 2
Info: should_be_3 = 3

```

The problem is in verilated_funcs.h? In the define for VL_TIME_UNITED_Q?

`#define VL_TIME_UNITED_Q(scale) (VL_TIME_Q() / static_cast<QData>(scale))
`

The following would fix?

```
#define VL_TIME_ROUND(t,p) ((t % p) >= (p/2)) ? ((t/p) + 1) : (t/p)
#define VL_TIME_UNITED_Q(scale) VL_TIME_ROUND(VL_TIME_Q(), static_cast<QData>(scale))

```

wsnyder, 2023-12-31:
Makes sense.  Can you turn this into a pull request including a test? Using SystemC versus pure verilog shouldn't matter related to this issue, so you should be able to make this into a normal test_regress test that uses just Verilog.   If any existing tests break because of this please make sure to test on existing simulators as the existing tests were thought to be pretty stable.  Thanks.

>#define VL_TIME_ROUND(t,p) ((t % p) >= (p/2)) ? ((t/p) + 1) : (t/p)

Faster:

`#define VL_TIME_ROUND(t,p) (((t) + ((p)/2)) / (p))
`

polmacanceart, 2024-01-01:
Done. I ran `make test_regress`, all other tests passed.

closed 2024-01-02 by wsnyder