verilator/verilator #4799: Verilator seg fault on openhwgroup/cvw

davidharrishmc, 2024-01-03:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
harris@chips:~/repos/cvwt$ git clone https://github.com/openhwgroup/cvw
Cloning into 'cvw'...
...
edit cvw/testbench/testbench.sv and change line 38 to 
  parameter TEST="arch64i";
harris@chips:~/repos/cvwt$ cd cvw/sim
harris@chips:~/repos/cvwt/cvw/sim$ ./verilate 
rv64gc simulating...
make: Entering directory '/home/harris/repos/cvwt/cvw/sim/obj_dir'
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-overloaded-virtual -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
...
Archive ar -rcs Vtestbench__ALL.a Vtestbench.o Vtestbench___024root__DepSet_hfc24d085__0.o Vtestbench___024root__DepSet_hfc24d085__1.o Vtestbench___024root__DepSet_hfc24d085__2.o Vtestbench___024root__DepSet_hfc24d085__3.o Vtestbench___024root__DepSet_hfc24d085__4.o Vtestbench___024root__DepSet_hed41eec4__0.o Vtestbench___024root__DepSet_hed41eec4__1.o Vtestbench___024root__DepSet_hed41eec4__2.o Vtestbench___024root__DepSet_hed41eec4__3.o Vtestbench___024root__DepSet_hed41eec4__4.o Vtestbench___024root__DepSet_hed41eec4__5.o Vtestbench___024root__DepSet_hed41eec4__6.o Vtestbench_cacheway__pi6__DepSet_hfd209fcc__0.o Vtestbench_cacheway__pi6__DepSet_hec3e220d__0.o Vtestbench_tlbram__Pz1_T20__DepSet_h3df7cb71__0.o Vtestbench_tlbram__Pz1_T20__DepSet_hab70f5b0__0.o Vtestbench_tlbcamline__Pz1_K34_S9__DepSet_ha51865b9__0.o Vtestbench_tlbcamline__Pz1_K34_S9__DepSet_ha51865b9__1.o Vtestbench_tlbcamline__Pz1_K34_S9__DepSet_h14165bf8__0.o Vtestbench__main.o Vtestbench__ConstPool_0.o Vtestbench___024root__Slow.o Vtestbench___024root__DepSet_hfc24d085__0__Slow.o Vtestbench___024root__DepSet_hfc24d085__1__Slow.o Vtestbench___024root__DepSet_hed41eec4__0__Slow.o Vtestbench___024root__DepSet_hed41eec4__1__Slow.o Vtestbench___024root__DepSet_hed41eec4__2__Slow.o Vtestbench___024root__DepSet_hed41eec4__3__Slow.o Vtestbench___024root__DepSet_hed41eec4__4__Slow.o Vtestbench___024unit__Slow.o Vtestbench___024unit__DepSet_hf87c9ffd__0__Slow.o Vtestbench_cacheway__pi6__Slow.o Vtestbench_cacheway__pi6__DepSet_hec3e220d__0__Slow.o Vtestbench_tlbram__Pz1_T20__Slow.o Vtestbench_tlbram__Pz1_T20__DepSet_h3df7cb71__0__Slow.o Vtestbench_tlbram__Pz1_T20__DepSet_hab70f5b0__0__Slow.o Vtestbench_tlbcamline__Pz1_K34_S9__Slow.o Vtestbench_tlbcamline__Pz1_K34_S9__DepSet_h14165bf8__0__Slow.o Vtestbench__Syms.o
g++ -fuse-ld=mold    verilated.o verilated_timing.o verilated_threads.o Vtestbench__ALL.a    -pthread -lpthread -latomic   -o Vtestbench
rm Vtestbench__ALL.verilator_deplist.tmp
make: Leaving directory '/home/harris/repos/cvwt/cvw/sim/obj_dir'
TEST is arch64i
./verilate: line 9: 1607893 Segmentation fault      (core dumped) ./obj_dir/Vtestbench
Verilation complete
```

What 'verilator' command line do we use to run your example?
see above

What 'verilator --version' are you using?  Did you try it with the git master version?
harris@chips:~/repos/cvwt/cvw/sim$ verilator --version
Verilator 5.019 devel rev v5.018-101-g54ad04b6e

What OS and distribution are you using?
Ubuntu 22.04LTS

May we assist you in trying to fix this in Verilator yourself?
Yes. We are trying to bring up our OpenHWGroup CORE-V Wally (cvw) core in Verilator.  It presently simulates in Questa and lints in Verilator.  It now compiles in Verilator, but seg faults.  Adding $display statements show that the crash occurs after the second rising edge of the clock (10 ticks into simulation) but before reading a memory file into the RAM.  I'm over my head on how to debug Verilator, but would be delighted to collaborate getting our core to work with Verilator.  I expect this may involve a combination of bug fixes in Verilator and perhaps code changes in the cvw testbench.

wsnyder, 2024-01-03:
To debug runtime errors, try this:
1. Verilate with "-CFLAGS -DVL_DEBUG -CFLAGS -D_GLIBCXX_DEBUG".  
2. Verilate with "-CFLAGS -ggdb -LDFLAGS -ggdb". 
3. Verilate with "-CFLAGS -fsanitize=address,undefined -LDFLAGS -fsanitize=address,undefined".
4. If you have your own makefiles be sure to likewise pass the appropriate compiler and load flags shown above.
5. Run the model under "gdb" and use "bt" to backtrace to the C++ code causing the issue.

davidharrishmc, 2024-01-06:
Thank you.  This is what I see.  The segfault is in a function that tries to read a file.

```
verilator -CFLAGS -DVL_DEBUG -CFLAGS -D_GLIBCXX_DEBUG -CFLAGS -ggdb -LDFLAGS -ggdb -CFLAGS -fsanitize=address,undefined -LDFLAGS -fsanitize=address,undefined --timescale "1ns/1ns" --timing --binary --top-module testbench "-I../config/shared" "-I../config/rv64gc" ../src/cvw.sv ../testbench/testbench.sv ../testbench/common/*.sv ../src/*/*.sv ../src/*/*/*.sv --relative-includes
...
harris@chips:~/cvw/sim$ gdb ./obj_dir/Vtestbench
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./obj_dir/Vtestbench...
(gdb) run
Starting program: /home/harris/cvw/sim/obj_dir/Vtestbench 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffef2e1b640 (LWP 228483)]
[New Thread 0x7ffef25fe640 (LWP 228484)]
...
TEST is arch64m

Thread 1 "Vtestbench" received signal SIGSEGV, Segmentation fault.
0x000055555ac2edc5 in Vtestbench___024root___nba_sequent__TOP__0 (vlSelf=vlSelf@entry=0x7ffef2e1d900) at Vtestbench___024root__DepSet_hfc24d085__0.cpp:3498

3498    VL_INLINE_OPT void Vtestbench___024root___nba_sequent__TOP__0(Vtestbench___024root* vlSelf) {
(gdb) 
(gdb) bt
#0  0x000055555ac2edc5 in Vtestbench___024root___nba_sequent__TOP__0 (vlSelf=vlSelf@entry=0x7ffef2e1d900) at Vtestbench___024root__DepSet_hfc24d085__0.cpp:3498
#1  0x000055555b13e499 in Vtestbench___024root___eval_nba (vlSelf=vlSelf@entry=0x7ffef2e1d900) at Vtestbench___024root__DepSet_hfc24d085__4.cpp:1820
#2  0x000055555bbb7cf3 in Vtestbench___024root___eval_phase__nba (vlSelf=vlSelf@entry=0x7ffef2e1d900) at Vtestbench___024root__DepSet_hed41eec4__6.cpp:4070
#3  0x000055555bbb7ff7 in Vtestbench___024root___eval (vlSelf=vlSelf@entry=0x7ffef2e1d900) at Vtestbench___024root__DepSet_hed41eec4__6.cpp:4118
#4  0x000055555abfbc67 in Vtestbench::eval_step (this=0x617000000080) at Vtestbench.cpp:126
#5  0x000055555c1c5a00 in Vtestbench::eval (this=0x617000000080) at /home/harris/cvw/sim/obj_dir/Vtestbench.h:125
#6  main (argc=<optimized out>, argv=<optimized out>) at Vtestbench__main.cpp:21
```

davidharrishmc, 2024-01-06:
You won't have that test vector file in the repo according to my pull directions because the file requires building, which requires a RISC-V tool chain, which is a bunch of work to set up.  However, I would expect a file not found error rather than a segfault.

Also, the relative path to the test vector file isn't right because Vtestbench is in obj_dir.  I tried copying it up one level so the path would be correct, and now get a different error.

```
harris@chips:~/cvw/sim$ cp obj_dir/Vtestbench .
harris@chips:~/cvw/sim$ ./Vtestbench 
TEST is arch64m
AddressSanitizer:DEADLYSIGNAL
=================================================================
==230106==ERROR: AddressSanitizer: stack-overflow on address 0x7ffdf33be788 (pc 0x5586287dedc5 bp 0x7ffdf3bbc7b0 sp 0x7ffdf33be788 T0)
    #0 0x5586287dedc5 in Vtestbench___024root___nba_sequent__TOP__0(Vtestbench___024root*) /home/harris/cvw/sim/obj_dir/Vtestbench___024root__DepSet_hfc24d085__0.cpp:3498
    #1 0x558628cee498 in Vtestbench___024root___eval_nba(Vtestbench___024root*) /home/harris/cvw/sim/obj_dir/Vtestbench___024root__DepSet_hfc24d085__4.cpp:1820
    #2 0x558629767cf2 in Vtestbench___024root___eval_phase__nba(Vtestbench___024root*) /home/harris/cvw/sim/obj_dir/Vtestbench___024root__DepSet_hed41eec4__6.cpp:4070
    #3 0x558629767ff6 in Vtestbench___024root___eval(Vtestbench___024root*) /home/harris/cvw/sim/obj_dir/Vtestbench___024root__DepSet_hed41eec4__6.cpp:4118
    #4 0x5586287abc66 in Vtestbench::eval_step() /home/harris/cvw/sim/obj_dir/Vtestbench.cpp:126
    #5 0x558629d759ff in Vtestbench::eval() /home/harris/cvw/sim/obj_dir/Vtestbench.h:125
    #6 0x558629d759ff in main /home/harris/cvw/sim/obj_dir/Vtestbench__main.cpp:21
    #7 0x7fe50730dd8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #8 0x7fe50730de3f in __libc_start_main_impl ../csu/libc-start.c:392
    #9 0x5586286cf614 in _start (/home/harris/cvw/sim/Vtestbench+0x55cb614)

SUMMARY: AddressSanitizer: stack-overflow /home/harris/cvw/sim/obj_dir/Vtestbench___024root__DepSet_hfc24d085__0.cpp:3498 in Vtestbench___024root___nba_sequent__TOP__0(Vtestbench___024root*)
==230106==ABORTING
```

davidharrishmc, 2024-01-06:
I'm in over my head here - I haven't used gdb in twenty years and don't have a good sense of what I'm trying to do with it or how the generated Vtestbench code relates to my Verilog and the root bug in Verilator.  I think the issue should be reproducible without the test vector file because the segfault seems to occur shortly before reading any files.  I would love to support getting this solved, and could collaborate over screen sharing if that would facilitate the process.

wsnyder, 2024-01-06:
I got the failure. Playing a hunch based on the code location and it moved:

```
  TEST is arch64i

  Thread 1 "Vtestbench" received signal SIGSEGV, Segmentation fault.
  0x000055555ac772dd in Vtestbench___024root___nba_sequent__TOP__0 (vlSelf=vlSelf@entry=0x7ffef2e1b900) at Vtestbench___024root__DepSet_hfc24d085__0.cpp:3498

$ ulimit -a
  real-time non-blocking time  (microseconds, -R) unlimited
  ...
  stack size                  (kbytes, -s) 9788
  ...

$ ulimit -s 100000
$ ulimit -a
  real-time non-blocking time  (microseconds, -R) unlimited
  stack size                  (kbytes, -s) 100000

$ gdb ./obj_dir/Vtestbench
  ...
  TEST is arch64i
  Couldn't find begin_signature in ../tests/riscof/work/riscv-arch-test/rv64i_m/I/src/add-01.S/ref/ref.elf.objdump.lab
  Couldn't find sig_end_canary in ../tests/riscof/work/riscv-arch-test/rv64i_m/I/src/add-01.S/ref/ref.elf.objdump.lab
  %Warning: ../tests/riscof/work/riscv-arch-test/rv64i_m/I/src/add-01.S/ref/ref.elf.memfile:0: $readmem file not found

```

So, just increase the stack size and should be good to go.  I don't know why Linux still has it default so small.

closed 2024-01-06 by wsnyder

davidharrishmc, 2024-01-06:
Thank you.  Your effort is amazing!

wsnyder, 2024-01-06:
Just committed predicting stack needed, and warning if not enough.  It predicts for your model needing ~120k and the model needs 59k to not crash (but might need more later when you add a .mem file).

```
$ obj_dir/Vtestbench
%Warning: System has stack size 9788 kb which may be too small; suggest at least 'ulimit -c 234612'
```

poemonsense, 2024-04-26:
Hi @wsnyder, we have been facing the same issue since October 2023, as well.

At that time, Verilator does not provide the stack check feature as well as the predicted stack size. Thus, [here](https://github.com/OpenXiangShan/difftest/pull/196) is how we addressed this issue with a constant stack size. We set the stack size during simulation-time using `setrlimit`.

However, as our design size continuously increases, we find that a constant value does not fundamentally fix the issue. Therefore, I think Verilated models should provide a public API `getPredictedStackSize` to access the predicted stack size, and decide whether they are going to enlarge the stack size. We can set the stack size in the user code to, like, `roundUpPowerOf2(1.5 * predictedStackSize)`. This would be a basic method the model could provide.

Furthermore, if it does not bring unexpected issues, I would propose to further introduce a public `updateStackSize` method to update the stack size if the system stack is smaller than the expected one. The user should manually call this function to update the stack size. If enabling it by default does not bring side effects, it would be much better if the VTop just calls it by default.

What's your opinion on both proposals? If you think they're good, we can implement them with PRs.

wsnyder, 2024-04-26:
I don't believe it's possible in Linux to make a `updateStackSize` function as stack is created before program starts.  But if you find a way please open a new issue describing the proposal.

poemonsense, 2024-04-27:
> I don't believe it's possible in Linux to make a `updateStackSize` function as stack is created before program starts. But if you find a way please open a new issue describing the proposal.

Thanks for your notice. I see what you mean. It makes sense. I'll have a check on why the `setrlimit` works for our environment. Probably it requires a second-time run (after the segmentation fault) for it to really take effect. Will come back later after the confirmation.