verilator/verilator #4800: warning: anonymous union cannot be 'const' [-Wpedantic]

atomlin, 2024-01-03:
Verilator 5.018 2023-10-30 rev v5.018
In verilated_trace.h 
the lines:
```
    struct CallbackRecord {
        // Note: would make these fields const, but some old STL implementations
        // (the one in Ubuntu 14.04 with GCC 4.8.4 in particular) use the
        // assignment operator on inserting into collections, so they don't work
        // with const fields...
        const union {  // The callback
            initCb_t m_initCb;
            dumpCb_t m_dumpCb;
            dumpOffloadCb_t m_dumpOffloadCb;
            cleanupCb_t m_cleanupCb;
        };

```
when compiled with clang++ 14 generate a warning
warning: anonymous union cannot be 'const' [-Wpedantic]

wsnyder, 2024-01-04:
Perhaps we just name the union?  Might you be willing to try that and if it works submit a pull request?  Thanks

atomlin, 2024-01-04:
Let me get the latest release and try again.

closed 2024-01-06 by wsnyder

atomlin, 2024-01-09:
Its not fixed yet, I have not had chance yet to investigate