verilator/verilator #4801: Verilator Internal error related to loop unrolling

Kamoulaz, 2024-01-03:
Hello,

We had these errors "Unsupported: Delayed assignment to array inside for loops (non-delayed is ok - see docs)". We fixed them through loop unrolling: --unroll-count 10000000 --unroll-stmts 10000000.

Now we are running into an internal error  "Unknown node type reached emitter: BEGIN"  at this particular code block

```
task xOne;
  integer i;
  begin
    for (i = 0; i < 8192; i = i + 1) begin
`ifdef delay
      u_ram.mem[i] <= #(`delay) {15{1'bx}};
`else
      u_ram.mem[i] <= #0.005 {15{1'bx}};
`endif
    end
    if (f_print) $display ("\tINFO %m ... ... %t", $realtime);
  end
endtask
```

We would appreciate your help on this.

Regards,
Kamel

wsnyder, 2024-01-04:
Perhaps related to  #4449?

Can you avoid using delays in that code as a workaround?  Especially in such a large loop they will slow things down and likely aren't needed functionally.

Kamoulaz, 2024-01-05:
Hello @wsnyder,

Unfortunately removing the delays is not possible as the file is 9000+ lines and we want to avoid creating a different set of files.

Do you have any ideas for fixing this using unrolling ?

wsnyder, 2024-01-05:
@kboronski-ant will need to comment on when he will have time, or better could you look into debugging it?

If you don't want to edit the code itself, you could use the --pipe-filter option and a script to cleanup the code on-the-fly as its read by Verilator.

wsnyder, 2024-01-26:
I was unable to reproduce this, if you can please modify the below to show the issue. This uses a new pragra for unrolling so you'll need the latest master git version, or instead use the appropriate unrolling command line switches (and indicate what they are), thanks.

verilator --binary

```
// See issue #4801

module t(/*AUTOARG*/);

   bit clk;
   initial forever begin #5; clk = !clk; end

   logic [15:0] mem [1024];

   task clear;
      integer i;
      begin
         $display ("[%0t] clear start", $time);
         // verilator unroll_full
         for (i = 0; i < 1024; i = i + 1) begin
            mem[i] <= #1 i[15:0];
         end
         $display ("[%0t] clear done", $time);
      end
   endtask

   int cyc;

   always_ff @ (posedge clk) begin
      cyc <= cyc + 1;
      if (cyc == 1) begin
         if (mem[5] !== '0 && mem[5] !== {16{1'bx}}) $stop;
      end
      else if (cyc == 2) begin
         clear();
      end
      else if (cyc == 3) begin
         if (mem[5] != 5) $stop;
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule
```

wsnyder, 2024-02-14:
Please reopen if you provide a small example to reproduce this.

closed 2024-02-14 by wsnyder