verilator/verilator #4804: Internal Error: AstNode is not of expected type, but instead of has type 'STRUCTSEL'

futurehome, 2024-01-05:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
[error_report.zip](https://github.com/verilator/verilator/files/13837258/error_report.zip)

What 'verilator' command line do we use to run your example?
verilator --cc error_report.v

What 'verilator --version' are you using?  Did you try it with the git master version?
5.018
What OS and distribution are you using?
Centos 7.
May we assist you in trying to fix this in Verilator yourself?
Yes

wsnyder, 2024-01-05:
With --debug:

```
- V3Delayed.cpp:650:  delayedAll:
%Error: Internal Error: error_report.v:14:18: ../V3Ast.h:2359: AstNode is not of expected type, but instead has type 'STRUCTSEL'
-node: STRUCTSEL 0x5555570d3b00 <e7527> {d14as} @dt=0x555556f01800@(w32)u[31:0]  __PVT__data
   14 |         data_out.data[j] <= data_in.data[j];
      |                  ^~~~
```

https://github.com/verilator/verilator/blob/1a0c8f15738a23100bdee27705ac2ff22801620f/src/V3Delayed.cpp#L248

currently doesn't handle other than packed arrays (ArraySel)s.  Perhaps you could look into fixing this?  Basically what is now building the "dimvalp" a queue of selects needs to become more complicated to also handle StructSel's (unpacked structure element references).

The temp workaround is use a packed structure.

futurehome, 2024-01-08:
But if I changed <= to =, then it worked.

I would like to have a try for what you've said, but it may not be an easy task for me.

Thanks,
Alex

wsnyder, 2025-09-09:
This passes on current master, presuming fixed earlier.