verilator/verilator #4805: Verilator does not work with SystemC 3.0.0

maveric2, 2024-01-05:
verilated_vcd_sc.h:46:35: fatal error: no member named 'add_trace_file' in 'sc_core::sc_simcontext'
        sc_get_curr_simcontext()->add_trace_file(this);

wsnyder, 2024-01-05:
[System 3.0.0](https://github.com/accellera-official/systemc/releases/tag/3.0.0_pub_rev_20231129) is only in draft review, as you probably know.  Could you please consider making a pull request to Verilator (and/or SystemC itself as now is the time) to support it?

donlon, 2024-01-05:
It seems that SystemC 3.0.0 remove phase callback and trace file related APIs, which Verilator's tracing function depends on currently. Instead the new version uses the new stage callback API to provide callbacks. I can fix it later quickly.

PR #4807 (2024-01-06): Support SystemC 3.0.0 public review version (#4805)
Closes #4805

It supports tracing with SystemC's new stage callback.

Not sure why t_var_sc_bv fails with SystemC 3.0.0, will have a look later.

```
$ test_regress/obj_vlt/t_var_sc_bv/Vt_var_sc_bv
        SystemC 3.0.0_pub_rev_20231124-Accellera
        Copyright (c) 1996-2023 by all Contributors,
        ALL RIGHTS RESERVED
free(): invalid pointer
```
closed 2024-01-06 by wsnyder