verilator/verilator #4810: Segfault when combining `--trace` and `-fno-inline`

sampsyo, 2024-01-07:
Hello! As always, thank you for Verilator!!

On git `master` and 5.018 on macOS/AArch64, I'm running into a segfault when I try to use both `--trace` and `-fno-inline`. The actual Verilog doesn't seem to matter; this input file suffices:

```verilog
module  TOP;
endmodule
```

Both of these runs work fine:

```
$ verilator_bin min.sv --debug --gdbbt --cc --trace
$ verilator_bin min.sv --debug --gdbbt --cc -fno-inline
```

But combining both `--trace` and `-fno-inline` results in this error:

```
$ verilator_bin min.sv --debug --gdbbt --cc --trace -fno-inline
Starting Verilator 5.021 devel rev v5.020-18-g68e4eeb53
- Verilator.cpp:616:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
[...]
- V3Ast.cpp:1280:     Dumping obj_dir/Vmin_051_splitas.tree
- V3TraceDecl.cpp:692:traceDeclAll: 
zsh: segmentation fault  [path]/bin/verilator_bin min.sv
```

Here is some location information from lldb:

```
Process 64134 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0xf0)
    frame #0: 0x00000001003d3118 verilator_bin`TraceDeclVisitor::fixupPlaceholder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, AstNodeStmt*) + 120
verilator_bin`TraceDeclVisitor::fixupPlaceholder:
->  0x1003d3118 <+120>: ldr    x8, [x8, #0xf0]
    0x1003d311c <+124>: mov    w19, #0x2
    0x1003d3120 <+128>: cbz    x8, 0x1003d3130           ; <+144>
    0x1003d3124 <+132>: ldrh   w8, [x8, #0x40]
Target 0: (verilator_bin) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0xf0)
  * frame #0: 0x00000001003d3118 verilator_bin`TraceDeclVisitor::fixupPlaceholder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, AstNodeStmt*) + 120
    frame #1: 0x00000001003d0480 verilator_bin`TraceDeclVisitor::fixupPlaceholders() + 256
    frame #2: 0x00000001003cfd28 verilator_bin`TraceDeclVisitor::TraceDeclVisitor(AstNetlist*) + 236
    frame #3: 0x00000001003cf918 verilator_bin`V3TraceDecl::traceDeclAll(AstNetlist*) + 52
    frame #4: 0x00000001000ef23c verilator_bin`verilate(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 2768
    frame #5: 0x00000001000ee044 verilator_bin`main + 444
    frame #6: 0x00000001839ebf28 dyld`start + 2236
(lldb) 
```

> What 'verilator --version' are you using?  Did you try it with the git master version?
> What OS and distribution are you using?

I reproduced this on revision 68e4eeb53e61ad3cbc7b00bb1ca70004e64c042b, the latest as of yesterday, and on a released version (5.018). All of this is on macOS 13.4.1 on an ARM machine.

Notably, I also have access to another machine (x86_64 Linux) with Verilator 5.006, where this problem does not exist.

> May we assist you in trying to fix this in Verilator yourself?

Definitely! I admit I'm not sure exactly where to start, other than that this is clearly happening somewhere in the tracing machinery.

wsnyder, 2024-01-07:
Just to double check I ran all the trace regressions with -fno-inline on Ubuntu and got no verilator segfaults.

First, use "verilator" not "verilator_bin" as it looks like you aren't getting the debug source filenames/lines.

>address=0xf0

probably the function is getting passed a nullptr, you'll need to trace it back.

sampsyo, 2024-01-07:
Thank you for the tips! I rebuilt Verilator with debug symbols to help track down where the nullptr comes from. It arises here:
https://github.com/verilator/verilator/blob/eec41fd039b809ac67fe4b0a872d03f1b8490773/src/V3TraceDecl.cpp#L262

It looks like `scopep->aboveCellp()` is null here. For this call to `fixupPlaceholder`, the `path` is just `TOP` (i.e., the only module in my no-op program); `scopep` is also the `TOP` module.

I admit I am really out of my Verilator depth here, but it would make sense that the `TOP` module has no "parent" scope… just to confirm that's what's going on, here's the lldb dump of `*scopep`:

```
(lldb) p *scopep
(const AstScope) $6 = {
  AstNode = {
    m_nextp = nullptr
    m_backp = 0x0000600003304000
    m_op1p = nullptr
    m_op2p = 0x0000000100e04080
    m_op3p = nullptr
    m_op4p = nullptr
    m_iterpp = nullptr
    m_type = (m_e = atScope)
    m_flags = (didWidth = false, doingWidth = false, protect = true, unused = '\0')
    m_brokenState = '3'
    m_cloneCnt = 0
    m_dtypep = nullptr
    m_headtailp = 0x00006000039044e0
    m_fileline = 0x0000600000c0e010
    m_clonep = nullptr
    m_user1u = {
      m_u = (up = 0x0000000000000001, ui = 1)
    }
    m_user1Cnt = 72
    m_user2Cnt = 0
    m_user2u = {
      m_u = (up = 0x0000000000000000, ui = 0)
    }
    m_user3u = {
      m_u = (up = 0x0000000000000000, ui = 0)
    }
    m_user3Cnt = 0
    m_user4Cnt = 0
    m_user4u = {
      m_u = (up = 0x0000000000000000, ui = 0)
    }
  }
  m_name = "TOP"
  m_aboveScopep = nullptr
  m_aboveCellp = nullptr
  m_modp = 0x0000600003d00000
}
```

Abstractly, it seems to me there are a few possibilities:

1. The `TOP` scope is supposed to have a parent scope.
2. We are never supposed to call `fixupPlaceholder` with `path == "TOP"` at all.
3. `m_pathToScopep.find(path)` is producing a subpath that it shouldn't (e.g., maybe it should be empty).

Without `-fno-inline`, `fixupPlaceholder` isn't called at all, which lends credence to theory 2. In that condition, `m_cellInitPlaceholders` and `m_ifaceRefInitPlaceholders` both appear to be empty here:
https://github.com/verilator/verilator/blob/eec41fd039b809ac67fe4b0a872d03f1b8490773/src/V3TraceDecl.cpp#L289

Even with a somewhat larger Verilog program, it seems like `m_cellInitPlaceholders` is empty without `-fno-inline` and decidedly nonempty with `-fno-inline`. So perhaps the problem is with things being detected as a subscope when they shouldn't be, here:
https://github.com/verilator/verilator/blob/eec41fd039b809ac67fe4b0a872d03f1b8490773/src/V3TraceDecl.cpp#L391

Does it seem like I might be on the right track? With inlining disabled, maybe some parentless scopes (such as top-level modules) are being misclassified as subscopes of something else I'm not quite seeing.

sampsyo, 2024-01-07:
It seems also relevant that `TOP` is added to `m_entries` here, only in the `-fno-inline` condition:
https://github.com/verilator/verilator/blob/eec41fd039b809ac67fe4b0a872d03f1b8490773/src/V3TraceDecl.cpp#L342-L345

Specifically, with `-fno-inline`, we gather this cell:
```
CELL 0x600001d14000 {d1ai}  __PVT__TOP -> MODULE 0x600001e181e0 {d1ai}  TOP  L0 [1ps]
```

Without `-fno-inline`, there are (as expected) no statements inside `TOP`, so nothing gets added to `m_entries`.

wsnyder, 2024-01-07:
This seems like the right track.  I would suggest open a virtual (or real) window on some Ubuntu 22.04 system and you can compare the MacOS with the expected results at each step.

sampsyo, 2024-01-07:
Surprisingly, when I tried the latest `master` on my Ubuntu/x86_64 machine, I ran into the same problem! Then I tried the `v5.006` tag and everything worked.

I followed this down the `git bisect` rabbit hole and was eventually able to blame 95c4ade718a21d51c661cbe37b252924c0a822cd. Unfortunately, that's a pretty big diff, even just in `V3TraceDecl.cpp`. But it's also the place where `fixupPlaceholders` is introduced in the first place, which would make sense. Perhaps @gezalore, the author of this commit, might have an idea at a glance?

As a seriously uninformed guess, the following patch does Work For Me™, but seems like it could really not be the right thing:

```diff
diff --git a/src/V3TraceDecl.cpp b/src/V3TraceDecl.cpp
index 47986dbd6..69219fab5 100644
--- a/src/V3TraceDecl.cpp
+++ b/src/V3TraceDecl.cpp
@@ -259,10 +259,12 @@ class TraceDeclVisitor final : public VNVisitor {
             const std::string name = path.substr(pos == string::npos ? 0 : pos + 1);
 
             // Compute the type of the scope being fixed up
-            AstNodeModule* const modp = scopep->aboveCellp()->modp();
-            const VTracePrefixType scopeType = VN_IS(modp, Iface)
-                                                   ? VTracePrefixType::SCOPE_INTERFACE
-                                                   : VTracePrefixType::SCOPE_MODULE;
+            const AstCell* const cellp = scopep->aboveCellp();
+            const VTracePrefixType scopeType = cellp ? (
+                VN_IS((cellp->modp()), Iface)
+                                ? VTracePrefixType::SCOPE_INTERFACE
+                                : VTracePrefixType::SCOPE_MODULE
+                ) : VTracePrefixType::SCOPE_MODULE;
 
             // Push the scope prefix
             AstNodeStmt* const pushp = new AstTracePushPrefix{flp, name, scopeType};
```

Troublingly, I still haven't found a good reason why this is happening for me but not you. This is Ubuntu 22.04, compiling Verilator with gcc 10.5.0.

wsnyder, 2024-01-11:
It might just be the data at that address when dereferenced is different on the different OSes.  Can you change your patch to a pull request please?

wsnyder, 2024-01-14:
Thanks for the debug and pull.

closed 2024-01-14 by wsnyder