verilator/verilator #4811: Verilator $sscanf reads into array with a delay

davidharrishmc, 2024-01-07:
Thanks for taking the time to report this.

I am experiencing strange behavior when trying to read lines from a file into an array one at a time.  The file has three lines:
10
11
12

When I run the testbench in Questa, I read them as expected.

```
harris@chips:~/cvw/examples/verilog/testcase$ qrun testbench.sv
...
# Read vectors.tv
# sig32[0] = 00000010   line: 00000010
# 
# sig32[1] = 00000011   line: 00000011
# 
# sig32[2] = 00000012   line: 00000012
# 
# ** Note: $finish    : testbench.sv(42)
```

But when I run it in Verilator, the first line shows up in the next array entry.

```
harris@chips:~/cvw/examples/verilog/testcase$ verilator --timescale "1ns/1ns" --timing --binary --top-module testbench testbench.sv
...
harris@chips:~/cvw/examples/verilog/testcase$ ./obj_dir/Vtestbench
Read vectors.tv
sig32[0] = **00000000**   line: 00000010

sig32[1] = 00000010   line: 00000011

sig32[2] = 00000011   line: 00000012

- testbench.sv:42: Verilog $finish
```

If I change the code to read into a temporary variable first, and then copy that variable into the array, Verilator matches Questa and my expectation.  This is done by commenting out line 31 and uncommenting lines 32 and 33.

```
harris@chips:~/cvw/examples/verilog/testcase$ verilator --timescale "1ns/1ns" --timing --binary --top-module testbench testbench.sv
...
harris@chips:~/cvw/examples/verilog/testcase$ ./obj_dir/Vtestbench
Read vectors.tv
sig32[0] = 00000010   line: 00000010

sig32[1] = 00000011   line: 00000011

sig32[2] = 00000012   line: 00000012

- testbench.sv:42: Verilog $finish
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

See attached testbench.sv and vectors.tv. (remove .txt suffix)
[vectors.tv.txt](https://github.com/verilator/verilator/files/13854177/vectors.tv.txt)
[testbench.sv.txt](https://github.com/verilator/verilator/files/13854178/testbench.sv.txt)

What 'verilator' command line do we use to run your example?

see above

What 'verilator --version' are you using?  Did you try it with the git master version?

harris@chips:~/cvw/examples/verilog/testcase$ verilator --version
Verilator 5.021 devel rev v5.020-16-g7507c5f56

What OS and distribution are you using?

Ubuntu 22.04LTS

May we assist you in trying to fix this in Verilator yourself?

Not my strength.

wsnyder, 2024-01-07:
This is due to a side effect of guarding that the array assignment is not out-of-bounds.  Verilator inserts a new statement with the actual array assignment, but this is after the other statements in the if.

The workaround is to use a temporary as you did.

The real fix is complicated, we'll need some new special select operator that is side-effect safe.

davidharrishmc, 2024-01-08:
Sorry to hear that it is a hard fix.

Is it easy to put in a warning that Verilator doesn't (or potentially doesn't) handle such code constructs reliably, and maybe suggest the workaround?  It cost me several hours to get to the root cause, and disrupts faith in that I don't know what other constructs might not be compatible with Verilator.

wsnyder, 2024-01-11:
>Is it easy for lint to emit a warning that Verilator doesn't (or potentially doesn't) handle such code constructs reliably, and maybe suggest the workaround? It cost me several hours to get to the root cause, and disrupts faith in that I don't know what other constructs might not be compatible with Verilator.

Not immediately obvious how to avoid false positives but will look into it.