verilator/verilator #4813: Incremental and parallel verilation in hierarchical mode

futurehome, 2024-01-09:
Thanks for taking the time to report this.

What would you like added/supported?

1. The design is large, so I use hierarchical mode for verilation, I hope when running verilation, it will regenerate sub-modules only with which RTL is updated, and all other sub-modules will not be re-generated. But in actual, all the sub-modules will be regenrated even none of the RTL is updated. 
2. Because there are a lot of sub-modules, how to run the sub-modules in different sessions(Not just threads)? Looks like it will execute all the makefiles one by one.

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this in Verilator yourself?

futurehome, 2024-01-09:
For 1, I think it's because the V*_hierMkArgs.f are always update when re-running  verilator.

wsnyder, 2024-01-09:
Can you explain more about what you mean by executing in different sessions?

futurehome, 2024-01-10:
I mean all the V*_hierMkargs.f can be executed in parallel.

futurehome, 2024-01-10:
BTW, because my design is extremely large, it takes several hours to verilate the whole design. But in hierarchical mode, there are several limitations, like no trace support, etc. Do you have plan to support trace for hierarchical mode?

wsnyder, 2024-01-11:
There are several open issues on hierarchical.  AFAIK there's no one actively fixing these so they are likely to be open for a long time.  We'd welcome your or others' work on them; feel free to post to the issues that you'd like to work on.

closed 2024-01-11 by wsnyder