verilator/verilator #4814: inconsistent results on signed operations

poemonsense, 2024-01-09:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
```verilog
module test();
  reg clock;
  initial begin
    clock = 0;
  end
  always #1 clock = ~clock;

  wire [63:0] in1 = 64'hfffffffbffffffff;
  wire [63:0] in2 = 64'hffffffffffffffdf;

  wire [63:0] div_data = $signed(in1) / $signed(in2);
  wire [63:0] result1 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      div_data;

  wire [63:0] result2 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      $signed(in1) / $signed(in2);

  always @(posedge clock) begin
    if (result1 != result2) begin
      $display("div_data: %x, $signed(in1) / $signed(in2): %x",
        div_data, $signed(in1) / $signed(in2));
      $display("result1: %x, result2: %x", result1, result2);
      $error();
    end
  end

endmodule
```

`result1` and `result2` are expected to be the same, is it? However, it says
```
div_data: 000000001f07c1f0, $signed(in1) / $signed(in2): 000000001f07c1f0
result1: 000000001f07c1f0, result2: 0000000000000000
[1] %Error: test.v:27: Assertion failed in TOP.test
%Error: test.v:27: Verilog $stop
Aborting...
```

What 'verilator' command line do we use to run your example?
`verilator --cc --exe --main --timing test.v`

What 'verilator --version' are you using?  Did you try it with the git master version?
`Verilator 5.021 devel rev v5.020-22-g19cb366e5`. It's the latest master branch.

What OS and distribution are you using?
Ubuntu 20.04

May we assist you in trying to fix this in Verilator yourself?
I'm not sure whether this is a bug or feature. I tried O0-O3 but all version produces the same result. I'm not having VCS installed now but I could have a try in the next days if needed.

poemonsense, 2024-01-09:
It looks like the inline version is treated as an unsigned operation.

If we change the operands to
```
wire [63:0] in1 = 64'hffffffffffffffff;
wire [63:0] in2 = 64'h2;
```

The results will be `result1: 0000000000000000, result2: 7fffffffffffffff`

wsnyder, 2024-01-09:
The "failure" is consistent with IEEE rules. Verilator results agree with other simulators.  Maybe you intended "wire signed"?

udif, 2024-01-09:
First, Verilator doesn't agree with other simulators. The test **passes** with Riviera on [edaplayground.com](https://edaplayground.com/x/H2py) (Other than replacing `always @(posedge clock)` with `initial`).

Second, Verilator doesn't agree with **itself**.

Please compare the following:

t1.v
```Verilog
module test();
  reg clock;
  initial begin
    clock = 0;
  end
  always #1 clock = ~clock;

  wire [63:0] in1 = 64'hfffffffbffffffff;
  wire [63:0] in2 = 64'hffffffffffffffdf;

  wire [63:0] div_data = $signed(in1) / $signed(in2);
  wire [63:0] result1 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      div_data;

  wire [63:0] result2 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      $signed(in1) / $signed(in2);

initial begin
    if (result1 != result2) begin
      $display("div_data: %x, $signed(in1) / $signed(in2): %x",
        div_data, $signed(in1) / $signed(in2));
      $display("result1: %x, result2: %x", result1, result2);
      $error();
    end
    $finish;
end

endmodule
```

t3.v
```
module test();
  reg clock;
  initial begin
    clock = 0;
  end
  always #1 clock = ~clock;

  reg [63:0] in1;
  always @(posedge clock) in1 = 64'hfffffffbffffffff;
  reg [63:0] in2;
  always @(posedge clock) in2 = 64'hffffffffffffffdf;

  wire [63:0] div_data = $signed(in1) / $signed(in2);
  wire [63:0] result1 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      div_data;

  wire [63:0] result2 =
    in2 == 64'h0 ? 64'hFFFFFFFFFFFFFFFF :
      $signed(in1) / $signed(in2);

initial begin
    @(posedge clock)
    if (result1 != result2) begin
      $display("div_data: %x, $signed(in1) / $signed(in2): %x",
        div_data, $signed(in1) / $signed(in2));
      $display("result1: %x, result2: %x", result1, result2);
      $error();
    end
    $finish;
end

endmodule
```

I've added an artificial time step in `t3.v`.
Why?
Because `t1.v` is statically compiled:
```
% grep result2 ./obj_dir/Vt1*.cpp
./obj_dir/Vt1___024root__DepSet_h1d7c7c72__0__Slow.cpp:    VL_WRITEF("div_data: 000000001f07c1f0, $signed(in1) / $signed(in2): 000000001f07c1f0\nresult1: 000000001f07c1f0, result2: 0000000000000000\n[%0t] %%Error: t1.v:25: Assertion failed in %Ntest\n",
```
All the calculations are done at compile time.

With t3.v, it is done at runtime:
```
% grep result2 ./obj_dir/Vt3*.cpp
./obj_dir/Vt3___024root__DepSet_h30230c7b__0.cpp:        VL_WRITEF("div_data: %x, $signed(in1) / $signed(in2): %x\nresult1: %x, result2: %x\n[%0t] %%Error: t3.v:28: Assertion failed in %Ntest\n",
```

The runtime results are different too:
```
% ./obj_dir/Vt1
div_data: 000000001f07c1f0, $signed(in1) / $signed(in2): 000000001f07c1f0
result1: 000000001f07c1f0, result2: 0000000000000000
[0] %Error: t1.v:25: Assertion failed in TOP.test
%Error: /home/udif/t1.v:25: Verilog $stop
Aborting...
Abort (core dumped)
% ./obj_dir/Vt3
- /home/udif/t3.v:30: Verilog $finish
```

So it seems something is broken in the constant expression handling logic.
I've ran  this on: `Verilator 5.021 devel rev v5.020-11-g1a0c8f157`

wsnyder, 2024-01-09:
Can you please convert your examples to tests that pass on edaplayground, and $stop on verilator, as opposed to looking at outputs?

udif, 2024-01-09:
After spending more time, and carefully running some examples again, I would like to retract my claims.

I ran the following examples against 2 commercial simulators, both producing identical results:
```Verilog
module t;
  wire [31:0]x1 = 32'hffffffff & ($signed(32'hffff_8000) / $signed(32'hffff_ff80));
  wire [31:0]x2 =                ($signed(32'hffff_8000) / $signed(32'hffff_ff80));
  initial
    #0
    $display("result: x1=%x x2=%x", x1, x2);
endmodule
```
The output is:
`result: x1=00000000 x2=00000100`

If we drop the `#0` we get:
```Verilog
module t;
  wire [31:0]x1 = 32'hffffffff & ($signed(32'hffff_8000) / $signed(32'hffff_ff80));
  wire [31:0]x2 =                ($signed(32'hffff_8000) / $signed(32'hffff_ff80));
  initial
    $display("result: x1=%x x2=%x", x1, x2);
endmodule
```
The output is:
`result: x1=xxxxxxxx x2=xxxxxxxx`

From what I see:
1. `initial` blocks are triggered before wires are initialized. Therefore the 2nd example produces X's.
2. The result of the 1st example can be explained by the following partial quote from section 11.8.2, describing how to determine the signess of an expression:
*Propagate the type and size of the expression (or self-determined subexpression) back down to the
context-determined operands of the expression. In general, any context-determined operand of an
operator shall be the same type and size as the result of the operator.*

If I understand (2) correctly:
* x1 is determined to be unsigned due to the unsigned `&` operator and it's 32'hffff_ffff left hand side argument, therefore all its arguments are *unsigned*, regardless of the `$signed()` calls on the right hand side of `&`. This yields an unsigned division 0xffff_8000 / 0xffff_ff80 which is less than 1, or simply zero. This is definitely non-intuitive, but this is what I understand from the spec and my experiments.
* x2, however, is determined to be signed. (According to the 1st paragraph of 11.8.1, the left hand side of the assignment is not taken into account for determining signess) , and so its expression is  (-32768/-128)==256 == 0x100.

As for Icarus Verilog, v0.9.7 failed to treat x2 as an unsigned division , while 0.10.0 agrees with the other simulators.
Both Icarus versions fails to run the initial block before the wire assignments, producing the same result as the '#0' version above, and not X's like the commercial sims.

Verilator correctly handles the signed/unsigned differences as shown by `x1` and `x2`, but produces the same result with or without #0, probably because it doesn't support X's anyhow, although I did expect it to handle the scheduling nuances, and produce an all 0's output perhaps for the 2nd example (without #0).

Conclusion:
* @poemonsense's issues were due to the peculiar signed/unsigned evaluation rules in 11.8.2 (an unsigned result is unsigned all the way down...).
* My separate tests produced anomalies because replacing the clock events with an `initial` block used x1,x2 before they had a chance to be initialized by the wire continuous assignment.
Therefore, in my opinion, there are no further issues to check in the original question.

poemonsense, 2024-01-10:
> 2. The result of the 1st example can be explained by the following partial quote from section 11.8.2, describing how to determine the signess of an expression:
>    _Propagate the type and size of the [...]

I still doubt the behaviors here are correct. As far as I've understood, @udif cited the spec to explain the non-intuitive conversion of the division operands (instead of the results). Given the overall expression should be unsigned, even if the operands of the division is signed, it will be implicitly converted to unsigned for division.

Here're the are I tried on Verilator.
```
  wire [31:0] in1 = 32'hffffffff;
  wire [31:0] in2 = 32'h2;

  wire [31:0] div_data_u = $signed(in1) / $signed(in2);
  wire signed [31:0] div_data_s = $signed(in1) / $signed(in2);

  wire [31:0] result0 = 32'hFFFFFFFF & div_data_u;
  wire [31:0] result1 = 32'hFFFFFFFF & $signed(div_data_u);
  wire [31:0] result2 = 32'hFFFFFFFF & div_data_s;
  wire [31:0] result3 = 32'hFFFFFFFF & $unsigned(div_data_s);
  wire [31:0] result4 = 32'hFFFFFFFF & ($signed(in1) / $signed(in2));
  wire [31:0] result5 = 32'hFFFFFFFF & (in1 / in2);
  wire [31:0] result6 = 32'hFFFFFFFF & $unsigned($signed(in1) / $signed(in2));
  wire [31:0] result7 = 32'hFFFFFFFF & $signed($signed(in1) / $signed(in2));
  wire [31:0] result8 = $signed(32'hFFFFFFFF) & div_data_u;
  wire [31:0] result9 = $signed(32'hFFFFFFFF) & $signed(div_data_u);
  wire [31:0] result10 = $signed(32'hFFFFFFFF) & div_data_s;
  wire [31:0] result11 = $signed(32'hFFFFFFFF) & $unsigned(div_data_s);
  wire [31:0] result12 = $signed(32'hFFFFFFFF) & ($signed(in1) / $signed(in2));
  wire [31:0] result13 = $signed(32'hFFFFFFFF) & (in1 / in2);
  wire [31:0] result14 = $signed(32'hFFFFFFFF) & $unsigned($signed(in1) / $signed(in2));
  wire [31:0] result15 = $signed(32'hFFFFFFFF) & $signed($signed(in1) / $signed(in2));
  wire signed [31:0] result16 = $signed(32'hFFFFFFFF) & div_data_u;
  wire signed [31:0] result17 = $signed(32'hFFFFFFFF) & $signed(div_data_u);
  wire signed [31:0] result18 = $signed(32'hFFFFFFFF) & div_data_s;
  wire signed [31:0] result19 = $signed(32'hFFFFFFFF) & $unsigned(div_data_s);
  wire signed [31:0] result20 = $signed(32'hFFFFFFFF) & ($signed(in1) / $signed(in2));
  wire signed [31:0] result21 = $signed(32'hFFFFFFFF) & (in1 / in2);
  wire signed [31:0] result22 = $signed(32'hFFFFFFFF) & $unsigned($signed(in1) / $signed(in2));
  wire signed [31:0] result23 = $signed(32'hFFFFFFFF) & $signed($signed(in1) / $signed(in2));
```

So here comes the results:
```
result0:  00000000
result1:  00000000
result2:  00000000
result3:  00000000
result4:  7fffffff
result5:  7fffffff
result6:  00000000
result7:  00000000
result8:  00000000
result9:  00000000
result10: 00000000
result11: 00000000
result12: 00000000
result13: 7fffffff
result14: 00000000
result15: 00000000
result16: 00000000
result17: 00000000
result18: 00000000
result19: 00000000
result20: 00000000
result21: 7fffffff
result22: 00000000
result23: 00000000
```

If what @udif explained is correct, `result4` and `result5` are the same. However, I noticed that `result12 != result13` and `result20 != result21`. If the conversion applies to the operands instead of the division result, they should be the same, right?

I'm not a SV spec expert. Just present these results for discussion. Thank you both for the time.

udif, 2024-01-10:
The example you prepared shows exactly what I wrote.

result12 is all signed, thereby producing 32'h00000000 as expected (-0.5 rouned to zero)
result13 is signed ANDed with an unsigned divide, giving unsigned result of 32'h7ffffff

result20 and result21 are identical to result12 and result13 respectively, because the rules says you don't look at the LHS of an assignment to determine the sign-ness of the RHS. You look only at the RHS itself.

Exact quote:
>11.8.1 Rules for expression types 
>The following are the rules for determining the resulting type of an expression:
> — Expression type depends only on the operands. It does not depend on the left-hand side (if any).

Generally speaking, for an expression to be signed it has to be purely signed. almost anything breaks the "sign-ness" of an expression:
* using a bit select or part select
* concatenation
* comparison result
* all non-decimal numbers are unsigned unless explicitly signed (e.g. 32'hs00000000)
* using any dyadic operator where one argument is unsigned and the other is signed

poemonsense, 2024-01-10:
> The example you prepared shows exactly what I wrote.
> 
> result12 is all signed, thereby producing 32'h00000000 as expected (-0.5 rouned to zero) result13 is signed ANDed with an unsigned divide, giving unsigned result of 32'h7ffffff

Why isn't `result13` the same as `result12` due to the implicit conversion to signed division? I thought the reason for `result4:  7fffffff` is that the division is converted to unsigned division because the expression is an unsigned `32'hFFFFFFFF & `. I assume following this rule an unsigned division should be converted to a signed division in `result12` because it is a signed AND.

poemonsense, 2024-01-10:
If you are saying `&` is for unsigned only, I change the operation to `+`. There is still `result12 != result13`. I thought they should be the same.

```verilog
  wire [31:0] result4 = 32'hFFFFFFFF + ($signed(in1) / $signed(in2));
  wire [31:0] result5 = 32'hFFFFFFFF + (in1 / in2);

  wire [31:0] result12 = $signed(32'hFFFFFFFF) + ($signed(in1) / $signed(in2));
  wire [31:0] result13 = $signed(32'hFFFFFFFF) + (in1 / in2);
```

Results:
```
result4:  7ffffffe
result5:  7ffffffe

result12: ffffffff
result13: 7ffffffe
```

I don't understand why (for `result4`) a signed division can be implicitly converted to an unsigned division because the other operand for ADD is unsigned, but (for `result13`) an unsigned division is not converted to a signed division because the other operand for ADD is signed.

wsnyder, 2024-01-12:
in1 and in2 are not "wire signed" so are unsigned values.  The rules are described in the IEEE standard for sign promotion.

closed 2024-01-12 by wsnyder

poemonsense, 2024-01-12:
> in1 and in2 are not "wire signed" so are unsigned values. The rules are described in the IEEE standard for sign promotion.

I tried `wire signed` but there still comes `result5:  7ffffffe`.

```
wire signed [31:0] in1 = 32'shffffffff;
wire signed [31:0] in2 = 32'sh2;

wire [31:0] result5 = 32'hFFFFFFFF + (in1 / in2);
```

So I don't agree it's the `wire` or `wire signed` that takes effects. The issue is that, `unsigned_A + (signed_B / signed_C)` will result in an ***unsigned*** division, even if both operands are ***signed***. This behavior is not affected by whether `signed_B` and `signed_C` is declared explicitly as `wire signed` or implicitly via `$signed()`.

poemonsense, 2024-01-15:
Thanks for your time. I just confirmed Verilator's behaviors against VCS.

Given `wire [31:0] in1 = 32'hffffffff` and `wire [31:0] in2 = 32'h2`, we have:

```
32'hFFFFFFFF + (in1 / in2)                          -> 7ffffffe
32'hFFFFFFFF + ($signed(in1) / $signed(in2))        -> 7ffffffe
32'hFFFFFFFF + $signed($signed(in1) / $signed(in2)) -> ffffffff
```

I'm going to submit issues for Chisel/CIRCT for generating inconsistent Verilog files...