verilator/verilator #4819: Compilation error on interface class usage

esynr3z, 2024-01-10:
Hello! The code below does not compile.

```verilog
module reproducer;
  interface class Courier;
    pure virtual function deliver();
  endclass

  class Person implements Courier;
    virtual function deliver();
      $display("Doing slow delivery");
    endfunction
  endclass

  class Vehicle; endclass
  class Car extends Vehicle implements Courier;
    virtual function deliver();
      $display("Doing fast delivery");
    endfunction
  endclass

  function do_delivery(Courier courier);
    courier.deliver();
  endfunction

  initial begin
    Car truck = new();
    do_delivery(truck); // !!!! this leads to compilation error
    $finish();
  end
endmodule
````

Error as follows:

```bash
In file included from /space/tools/verilator/v5.020/share/verilator/include/verilated.h:942,
                 from Vreproducer__pch.h:21,
                 from Vreproducer.cpp:4,
                 from Vreproducer__ALL.cpp:3:
/space/tools/verilator/v5.020/share/verilator/include/verilated_types.h: In instantiation of ‘VlClassRef<T_Class>& VlClassRef<T_Class>::operator=(const VlClassRef<T_OtherClass>&) [with T_OtherClass = Vreproducer_reproducer__03a__03aCar; T_Class = Vreproducer_reproducer__03a__03aCourier]’:
Vreproducer_reproducer__DepSet_h37a7cadd__0__Slow.cpp:22:39:   required from here
/space/tools/verilator/v5.020/share/verilator/include/verilated_types.h:1681:20: error: comparison between distinct pointer types ‘Vreproducer_reproducer__03a__03aCourier*’ and ‘Vreproducer_reproducer__03a__03aCar*’ lacks a cast
 1681 |         if (m_objp == copied.m_objp) return *this;
      |             ~~~~~~~^~~~~~~~~~~~~~~~
/space/tools/verilator/v5.020/share/verilator/include/verilated_types.h:1683:25: error: cannot convert ‘Vreproducer_reproducer__03a__03aCar* const’ to ‘Vreproducer_reproducer__03a__03aCourier*’ in assignment
 1683 |         m_objp = copied.m_objp;
      |                  ~~~~~~~^~~~~~
      |                         |
      |                         Vreproducer_reproducer__03a__03aCar* const
make: *** [/space/tools/verilator/v5.020/share/verilator/include/verilated.mk:269: Vreproducer__ALL.o] Error 1
make: Leaving directory '/space/projects/verilator-bug-report/obj_dir'
%Error: make -C obj_dir -f Vreproducer.mk -j 1 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /space/tools/verilator/v5.020/bin/verilator_bin reproducer.sv --cc --binary --timing -Wno-IGNOREDRETURN -O1 -o Vreproducer --top-module reproducer
```

>What 'verilator' command line do we use to run your example?
```bash
verilator reproducer.sv --cc --binary --timing -Wno-IGNOREDRETURN -O1 -o Vreproducer --top-module reproducer
```

>What 'verilator --version' are you using?  Did you try it with the git master version?

Verilator 5.020 2024-01-01 rev v5.020

>What OS and distribution are you using?

Ubuntu 22.04.3 LTS

wsnyder, 2024-01-12:
Verilator hadn't previously handled multiple inheritance.  V3EmitCFunc.h needs to put out multiple "public virtual" in the class declaration for the implements classes. Also the constructor for the class needs to then construct the implements classes in V3EmitCHeaders.cpp.

Another related small part of this is to properly include the base classes. I just committed a fix for this.

If this something you'd be willing to look at a patch for?

esynr3z, 2024-01-13:
I'd like to help, but unfortunately I have no expertise to assist or review C++ code. However, if you provide some information, I'll try.

closed 2024-01-24 by wsnyder