verilator/verilator #4820: commit 95c4ade causes new maybe-uninitialized errors/warnings when building user code

ldoolitt, 2024-01-12:
I'm sure @gezalore's commit 95c4ade (2023-10-24) is great in general, but it has the side effect of causing new compiler warnings/errors when building user code.
For example with g++ 12.2.0:
```
/redacted/share/verilator/include/verilated_fst_c.cpp: In member function ‘void VerilatedFst::declare(uint32_t, const char*, int, VerilatedTraceSigDirection, VerilatedTraceSigKind, VerilatedTraceSigType, bool, int, bool, int, int)’:
/redacted/share/verilator/include/verilated_fst_c.cpp:218:33: warning: ‘varDir’ may be used uninitialized [-Wmaybe-uninitialized]
  218 |             = fstWriterCreateVar(m_fst, varType, varDir, bits, name_str.c_str(), 0);
      |               ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/redacted/share/verilator/include/verilated_fst_c.cpp:177:15: note: ‘varDir’ was declared here
  177 |     fstVarDir varDir;
      |               ^~~~~~
```

Depending on some mix of configuration and compiler version, I've seen it show up either as a harmless-but-ugly warning, or as a fatal error (`cc1plus: all warnings being treated as errors`).

A brainless one-line patch eliminates the regression, and is unlikely to change any actual behavior:
```
diff --git a/include/verilated_fst_c.cpp b/include/verilated_fst_c.cpp
index 79c5c9ee9..ffc78ec99 100644
--- a/include/verilated_fst_c.cpp
+++ b/include/verilated_fst_c.cpp
@@ -174,7 +174,7 @@ void VerilatedFst::declare(uint32_t code, const char* name, int dtypenum,

     if (dtypenum > 0) fstWriterEmitEnumTableRef(m_fst, m_local2fstdtype[dtypenum]);

-    fstVarDir varDir;
+    fstVarDir varDir = FST_VD_IMPLICIT;
     switch (direction) {
     case VerilatedTraceSigDirection::INOUT: varDir = FST_VD_INOUT; break;
     case VerilatedTraceSigDirection::OUTPUT: varDir = FST_VD_OUTPUT; break;
```

Of course, one can also work around the effect by adding `-CFLAGS -Wno-maybe-uninitialized`
to the verilator command line.

wsnyder, 2024-01-12:
Thanks for determining the fix. Could you make that into a pull please?

wsnyder, 2024-01-12:
Thanks for the pull.