verilator/verilator #4821: Verilator sometimes misorders `$dumpvars` before `$dumpfile`

FanShupei, 2024-01-12:
## Can you attach an example that shows the issue?
Here is a minified show case extracted from the problem I encountered in my project.

It contains two file `top.sv` and `top_dpi.cpp`. The `top_dpi.cpp` is just trivial.

```sv
// top.sv
`timescale 1ns/1ps

import "DPI-C" function string dpi_waveform_filename();
import "DPI-C" function int dpi_waveform_enable();

module sim_top;
    bit clk;

    clock_gen U_clock(.clk);
    initial begin
        if (dpi_waveform_enable() != 0) begin
            automatic string waveform_filename = dpi_waveform_filename();
            $dumpfile(waveform_filename);
            $dumpvars;
        end

        #10 $finish();
    end
endmodule

module clock_gen(clk);
    output bit clk = 1'b0;
    always #1 clk = ~clk;
endmodule
```

```c++
// top_dpi.cpp
#include "Vtop__Dpi.h"
int dpi_waveform_enable() {
    return 1;
}
const char* dpi_waveform_filename() {
    return "wave.vcd";
}
```

## What 'verilator' command line do we use to run your example?

`verilator --cc --trace-fst --timing --timescale 1ns/1ps --top-module sim_top sim/sim_top.sv`

It's expected generated a waveform `wave.vcd`.

However, it gives following warning without any waveform.

```
%Warning: $dumpvar ignored as not preceded by $dumpfile
- top.sv:17: Verilog $finish
```

## What 'verilator --version' are you using?

`Verilator 5.020 2024-01-01 rev UNKNOWN.REV` on ArchLinux

May we assist you in trying to fix this in Verilator yourself?

# Analysis

I inspect `Vtop___024root__DepSet_h84412442__0.cpp`, and the following code seems suspicious.
```c++
VL_INLINE_OPT VlCoroutine Vtop___024root___eval_initial__TOP__Vtiming__0(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___eval_initial__TOP__Vtiming__0\n"); );
    // Init
    IData/*31:0*/ __Vfunc_dpi_waveform_enable__0__Vfuncout;
    __Vfunc_dpi_waveform_enable__0__Vfuncout = 0;
    std::string __Vfunc_dpi_waveform_filename__1__Vfuncout;
    // Body
    if ((0U != ([&]() {
                    Vtop___024unit____Vdpiimwrap_dpi_waveform_enable_TOP____024unit(__Vfunc_dpi_waveform_enable__0__Vfuncout);
                }(), __Vfunc_dpi_waveform_enable__0__Vfuncout))) {
        Vtop___024unit____Vdpiimwrap_dpi_waveform_filename_TOP____024unit(__Vfunc_dpi_waveform_filename__1__Vfuncout);
        vlSymsp->_traceDumpOpen();
        vlSelf->sim_top__DOT__unnamedblk1__DOT__waveform_filename 
            = __Vfunc_dpi_waveform_filename__1__Vfuncout;
        vlSymsp->_vm_contextp__->dumpfile(VL_CVT_PACK_STR_NN(vlSelf->sim_top__DOT__unnamedblk1__DOT__waveform_filename));
    }
    co_await vlSelf->__VdlySched.delay(0x2710ULL, nullptr, 
                                       "top.sv", 17);
    VL_FINISH_MT("top.sv", 17, "");
}
```

If I understand correctly, `vlSymsp->_traceDumpOpen()` corresponds to `$dumpvar`. It explains everything, but I have no idea how and why Verilator performs the reordering.

FanShupei, 2024-01-12:
The problem is reproducible on master `Verilator 5.021 devel rev v5.020-25-gfcdd8e5da`, with the same behavior.

Some more explanation on the show case. The dpi wrapper just acts as a blackbox. The bug is only triggered with specific code structure. If I tweak it slightly, the bug disappears.

```
initial begin // bad
    if (dpi_waveform_enable() != 0) begin
        automatic string waveform_filename = dpi_waveform_filename();
        $dumpfile(waveform_filename);
        $dumpvars;
    end
    #10 $finish();
end

initial begin // good
    automatic string waveform_filename = dpi_waveform_filename();
    $dumpfile(waveform_filename);
    $dumpvars;
    #10 $finish();
end

initial begin // good
    if (dpi_waveform_enable() != 0) begin
        automatic string waveform_filename = "wave.vcd";
        $dumpfile(waveform_filename);
        $dumpvars;
    end
    #10 $finish();
end

initial begin // good
    if (dpi_waveform_enable() != 0) begin
        $dumpfile(dpi_waveform_filename());
        $dumpvars;
    end
    #10 $finish();
end
```

wsnyder, 2024-01-12:
Thanks for the good test case.  -fno-reorder is a workaround for this, fix pushed.

closed 2024-01-12 by wsnyder