verilator/verilator #4823: Unrecognized argument error while building Verilator from git

Shreesh-Kulkarni, 2024-01-12:
[image]

I came across this error while building verilator from git master. I followed the documentation for installation steps.
I could have used the package manager version, but for my specific project, the version from the package manager install isn't supported. Hence, the git master version is suitable.

I am running the build on WSL2 with Ubuntu 20.04 LTS.

I would appreciate any suggestions to fix this error myself, so that I can contribute for the same.

wsnyder, 2024-01-12:
Either upgrade "bear" to >= version 3.0, or unistall bear, or use

`make BEAR=`

Closing as this should solve it but feel free to followup here.

P.S. in future please avoid screenshots and post text, or convert screenshot to text with OCR. Thanks.

closed 2024-01-12 by wsnyder

Shreesh-Kulkarni, 2024-01-12:
Thanks! I'll keep it in mind to use text instead next time.