verilator/verilator #4824: Invalid verilated code generated

ollie-etl, 2024-01-12:
I'm running a testbench (via cocotb) with a parameterised data width. At 256 bits, all is well. At 512 bits, the generated code is invalid. I'm not going to be able to release the code, but i'll try and provide as much info as i can.

The error is this:

```console
INFO     cocotb:simulator.py:302 Vtop___024root__DepSet_heccd7ead__1.cpp: In function ‘void Vtop___024root___eval(Vtop___024root*)’:
INFO     cocotb:simulator.py:302 Vtop___024root__DepSet_heccd7ead__1.cpp:5334:5: error: ‘__Vilp’ was not declared in this scope
```

Looking at that line, we have this:

```cpp
void Vtop___024root___eval(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___eval\n"); );
    // Init
    __Vilp = 0U;
    while ((__Vilp <= 0x2aU)) {
         // I've changed signal names here
         vlSelf->__Vsampled__TOP__ip__DOT__ip1__DOT__m03_axis_f__DOT__axis_packed[__Vilp] 
            = vlSelf->ip__DOT__ip1__DOT__m03_axis_f__DOT__axis_packed[__Vilp];
        __Vilp = ((IData)(1U) + __Vilp);
    }
```
using grep -R __Vilp, I can find _nowhere_ in the generated verilated code which declares `__Vilp`

Comparing to the code in the 256 bit width version

```cpp
void Vtop___024root___eval(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___eval\n"); );
      // Init
      vlSelf->__Vsampled__TOP__ip__DOT__ip1__DOT__m03_axis_f__DOT__axis_packed[0u] 
            = vlSelf->ip__DOT__ip1__DOT__m03_axis_f__DOT__axis_packed[0u];
```

It looks like in this case, it does the initalisation with loop unrolling. It looks like in the none unrolled case, the variable needs declaring first?

I've tried this on verilator 5.010, and again at 5.020. I'm on ubuntu, but all the tools are packaged with nix

The command invoking verilator is below. I've stripped out an aweful lot of includes
```console
perl /nix/store/g352nrsp61alrgqsd2825wqbqfg82srb-verilator-5.020/bin/verilator -cc --exe \
    -Mdir /home/user/ip/sim_build/ip-4-1-512 \
    -DCOCOTB_SIM=1 \
    --top-module ip \
    --vpi \
    --public-flat-rw \
    --prefix Vtop \
    -o ip \
    -LDFLAGS -Wl,-rpath,/nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/libs \
    -L/nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/libs \
    -lcocotbvpi_verilator \
    -I/home/user/ip0 \
    -I/home/user/ip1 \
    -I/home/user/ip2 \
   -DOBSOLETE -DONESPIN \
   -Wno-fatal \
   --assert -DFORMAL \
   --coverage-line \
    -j 0 \
   -O3 \
   -CFLAGS -O3 \
   --threads 24 \
   -GDATA=512 \
   -GID=1 \
   -GDEST=4 \
   /nix/store/bvcp6l3g4ac52gv3fjl8ig1dyfwwhczv-python3-3.9.13-env/lib/python3.9/site-packages/cocotb/share/lib/verilator/verilator.cpp \
   .verilator.vlt \
   /home/user/ip/ip.sv
```

wsnyder, 2024-01-12:
Please run with --debug and see if the VAR is declared in a .tree file, and if so the name of the last file it appears in.  Presumably something is removing it.

ollie-etl, 2024-01-12:
wow, thats a lot of output. I'm going to guess at te interesting bits

```console
O     cocotb:simulator.py:302 - V3Unroll.cpp:510:   unrollAll:
INFO     cocotb:simulator.py:302 - V3Unroll.cpp:209:   Unable to simulate
INFO     cocotb:simulator.py:302 - V3Unroll.cpp:61:       Can't Unroll: Unable to simulate loop :WHILE 0x9307880 <e25892> {n127af}
INFO     cocotb:simulator.py:302 - V3Unroll.cpp:61:       Can't Unroll: Unable to simulate loop :WHILE 0x8763dd0 <e105680> {p1114ah}
```

There are hundreds of these
```console
INFO     cocotb:simulator.py:302 - V3Simulate.h:381:   Unknown node type in SimulateVisitor: CEXPR
INFO     cocotb:simulator.py:302 - V3Simulate.h:381:   Unknown node type in SimulateVisitor: SAMPLED
```

But probably this is most relevant (and looks like my bug
```console
INFO     cocotb:simulator.py:302 %Error: Internal Error: /home/user/ip/axis_formal.sv:73:22: ../V3Broken.cpp:284: Local variable not in scope where referenced: VAR 0x1f340790 <e7531026#> {m73aw} @dt=0x1874e70@(G/w32)  __Vilp STMTTEMP
INFO     cocotb:simulator.py:302 -node: VARREF 0x1f3409d0 <e7531037#> {m73aw} @dt=0x1874e70@(G/w32)  __Vilp [LV] => VAR 0x1f340790 <e7531026#> {m73aw} @dt=0x1874e70@(G/w32)  __Vilp STMTTEMP
INFO     cocotb:simulator.py:302    73 |     DATA) -2 -1 : 0] axis_packed;
```

Ok, i did some digging around there. That line is a variable declaration, with several layers of macros. Manually expanding doesn't help
```console
INFO     cocotb:simulator.py:302 %Error: Internal Error: /home/user/ip/axis_formal.sv:75:128: ../V3Broken.cpp:284: Local variable not in scope where referenced: VAR 0x1f34da40 <e7531026#> {m75ey} @dt=0x1874e70@(G/w32)  __Vilp STMTTEMP
INFO     cocotb:simulator.py:302 -node: VARREF 0x1f34dc80 <e7531037#> {m75ey} @dt=0x1874e70@(G/w32)  __Vilp [LV] => VAR 0x1f34da40 <e7531026#> {m75ey} @dt=0x1874e70@(G/w32)  __Vilp STMTTEMP
INFO     cocotb:simulator.py:302    75 |   logic [(2 + (HAS_LAST ? 1: 0) + (HAS_STRB ? (DATA/8) : 0) + (HAS_KEEP ? (DATA/8) : 0) + ID + DEST + USER +  DATA) -2 -1 : 0] axis_packed;
```

I've checked, and all those variables are in scope, and declared in the parameter list of the module. This only seems to be an issue when DATA == 512, not DATA 256 or less.

ollie-etl, 2024-01-12:
I'm not exactly sure of the values of the other variables, as there a few instances of this in this design, but USER ~750, so the width of this vector will be something like ~1500 in total. Nothing too outrageous.

wsnyder, 2024-01-13:
>INFO     cocotb:simulator.py:302 -node: VARREF 0x1f3409d0 <e7531037#> {m73aw} @dt=0x1874e70@(G/w32)  __Vilp [LV] => VAR 0x1f340790 <e7531026#> {m73aw} @dt=0x1874e70@(G/w32)  __Vilp STMTTEMP

Yes, that's it.  In what .tree file did the VAR last exist, then disappear?

There's something different about where the loop is that's causing the bug, try a little example with that loop in a similar code place.

The width of signals doesn't directly matter, it's just related to crossing unroll loop thresholds.

ollie-etl, 2024-01-13:
Not exactly sure what you mean, but assuming the numbers are meaningful, the list returned by

> grep -cRL __Vilp sim_build/test_ip-4-1-512/*.tree

gives a list, with `sim_build/test_ip-4-1-512/Vtop_080_deadAll.tree` the highest numbered .tree file

wsnyder, 2024-01-17:
Can you attach the --debug stdout?  You can remove the V3Simulate warnings.

wsnyder, 2024-02-14:
I'm presuming this got figured out; otherwise please reopen with an example we can use to reproduce the issue.

ollie-etl, 2024-02-14:
Not figured out, but worked around by simulating something slightly different. Will reopen if this reappears

justinyaodu, 2024-03-02:
If anyone else runs into this, a workaround I used was adding [-fno-reloop](https://veripool.org/guide/latest/exe_verilator.html#cmdoption-fno-reloop) to the verilator command line.

wsnyder, 2024-03-02:
@justinyaodu if you can provide a standalone verilog code example that shows the issue, please post and reopen.

justinyaodu, 2024-03-03:
Verilator command line:

```sh
verilator --timing --binary --assert --top-module testbench testbench.sv
```

Verilator version: commit 5e1fc6e

OS: Ubuntu 24.04

Contents of `testbench.sv`:

```sv
parameter WIDTH = 8;
parameter SIZE = 1 << WIDTH;
typedef logic [WIDTH - 1 : 0] Word;
typedef logic [255:0] BigItem;

module shuffler (
    input logic clk,
    input logic rst_n,
    output logic odd,
    output logic [(1 << WIDTH) - 1 : 0][WIDTH - 1 : 0] shuffle
);
    Word ctr;
    assign odd = ctr[0];

    always_ff @(posedge clk) begin
        if (!rst_n) begin
            ctr <= 0;
        end
        else begin
            ctr <= ctr + 1;
        end
    end

    for (genvar i = 0; i < SIZE; i++) always_comb begin
        shuffle[i] = Word'(i) - ctr;
    end

    for (genvar i = 0; i < SIZE; i++) begin
        assert property (@(posedge clk) shuffle[ctr + Word'(i)] == i);
    end
endmodule

interface big_port();
    BigItem big;

    function automatic BigItem get_big();
        return big;
    endfunction

    modport reader(import get_big);
endinterface

module foo (
    input clk,
    input rst_n,
    big_port.reader big
);
    logic odd;
    Word[SIZE - 1 : 0] shuffle;
    shuffler fifo (
        .clk,
        .rst_n,
        .odd,
        .shuffle
    );

    BigItem bigs[SIZE];
    for (genvar i = 0; i < SIZE; i++) always_comb begin
        bigs[i] = odd ? big.get_big() : 0;
    end
endmodule

module testbench;
    logic clk = 0;
    always #5 clk = !clk;

    logic rst_n = 0;

    big_port big();

    foo foo (
        .clk,
        .rst_n,
        .big
    );

    initial begin
        #10;
        $finish;
    end
endmodule
```

Command line output:

```
make: Entering directory '/obj_dir'
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_timing.o /usr/local/share/verilator/include/verilated_timing.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_threads.o /usr/local/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtestbench.cpp Vtestbench___024root__DepSet_hfc24d085__0.cpp Vtestbench___024root__DepSet_hed41eec4__0.cpp Vtestbench__main.cpp Vtestbench___024root__Slow.cpp Vtestbench___024root__DepSet_hfc24d085__0__Slow.cpp Vtestbench___024root__DepSet_hed41eec4__0__Slow.cpp Vtestbench___024unit__Slow.cpp Vtestbench___024unit__DepSet_hf87c9ffd__0__Slow.cpp Vtestbench__Syms.cpp > Vtestbench__ALL.cpp
ccache g++ -Os  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -fcoroutines -c -o Vtestbench__ALL.o Vtestbench__ALL.cpp
In file included from Vtestbench__ALL.cpp:5:
Vtestbench___024root__DepSet_hed41eec4__0.cpp: In function 'void Vtestbench___024root___eval(Vtestbench___024root*)':
Vtestbench___024root__DepSet_hed41eec4__0.cpp:122:5: error: '__Vilp' was not declared in this scope
  122 |     __Vilp = 0U;
      |     ^~~~~~
make: *** [/usr/local/share/verilator/include/verilated.mk:266: Vtestbench__ALL.o] Error 1
make: Leaving directory '/obj_dir'
%Error: make -C obj_dir -f Vtestbench.mk -j 1 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/share/verilator/bin/verilator_bin --timing --binary --assert --top-module testbench testbench.sv
```

Sorry for the long example; I had trouble making this smaller while still triggering the error.

justinyaodu, 2024-03-03:
@wsnyder I'm not sure if I have the right permissions to reopen a closed issue; would you mind doing that if you feel it's appropriate? Thanks :sweat_smile:

wsnyder, 2024-03-03:
Thanks for the example, fixed.

Before fix, with --debug gave a clean assert, and workaround was -fno-reloop.

closed 2024-03-03 by wsnyder

justinyaodu, 2024-03-03:
Wow, that was super fast - thank you so much! :grinning: