verilator/verilator #4830: Improve unique case support

yTakatsukasa, 2024-01-15:
IEEE 1800-2017 12.5.3 requires simulators to issue a violation report
1.  if more than two case items matched (both unique and unique0)
2.  if no case item matches (only unique case)

These checks are not implemented yet.

I think `%Warning-CASEINCOMPLETE` is not necessary for `unique case` even for non-enum types such as logic.
I understand `unique case` is successor of `full_case unique_case` directive for synthesis in the past.
Designer wants uncovered case item to be treated as a dont_care for optimization.

I'd like to find the common interpretation of the spec before making a PR.

yTakatsukasa, 2024-01-15:
https://github.com/verilator/verilator/issues/4357#issuecomment-1632899995 looks different opinion from me.

wsnyder, 2024-01-17:
I think some of what you suggest might already be found at runtime - first step is to check the tests or improve them to create these cases.

yTakatsukasa, 2024-02-18:
After walking around verilator code base, I'd like to propose the following changes.
1. Add "no match" check for unique/priority case when `-assert` is not set. The check has little overhead, just adding `default` term during verilation. ("multiple much" check is expensive, so it is out of scope for now).
2. "no match" check just emits warning and continues the simulation instead of calling $finish/$stop. This is default behavior in Icarus and at least one commercial simulator.

Any feedback is appreciated.

wsnyder, 2024-02-18:
There is a performance cost to adding this by default, as. the $stop will disable some math optimizations. For designs which are known good and e.g. providing a software model the asserts are not useful.

It seems inconsistent we would detect none, but not overlaps by default.

Maybe add --assert-case that does both and --assert sets it?

yTakatsukasa, 2024-02-18:
I mean just warning message, no $stop nor $finish.
I'm not sure if $display is still burden for math optimization.

How about
- add `--assert-case` that enables only case check as you proposed
- also add option to customize action when assertion fails (`--stop-on-assertion-failure` ?)
   - `$stop` as now
   - keep simulation as other simulators do

wsnyder, 2024-02-18:
display also disables some optimizations as is not pure.

Ok, --assert-case

Maybe --no-stop-fail?  Would suggest this be a separate pre or post pull.

yTakatsukasa, 2024-02-18:
Sure. I will send 2 PRs, `--assert-case` and `--no-stop-fail`.

yTakatsukasa, 2024-02-23:
I'm good with recent PRs. Thanks for comments!

closed 2024-02-23 by yTakatsukasa