verilator/verilator #4832: Wrong runtime output on small combinational circuit

flaviens, 2024-01-15:
Hi there!

Thank you very much for your contributions with Verilator!
I stumbled into a simple circuit that produces a wrong runtime value, which I minimized and packaged it into a [dedicated repo](https://github.com/flaviens/verilator-b7).

In short, the design looks like
```
module top(in_data, out_data);
  bit celloutsig_10z;
  bit [3:0] celloutsig_1z;
  bit [5:0] celloutsig_2z;
  bit celloutsig_5z;
  bit celloutsig_6z;
  bit celloutsig_7z;
  input [95:0] in_data;
  bit [95:0] in_data;
  output [95:0] out_data;
  bit [95:0] out_data;
  assign celloutsig_5z = ~(celloutsig_1z[3] & in_data[67]);
  assign celloutsig_6z = ~(in_data[86] & celloutsig_5z);
  assign celloutsig_7z = ~(celloutsig_6z);
  assign celloutsig_10z = in_data[3] & ~(celloutsig_7z);
  assign celloutsig_2z = ~ { in_data[43:42], celloutsig_1z };
  assign celloutsig_1z = in_data[18:15] ^ { in_data[42:40], 1'b1 };
  assign out_data[0] = celloutsig_10z;
endmodule
```

I tested a slightly older version of Verilator and confirm that it still exists in the last version `Verilator 5.021 devel rev v5.020-31-g54922c036`.
Notably, the bug also persists if tracing is enabled.

Please let me know if you need any more info.

Thanks!
Flavien

yTakatsukasa, 2024-01-15:
Thanks for the brief example.
Disabling bit op tree using `-fno-const-bit-op-tree` option changes the result.

icarus crashes on my side, so I cannot compare the result with it. (I'm using `Icarus Verilog runtime version 12.0 (stable)`)

Please let me know whether the result with `-fno-const-bit-op-tree`  is correct or not.

yTakatsukasa, 2024-01-16:
Thanks for the info.
I've even simplified your case to t_regress format.
I need to recover my memory. Hopefully I can try this weekend.

PR #4847 (2024-01-21): Fix #4832 of incorrect bit-op-tree optimization
Fixes #4832.

`a & b` in `!(a & b) & c` has to be untouched, but was not.

As usual, push the test to reproduce #4832, then will push the fix for it later.

I'm a bit surprised at how long such a simple bug has been there.

ext tests pass.
commit 50df902b (2024-01-21): Fix incorrect bit-op-tree NOT optimization (#4832) (#4847)

* Fix #4832. Bit-op-tree opt. should not touch a subtree under NOT in AND/OR tree.
closed 2024-01-21 by wsnyder