verilator/verilator #4833: Segmentation fault in eval_initial

flaviens, 2024-01-16:
Hi there!

I stumbled into a segfault in Verilator (with `Verilator 5.021 devel rev v5.020-31-g54922c036`).

```
0x00005555555690cd in Vtop_top___eval_initial__TOP__top (vlSelf=0x240) at Vtop_top__DepSet_h6315917a__0__Slow.cpp:10                                                                                                                                                                      
10          Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;                                                                                                                                                                                                        
(gdb) bt                                                                                                                                                                                                                                                                                  
#0  0x00005555555690cd in Vtop_top___eval_initial__TOP__top (vlSelf=0x240) at Vtop_top__DepSet_h6315917a__0__Slow.cpp:10                                                                                                                                                                  
#1  0x0000555555567533 in Vtop___024root___eval_initial (vlSelf=0x5555555a37c0) at Vtop___024root__DepSet_h84412442__0__Slow.cpp:16                                                                                                                                                       
#2  0x00005555555671d9 in Vtop::eval_step (this=0x5555555a3600) at Vtop.cpp:57                                                                                                                                                                                                            
#3  0x0000555555558da7 in Vtop::eval (this=0x5555555a3600) at ./Vtop.h:58                                                                                                                                                                                                                 
#4  run_test (my_module=0x5555555a3600, simlen=205, trace_filename=...) at /path/to/tb_base.cc:206                                                                                                                                                          
#5  0x000055555555893a in main (argc=<optimized out>, argv=<optimized out>, env=<optimized out>) at /path/to/tb_base.cc:251                                                                                                                                 
```

[Here is a Valgrind log.](https://github.com/verilator/verilator/files/13950315/valgrind.log)

[Here is an archive with the sources.](https://github.com/verilator/verilator/files/13950508/segfault_eval_init.zip)

My apologies for the complex test case.
It may related to #4759.

Thank you!
Flavien

wsnyder, 2024-01-16:
The code provided doesn't compile, and when I tried a few rounds of fixes I still had problems. Can you please repost and make sure it's as simplified as possible? Thanks.

flaviens, 2024-01-17:
Thank you for your response! I created a repository for #4835 that has the same sources but that gives another segfault this time. I close this issue to the profit of the other one.
Thanks!
Flavien

closed 2024-01-17 by flaviens