verilator/verilator #4834: Runtime error after a xor of concatenations after a latch

flaviens, 2024-01-16:
Hi there!

I stumbled into a runtime error with Icarus, where the output of a latch is wrong after consumed by a xor.
Verilator produces the value that one would expect. To the  best of my knowledge, there is no structural race condition in this circuit.

[Here is a repository to reproduce the issue](https://github.com/flaviens/icarus-b1).
I'm using `Icarus Verilog version 13.0 (devel) (s20221226-331-g77d7f0b8f)`.

Please let me know if you need any more information.

wsnyder, 2024-01-16:
Based on your description this seems like an Icarus bug.  if so please refile in Icarus's bug tracker, if it is against Verilator please reword appropriately.

flaviens, 2024-01-16:
Oh so sorry I posted it in the wrong repository :facepalm:  my apologies for bothering you :disappointed:

closed 2024-01-16 by flaviens