verilator/verilator #4835: Segmentation fault in traceInit

flaviens, 2024-01-17:
Hi there!

I stumbled into a segfault in Verilator (with `Verilator 5.021 devel rev v5.020-31-g54922c036`).

```bash
Thread 1 "Vtop" received signal SIGSEGV, Segmentation fault.
trace_init (voidSelf=0x5555555b1740, tracep=0x5555555d7958, code=1) at Vtop.cpp:117
117         if (!vlSymsp->_vm_contextp__->calcUnusedSigs()) {
(gdb) bt
#0  trace_init (voidSelf=0x5555555b1740, tracep=0x5555555d7958, code=1) at Vtop.cpp:117
#1  0x00005555555689c7 in VerilatedTrace<VerilatedVcd, VerilatedVcdBuffer>::traceInit (this=0x5555555d7958) at /path/to/verilator/include/verilated_trace.h:343
#2  0x000055555556aaf4 in VerilatedVcd::open (this=0x5555555d7958, filename=<optimized out>) at /path/to/verilator/include/verilated_vcd_c.cpp:123
#3  0x000055555555a0a3 in run_test (my_module=0x5555555b15c0, simlen=205, trace_filename=...) at /usr/include/c++/10/bits/basic_string.h:2312
#4  0x0000555555559bfa in main (argc=<optimized out>, argv=<optimized out>, env=<optimized out>) at /path/to/some/tb_base.cc:251
```

[Here is a dedicated repository.](https://github.com/flaviens/verilator-b8)

My apologies for the complex test case.
It exposes another segfault than #4833 with the same source but another context.

Thank you!
Flavien

wsnyder, 2024-01-17:
When I compile it it always works fine.  I'll need an example I can compile that shows the problem to debug it.

From the backtrace you provided it looks like you are calling the model before it's properly been created, and this was using a different .cc file than you provided so I can't see what you are doing.

wsnyder, 2024-01-17:
Also make sure to use the debug flags recommended in the documentation: https://veripool.org/guide/latest/simulating.html#runtime-debugging

robotman2412, 2024-01-29:
I've encountered a segfault like this one in one of my projects (https://github.com/robotman2412/sv-float).
The model in question exists at `test/special` and can be run using `make`.
This issue exists on 5.020 but not on 5.008, with the same code.

wsnyder, 2024-01-30:
For me, the core dump appears in your code due to not checking fopen for a non-existent file.  (I suspect this is a mistake in what you provided, and not the real issue, so please repair and send again)

```
#0  __vfscanf_internal (s=0x0, format=0x555555557079 "0x%08x 0x%08x 0x%08x 0x%08x\n", argptr=argptr@entry=0x7fffffffd1b0, mode_flags=mode_flags@entry=2) at ./std\
io-common/vfscanf-internal.c:345
#1  0x00007ffff7cff21d in __isoc99_fscanf (stream=<optimized out>, format=<optimized out>) at ./stdio-common/isoc99_fscanf.c:30
#2  0x000055555555568e in run_sim (a=nan(0x400000), b=nan(0x400000)) at test.c:93
#3  0x00005555555557b5 in sim_mul (a=nan(0x400000), b=nan(0x400000)) at test.c:109

```

robotman2412, 2024-01-30:
This isn't the Verilator model and the Verilator model crashing is what causes this.
I should have noted that.
However, the Verilator model bench.cpp (compiled to obj_dir/sim) exhibits the behaviour I described.

wsnyder, 2024-01-30:
Ok, but the other problem I then get is reading data bad over the stack because of this: 

   trace->open(argv[2]);

but the makefile doesn't pass any arguments so that is really past the stack.   Please make your test standalone and simplify away all argument, other files, and similar dependencies.

robotman2412, 2024-01-30:
That's because the other program being compiled invokes the model, which I also could have said earlier.
But it your concern about the stack is simply not true, bench.cpp checks the value of argc before doing anything else and exits if there is an incorrect amount of arguments.
Believe me that I tested on both 5.020 and 5.008 and it is working normally in the older version of Verilator.

wsnyder, 2024-01-30:
I'm sure it does change between versions, but to debug I need the address sanitizer to be clean as that is how I debug these sort of problems.   (The argv is reported by it - there is no an argc check).  So, please update to post something as simplified as possible.

pberuto, 2024-02-27:
Hello,
I think I've found the same problem.

Thread 1 "mcu_sim" received signal SIGSEGV, Segmentation fault.
0x000055555555e04d in trace_init(void*, VerilatedFst*, unsigned int) ()
(gdb) backtrace
#0  0x000055555555e04d in trace_init(void*, VerilatedFst*, unsigned int) ()
#1  0x00005555555eb210 in VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() ()
#2  0x00005555555eb577 in VerilatedFst::open(char const*) ()
#3  0x00005555555bfcb0 in simulationT::simulationT(int, char* const*) ()
#4  0x00005555555bdde2 in main ()

Unfortunately, if I add the -ggdb flags, this problem doesn't happen.
Also, I've noticed that changing CFLAGS from -O3 to -O2 makes a difference, depending on the actual verilog model.
I mean, if you change the verilog code, the problem may show up or not, and changing the CFLAGS sometimes makes the difference between SIGSEGV or not. But the backtrace is always the same.

Additionally, changing the g++ version also makes a difference (sometimes). Again, by "sometimes" I mean that is the combination of g++ version, CFLAGS and the actual verilog code that makes the difference, but in a totally random manner.

In my humble opinion, when the code is sensitive to these kind of changes, it is likely we're facing a race of sort.
It seems to me that the g++ version/CFLAGS is merely affecting the timing of code execution.

Additional info: If I compile the very same code on the very same virtual machine (i mean, a real clone) but on a PC with a different clock speed, again, the problem may show/not show. Again, this seems to me a symptom of time-sensitive race conditions.

Please, let me know if I can help more.

gezalore, 2024-02-27:
What you describe indicates a memory safety issue, could you run the simulation with valgrind please and report any invalid accesses it might flag up.

pberuto, 2024-02-27:
> What you describe indicates a memory safety issue, could you run the simulation with valgrind please and report any invalid accesses it might flag up.

I was able to reproduce the problem running Valgrind. Please see the following log.
Unfortunately, I could not reproduce this with debug symbols enabled (-ggdb CFLAGS). I will try more.
It looks like a NULL pointer access to me. Although, it is not systematic.

==629928== Memcheck, a memory error detector
==629928== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==629928== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==629928== Command: ./mcu_sim
==629928== 
==629928== Use of uninitialised value of size 8
==629928==    at 0x11204D: trace_init(void*, VerilatedFst*, unsigned int) (in mcu_sim)
==629928==    by 0x19F20F: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (in mcu_sim)
==629928==    by 0x19F576: VerilatedFst::open(char const*) (in mcu_sim)
==629928==    by 0x173CAF: simulationT::simulationT(int, char* const*) (in mcu_sim)
==629928==    by 0x171DE1: main (in mcu_sim)
==629928== 
==629928== Invalid read of size 8
==629928==    at 0x11204D: trace_init(void*, VerilatedFst*, unsigned int) (in mcu_sim)
==629928==    by 0x19F20F: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (in mcu_sim)
==629928==    by 0x19F576: VerilatedFst::open(char const*) (in mcu_sim)
==629928==    by 0x173CAF: simulationT::simulationT(int, char* const*) (in mcu_sim)
==629928==    by 0x171DE1: main (in mcu_sim)
==629928==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==629928== 
==629928== 
==629928== Process terminating with default action of signal 11 (SIGSEGV)
==629928==  Access not within mapped region at address 0x0
==629928==    at 0x11204D: trace_init(void*, VerilatedFst*, unsigned int) (in mcu_sim)
==629928==    by 0x19F20F: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (in mcu_sim)
==629928==    by 0x19F576: VerilatedFst::open(char const*) (in mcu_sim)
==629928==    by 0x173CAF: simulationT::simulationT(int, char* const*) (in mcu_sim)
==629928==    by 0x171DE1: main (in mcu_sim)
==629928==  If you believe this happened as a result of a stack
==629928==  overflow in your program's main thread (unlikely but
==629928==  possible), you can try to increase the size of the
==629928==  main thread stack using the --main-stacksize= flag.
==629928==  The main thread stack size used in this run was 8388608.
==629928== 
==629928== HEAP SUMMARY:
==629928==     in use at exit: 138,608,076 bytes in 117 blocks
==629928==   total heap usage: 142 allocs, 25 frees, 138,697,347 bytes allocated
==629928== 
==629928== LEAK SUMMARY:
==629928==    definitely lost: 0 bytes in 0 blocks
==629928==    indirectly lost: 0 bytes in 0 blocks
==629928==      possibly lost: 2,432 bytes in 8 blocks
==629928==    still reachable: 138,605,644 bytes in 109 blocks
==629928==         suppressed: 0 bytes in 0 blocks
==629928== Rerun with --leak-check=full to see details of leaked memory
==629928== 
==629928== Use --track-origins=yes to see where uninitialised values come from
==629928== For lists of detected and suppressed errors, rerun with: -s
==629928== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)

pberuto, 2024-02-27:
Actually, I got lucky and I could get the same problem with debug symbols enabled.
**See below:**

==630148== Memcheck, a memory error detector
==630148== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==630148== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==630148== Command: ./mcu_sim
==630148== 
==630148== Use of uninitialised value of size 8
==630148==    at 0x153AFA: Vmcu___024root__trace_init_sub__TOP__0(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:12)
==630148==    by 0x1B3B42: Vmcu___024root__trace_init_top(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:6480)
==630148==    by 0x110E74: trace_init(void*, VerilatedFst*, unsigned int) (Vmcu.cpp:148)
==630148==    by 0x12E8E2: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (verilated_trace_imp.h:331)
==630148==    by 0x12EC24: VerilatedFst::open(char const*) (verilated_fst_c.cpp:87)
==630148==    by 0x112E5F: simulationT::simulationT(int, char* const*) (sim_mcu.cc:72)
==630148==    by 0x110FAC: main (sim_mcu.cc:311)
==630148== 
==630148== Invalid read of size 4
==630148==    at 0x153AFA: Vmcu___024root__trace_init_sub__TOP__0(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:12)
==630148==    by 0x1B3B42: Vmcu___024root__trace_init_top(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:6480)
==630148==    by 0x110E74: trace_init(void*, VerilatedFst*, unsigned int) (Vmcu.cpp:148)
==630148==    by 0x12E8E2: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (verilated_trace_imp.h:331)
==630148==    by 0x12EC24: VerilatedFst::open(char const*) (verilated_fst_c.cpp:87)
==630148==    by 0x112E5F: simulationT::simulationT(int, char* const*) (sim_mcu.cc:72)
==630148==    by 0x110FAC: main (sim_mcu.cc:311)
==630148==  Address 0x1c is not stack'd, malloc'd or (recently) free'd
==630148== 
==630148== 
==630148== Process terminating with default action of signal 11 (SIGSEGV)
==630148==  Access not within mapped region at address 0x1C
==630148==    at 0x153AFA: Vmcu___024root__trace_init_sub__TOP__0(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:12)
==630148==    by 0x1B3B42: Vmcu___024root__trace_init_top(Vmcu___024root*, VerilatedFst*) (Vmcu__Trace__0__Slow.cpp:6480)
==630148==    by 0x110E74: trace_init(void*, VerilatedFst*, unsigned int) (Vmcu.cpp:148)
==630148==    by 0x12E8E2: VerilatedTrace<VerilatedFst, VerilatedFstBuffer>::traceInit() (verilated_trace_imp.h:331)
==630148==    by 0x12EC24: VerilatedFst::open(char const*) (verilated_fst_c.cpp:87)
==630148==    by 0x112E5F: simulationT::simulationT(int, char* const*) (sim_mcu.cc:72)
==630148==    by 0x110FAC: main (sim_mcu.cc:311)
==630148==  If you believe this happened as a result of a stack
==630148==  overflow in your program's main thread (unlikely but
==630148==  possible), you can try to increase the size of the
==630148==  main thread stack using the --main-stacksize= flag.
==630148==  The main thread stack size used in this run was 8388608.
==630148== 
==630148== HEAP SUMMARY:
==630148==     in use at exit: 138,608,116 bytes in 117 blocks
==630148==   total heap usage: 144 allocs, 27 frees, 138,701,523 bytes allocated
==630148== 
==630148== LEAK SUMMARY:
==630148==    definitely lost: 0 bytes in 0 blocks
==630148==    indirectly lost: 0 bytes in 0 blocks
==630148==      possibly lost: 2,432 bytes in 8 blocks
==630148==    still reachable: 138,605,684 bytes in 109 blocks
==630148==         suppressed: 0 bytes in 0 blocks
==630148== Rerun with --leak-check=full to see details of leaked memory
==630148== 
==630148== Use --track-origins=yes to see where uninitialised values come from
==630148== For lists of detected and suppressed errors, rerun with: -s
==630148== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)

Segmentation fault

**And this is the content of the Vmcu__Trace__0__Slow.cpp:**

7  VL_ATTR_COLD void Vmcu___024root__trace_init_sub__TOP__0(Vmcu___024root* vlSelf, VerilatedFst* tracep) {
8      if (false && vlSelf) {}  // Prevent unused
9      Vmcu__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
10     VL_DEBUG_IF(VL_DBG_MSGF("+    Vmcu___024root__trace_init_sub__TOP__0\n"); );
11 // Init
**12  const int c = vlSymsp->__Vm_baseCode;**
13 // Body

So it seems vlSymsp is NULL?

wsnyder, 2024-03-03:
>by 0x12EC24: VerilatedFst::open(char const*) (verilated_fst_c.cpp:87)
==630148== by 0x112E5F: simulationT::simulationT(int, char* const*) (sim_mcu.cc:72)

It looks like you might be calling tracing in a constructor before the model has completed constructing.  Can you share your .cpp code?

pberuto, 2024-03-09:
> > by 0x12EC24: VerilatedFst::open(char const*) (verilated_fst_c.cpp:87)
> > ==630148== by 0x112E5F: simulationT::simulationT(int, char* const*) (sim_mcu.cc:72)
> 
> It looks like you might be calling tracing in a constructor before the model has completed constructing. Can you share your .cpp code?

Sure,
this is the constructor of my class, which is instanced in the main function. I added more comments to make it more clear.
Please, note that the problem is NOT systematic, as I reported above. Most of the time, it works.

simulationT::simulationT(int argc, char* const argv[]) : _log("main")
{
   _parseCmdLine(argc, argv);
   
   std::string dir, exe;
   pem::splitPathName(argv[0], dir, exe);
   
   // change directory before instancing the verilated model
   int ret = chdir(dir.c_str());
   
   // logger configuarion
   pem::loggerT::setDefLogLev(pem::log::info);
   pem::loggerT::setDefLogDev(
      pem::logDevT::handleT(new pem::fileLogDevT("sim.log"))
   );
   
   // parse command line options for verilator
   Verilated::commandArgs(argc, (const char **) &argv[0]);

   VerilatedContext *ctx = new VerilatedContext;
   _ctx.reset(ctx);
   
   **// instance the verilated model**
   _mcu = new Vmcu(ctx);

   // instance non-verilated models of the peripherals
   _uart = simUART::create(_mcu, clock_speed_hz, uart_baud_hz);
   _phy = simPHY::create(_mcu, clock_speed_hz, phy0_link_speed_hz);
   _boot = simBOOT::create(_mcu, _bootFileNames);

#ifdef __SIM_TRACE__
   Verilated::traceEverOn(true);
   
   _trace = new VerilatedFstC;
   _mcu->trace(_trace, 99);
   _trace->open("wave.fst");     // <-- this is what **sometimes** gives segfault.
#endif
}

wsnyder, 2024-03-09:
Try adding an eval() call before you make the trace.  That's what we do in all other places - the idea is you might start trace half way through an execution if it is long.  I don't know why it would be indeterminate, as you are making it in program order and not e.g. as static objects where it is up to the compiler.

wsnyder, 2024-04-13:
There may still be a problem here - but we still don't have a obvious reproducer, if there is one that is otherwise sanitizer-clean please reopen.

closed 2024-04-13 by wsnyder

mbekmyrz, 2024-04-26:
Having similar problems with random segfaults as mentioned above.