verilator/verilator #4836: Wrong runtime value when tracing is disabled

flaviens, 2024-01-17:
Hi there!

I stumbled into a runtime value error in Verilator (`Verilator 5.021 devel rev v5.020-31-g54922c036`), which is corrected when tracing, but not corrected by any individual `-fno-` flag.

[Here is  a dedicated repository](https://github.com/flaviens/verilator-b9) with a small source file for reproducing the bug.

Please let me know if you need any more information!
Thank you,
Flavien

wsnyder, 2024-01-18:
Run with -Wall and there are undriven nets which need to be fixed, and Verilator will use indeterminate values.  When I assign those to zero the results match.

closed 2024-01-18 by wsnyder

flaviens, 2024-01-18:
Hi @wsnyder,

Thank you for your answer!
In fact, the undriven nets come from simplifying stateful cells that did not provide a reset value (conceptually providing `x`).  I imagine that one decision would be to tolerate these mismatches between optimizations for the sake of simplicity/speed.

Thank you!
Flavien