verilator/verilator #4837: Wrong runtime output corrected exclusively by -fno-expand -fno-const-bit-op-tree

flaviens, 2024-01-17:
Hi there!

Thank you very much for your contributions with Verilator!
I stumbled into a simple circuit that produces a wrong runtime value, which I minimized and packaged it into a [dedicated repo](https://github.com/flaviens/verilator-b10).

I tested it on `Verilator 5.021 devel rev v5.020-31-g54922c036`.
Notably, the bug also persists if tracing is enabled.

The flags that correct this bug are `-fno-expand` `-fno-const-bit-op-tree`.

Please let me know if you need any more info.

Thanks!
Flavien

yTakatsukasa, 2024-01-17:
Let me confirm one thing: `Actual output` and `Expected output` looks opposite.

I mean

This should be `Actual output`
https://github.com/flaviens/verilator-b10/blob/c3d62860bbe4b73aef1ef58343a8b8de18f1c7ec/Readme.md?plain=1#L8-L14

and below should be `Expected output`
https://github.com/flaviens/verilator-b10/blob/c3d62860bbe4b73aef1ef58343a8b8de18f1c7ec/Readme.md?plain=1#L16-L22

flaviens, 2024-01-18:
Thank you for your response! Yes that's correct sorry for the little mistake ^^' you git it correctly. Now corrected in the readme.

wsnyder, 2024-01-21:
Thanks @yTakatsukasa !

closed 2024-01-21 by wsnyder