verilator/verilator #4839: Problem compiling design under Verilator 5.019

jordankrim, 2024-01-19:
My test is ending without any errors running under Verilator 4.226.  I attempted to use valgrind, but cannot interpret the results.  I also used GDB, but it just runs to the end and then there is no stack and I have no idea where the issue is so sneaking up on it is almost impossible.  

So, I decided to move to a newer version of Verilator (5.019 devel rev v5.018-44-gb8417abee) to see if it has the same issue.  Unfortunately, I did a make clean before starting so that should have cleared out all the old files but the compile ends with the following fatal error for a missing file:

<command-line>: fatal error: VTestHarness__pch.h.fast: No such file or directory
compilation terminated.
make[1]: *** [/home/jkrim/verilator/include/verilated.mk:272: VTestHarness.o] Error 1
make[1]: Leaving directory '/mnt/data/projects/project/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig'
make: *** [Makefile:273: /mnt/data/projects/project/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/simulator-chipyard-CEPVerilatorRocketConfig] Error 2

The lines in Makefile (272-273 are not the commented ones below) are these:

#########################################################################################
# invoke make to make verilator sim rules
#########################################################################################
$(sim): $(model_mk) $(dramsim_lib)
	$(MAKE) VM_PARALLEL_BUILDS=1 -C $(model_dir) -f V$(VLOG_MODEL).mk

wsnyder, 2024-01-19:
Please see if this relates to #4785 with how you are passing includes.

jordankrim, 2024-01-19:
I don't think this is the same issue.   I searched the directory for the file that it is mising and I don't see it at all.  Here is the list of files I see:

./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__verFiles.dat
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__44__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__17.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__22__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__36__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__4__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__68.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__48.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__47.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__66.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__4__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__19__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__8.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__46.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__58.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__11__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_h30a6511e__2.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__4.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__77.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__7.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__72.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024unit__DepSet_h66f95809__0__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__12__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__0__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__18.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Dpi.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__11__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_h30a6511e__1__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_h30a6511e__0.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__37__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Syms__2.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__34__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__20__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness__Trace__9__Slow.cpp
[... cut ...]
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__21__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__10__Slow.cpp
./generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness___024root__DepSet_hcfd557fa__21.cpp

So, is there some compiler option or such that I am missing with this "newer" version of Verilator because 4.226 compiles fine for me with no problem?

wsnyder, 2024-01-20:
Also please use the latest master git version, to make sure it's not something already fixed.

jordankrim, 2024-01-20:
Downloaded and built from the master (which appears to be 5.021).  Same problem.  I'm a bit lost since I didn't write these Makefiles that are creating the Verilator executable, but here is a line with the offending VTestHarness__pch.fast.  No idea where to go from here.

g++ -Os  -I.  -MMD -I/home/jkrim/verilator_5.021/include -I/home/jkrim/verilator_5.021/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O3 -std=c++14 -I/mnt/data/projects/ipvault/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/riscv-tools/include -I/mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/tools/DRAMSim2 -I/mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig    -D__STDC_FORMAT_MACROS -DTEST_HARNESS=VTestHarness -DVERILATOR -include /mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig.plusArgs -include /mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/verilator.h -include /mnt/data/projects/ipvault/jkrim/CEP-Setup/cep_v4_4_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness.h   -fcoroutines -include **VTestHarness__pch.h.fast** -c -o VTestHarness.o VTestHarness.cpp

wsnyder, 2024-01-20:
There should be another line above that created VTestHarness__pch.h.fast.

Please attach your whole logfile, and again still asking your OS Distro.

jordankrim, 2024-01-21:
My distro is Red Hat Enterprise Linux release 8.9 (Ootpa).  Output is attached.
[compile_5.012.log](https://github.com/verilator/verilator/files/14000046/compile_5.012.log)

wsnyder, 2024-01-21:
>g++ -Os  -I.  -MMD -I/home/jkrim/verilator_5.021/include ...-x c++-header VTestHarness__pch.h -o VTestHarness__pch.h.fast.gch

This should have created .fast.gch, this is the same issue described in #4785.  Try changing include/Makefile.in 

   CFG_CXXFLAGS_PCH_I=-include-pch
   CFG_GCH_IF_CLANG=.gch

As described in #4785 given RHEL 8 distro will be end of life in a few months we're unlikely to fix this.

jordankrim, 2024-01-22:
I don't see any file called inclue/Makefile.in
I see this code in configure.ac:
# Compiler precompiled header options (assumes either gcc or clang++)
AC_MSG_CHECKING([for $CXX precompile header include option])
if $CXX --help | grep include-pch >/dev/null 2>/dev/null ; then
   # clang
   CFG_CXXFLAGS_PCH_I=-include-pch
   CFG_GCH_IF_CLANG=.gch
else
   # GCC
   CFG_CXXFLAGS_PCH_I=-include
   CFG_GCH_IF_CLANG=
fi

So I changed the last line in this to CFG_GCH_IF_CLANG=.gch
I ran autoconf, ./configure, then make

I am getting thousands of errors similar to this line:
./V3PchAstMT.h.gch:16732:33: error: stray '\20' in program
16732 |       <U+0000><U+0000><U+0000><U+0000><U+0000><98>o<U+0002><U+0010><U+0000><U+0000><U+0000><U+007F><U+0000><U+0002><U+0000><U+0000><U+0000><U+0000><U+0000><U+0601><d6>       **<U+0010>**<U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><a0><ed> <U+0000><U+0010><U+0000><U+0000><U+0000><U+0003><U+0000><U+0000><U+0000><U+0001><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0000><U+0010>V<90>
      |                                                                                                                                                                               ^~~~~~~~

wsnyder, 2024-01-23:
This is an old version of g++ so you'll need to figure out the magic to make it happy, or just use "clang", or use a newer redhat or Ubuntu release (which you'll need to in a few months anyways).

wsnyder, 2024-01-23:
P.S. to use clang

```
CXX=clang++ ./configure
make
```

and rebuild per the normal Verilator instructions.

jordankrim, 2024-01-23:
This seems to have resolved the compile issue, but then doing a make test fails:
clang++    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o verilated_threads.o Vt_a1_first_cc__ALL.a    -pthread -lpthread   -o Vt_a1_first_cc
verilated.o: In function `std::__atomic_base<unsigned long>::is_lock_free() const':
verilated.cpp:(.text._ZNKSt13__atomic_baseImE12is_lock_freeEv[_ZNKSt13__atomic_baseImE12is_lock_freeEv]+0x19): undefined reference to `__atomic_is_lock_free'
verilated_threads.o: In function `std::__atomic_base<unsigned int>::is_lock_free() const':
verilated_threads.cpp:(.text._ZNKSt13__atomic_baseIjE12is_lock_freeEv[_ZNKSt13__atomic_baseIjE12is_lock_freeEv]+0x19): undefined reference to `__atomic_is_lock_free'
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [Vt_a1_first_cc.mk:65: Vt_a1_first_cc] Error 1
rm Vt_a1_first_cc__ALL.verilator_deplist.tmp
driver: Leaving directory '/home/jkrim/verilator_5.021/test_regress/obj_vlt/t_a1_first_cc'
%Warning: vlt/t_a1_first_cc: Exec of make failed: clang++  -I.  -MMD -I/home/jkrim/verilator_5.021/test_regress/../include -I/home/jkrim/verilator_5.021/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -DVL_LOCK_SPINS=10000 -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp

vlt/t_a1_first_cc: %Error: Exec of make failed: clang++  -I.  -MMD -I/home/jkrim/verilator_5.021/test_regress/../include -I/home/jkrim/verilator_5.021/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -DVL_LOCK_SPINS=10000 -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
vlt/t_a1_first_cc: FAILED: Exec of make failed: clang++  -I.  -MMD -I/home/jkrim/verilator_5.021/test_regress/../include -I/home/jkrim/verilator_5.021/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -DVL_LOCK_SPINS=10000 -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
==SUMMARY: Passed 0  Failed 1  Time 0:09
==SUMMARY: Passed 0  Failed 1  Time 0:09

======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of make failed: clang++  -I.  -MMD -I/home/jkrim/verilator_5.021/test_regress/../include -I/home/jkrim/verilator_5.021/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFIX_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -DVL_LOCK_SPINS=10000 -O0 -c -o Vt_a1_first_cc__main.o ../../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:09
make: *** [Makefile:170: smoke-test] Error 10

wsnyder, 2024-01-24:
Googling the error... see: https://www.reddit.com/r/C_Programming/comments/qnlw5k/undefined_reference_to_atomic_is_lock_free/
so try `-LDFLAGS -latomic`

jordankrim, 2024-01-24:
Is that -LDFLAGS -latomic supposed to be added to the compile of Verilator options (and if so where exactly)?

wsnyder, 2024-01-24:
Add it on a single test you are trying to run, or in configure.ac try this

```
diff --git a/configure.ac b/configure.ac
index 1aa353ed7..9315730f4 100644
--- a/configure.ac
+++ b/configure.ac
@@ -474,6 +474,7 @@ AC_SUBST(CFG_LDFLAGS_VERILATED)
 # pthreads is harmless otherwise.
 CFG_LIBS="$LIBS $CFG_LIBS"
 _MY_LDLIBS_CHECK_OPT(CFG_LIBS, -lpthread)
+_MY_LDLIBS_CHECK_OPT(CFG_LIBS, -latomic)

 # Check libraries for MingW
 _MY_LDLIBS_CHECK_OPT(CFG_LIBS, -lbcrypt)

```

jordankrim, 2024-01-24:
I did a git pull and there is this in the configure.ac:

# pthreads is harmless otherwise.
CFG_LIBS="$LIBS $CFG_LIBS"
_MY_LDLIBS_CHECK_OPT(CFG_LIBS, -lpthread)
_MY_LDLIBS_CHECK_OPT(CFG_LIBS, -latomic)

# Check libraries for MingW
_MY_LDLIBS_CHECK_OPT(CFG_LIBS, -lbcrypt)

The make test still fails.  I did not try to add it to the test.  I need to investigate/understand where the -LDFLAGS -latomic would be added to a test.

jordankrim, 2024-01-26:
Tried something else since no response:

Linking ../../bin/verilator_bin_dbg...
clang++ -static-libgcc -static-libstdc++ -Xlinker -gc-sections -latomic -o ../../bin/verilator_bin_dbg V3Const__gen.o V3Error.o V3FileLine.o V3Graph.o V3GraphAcyc.o V3GraphAlg.o V3GraphPathChecker.o V3GraphTest.o V3Hash.o V3OptionParser.o V3Os.o V3ParseGrammar.o V3ParseImp.o V3ParseLex.o V3PreProc.o V3PreShell.o V3String.o V3ThreadPool.o V3Waiver.o Verilator.o  V3Ast.o V3AstNodes.o V3Broken.o V3Config.o V3EmitCBase.o V3EmitCConstPool.o V3EmitCFunc.o V3EmitCHeaders.o V3EmitCImp.o V3EmitCInlines.o V3EmitCPch.o V3EmitV.o V3File.o V3Global.o V3Hasher.o V3Number.o V3Options.o V3Stats.o V3StatsReport.o  V3Active.o V3ActiveTop.o V3Assert.o V3AssertPre.o V3Begin.o V3Branch.o V3CCtors.o V3CUse.o V3Case.o V3Cast.o V3Class.o V3Clean.o V3Clock.o V3Combine.o V3Common.o V3Coverage.o V3CoverageJoin.o V3Dead.o V3Delayed.o V3Depth.o V3DepthBlock.o V3Descope.o V3Dfg.o V3DfgAstToDfg.o V3DfgDecomposition.o V3DfgDfgToAst.o V3DfgOptimizer.o V3DfgPasses.o V3DfgPeephole.o V3DupFinder.o V3EmitCMain.o V3EmitCMake.o V3EmitCModel.o V3EmitCSyms.o V3EmitMk.o V3EmitXml.o V3Expand.o V3Force.o V3Fork.o V3Gate.o V3HierBlock.o V3Inline.o V3Inst.o V3InstrCount.o V3Interface.o V3Life.o V3LifePost.o V3LinkCells.o V3LinkDot.o V3LinkInc.o V3LinkJump.o V3LinkLValue.o V3LinkLevel.o V3LinkParse.o V3LinkResolve.o V3Localize.o V3MergeCond.o V3Name.o V3Order.o V3Param.o V3Partition.o V3Premit.o V3ProtectLib.o V3Randomize.o V3Reloop.o V3Sched.o V3SchedAcyclic.o V3SchedPartition.o V3SchedReplicate.o V3SchedTiming.o V3SchedVirtIface.o V3Scope.o V3Scoreboard.o V3Slice.o V3Split.o V3SplitAs.o V3SplitVar.o V3StackCount.o V3Subst.o V3TSP.o V3Table.o V3Task.o V3Timing.o V3Trace.o V3TraceDecl.o V3Tristate.o V3Undriven.o V3Unknown.o V3Unroll.o V3VariableOrder.o V3Width.o V3WidthCommit.o V3WidthSel.o   -lpthread -lm
/usr/bin/ld: cannot find -latomic
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../Makefile_obj:341: ../../bin/verilator_bin_dbg] Error 1
make[2]: Leaving directory '/home/jkrim/verilator_5.021/src/obj_dbg'
make[1]: *** [Makefile:85: ../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/home/jkrim/verilator_5.021/src'
make: *** [Makefile:146: verilator_exe] Error 2

jordankrim, 2024-01-26:
I think the root of my problem is:

clang++ --version
clang version 16.0.6 (Red Hat 16.0.6-2.module+el8.9.0+19521+190d7aba)
Target: x86_64-redhat-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

When the make test starts, I see this:

checking whether clang++ linker accepts -latomic... no

So, why is this and can I get it to accept -latomic as that would seemingly solve all my test failures?

wsnyder, 2024-01-27:
configure.ac says try -latomic and if the linker accepts it, use it. Based on your earlier experiment it seemed -latomic didn't work, but that conflicts with your thinking it should have configured to use it?

I'd suggest manually figure out the correct compiler and linker flags, and once you get it working I can help indicate how to automate the discovery of the flags.

jordankrim, 2024-01-27:
I have Ubuntu (22.04.3) in a Oracle VM Box.  I thought that I could help debug my problem on Redhat with that.  I am able to compile the Ubuntu one "normally" (i.e. no CXX=clang++).  But, I thought it would be a better comparison if I could get the Ubuntu compiled with clang++ as I am trying to do on the Redhat one.  Unfortunately, when I try to do the configure CXX=clang++ on Ubuntu I get this:

configuring for Verilator 5.021 devel
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether clang++ accepts -g... yes
checking for a BSD-compatible install... /home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/install -c
compiler is clang++ --version = Ubuntu clang version 14.0.0-1ubuntu1.1
checking that C++ compiler can compile simple program... no
configure: error: a working C++ compiler is required

wsnyder, 2024-01-27:
Something is incorrect in your clang setup then, we run 22.04 clang against every commit, e.g. https://github.com/verilator/verilator/actions/runs/7668257285/job/20899722788

wsnyder, 2024-01-27:
And see the recipe here: https://github.com/verilator/verilator/blob/master/.github/workflows/build.yml

jordankrim, 2024-01-27:
I will attempt to wrestle with the Ubuntu installation, but I am a Linux user and not unfortunately a Sys Admin, so I have no idea yet what is or is not installed correctly.  I will have to either use the Internet to try to figure it out or better yet, I will see if any of my colleagues who are more well versed in this can help.

Meanwhile, comparing what I have working (Ubuntu with g++) against what is not working (Redhat), I return to what I showed above.  The difference is that in the Ubuntu case "-latomic" is on the Linker line and in the Redhat case, when I try to add it manually to the linker command line I get what I posted above:

/usr/bin/ld: cannot find -latomic
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../Makefile_obj:341: ../../bin/verilator_bin_dbg] Error 1
make[2]: Leaving directory '/home/jkrim/verilator_5.021/src/obj_dbg'
make[1]: *** [Makefile:85: ../bin/verilator_bin_dbg] Error 2
make[1]: Leaving directory '/home/jkrim/verilator_5.021/src'
make: *** [Makefile:146: verilator_exe] Error 2

Since I do not have admin provileges on the Redhat machine, do I need to have them install the "atomic" package and if so do you know what the command is?

jordankrim, 2024-01-27:
OK.  I have no idea what happened. but now I am able to compile Verilator 5.021 and run the tests without errors on Redhat with gcc (g++).

wsnyder, 2024-01-28:
Glad it works (somehow!)

closed 2024-01-28 by wsnyder

jordankrim, 2024-01-29:
But, unfortunately, I took a big tangent and apparently lost focus as to the original issue and it is not resolved as I still have the original problem: <command-line>: fatal error: VTestHarness__pch.h.fast: No such file or directory

jordankrim, 2024-01-29:
Is there a way to get rid of this error?

clang-16: error: unknown argument: '-fcoroutines'

I can't find the place where it is added to the clang++ (or really g++) command line.

wsnyder, 2024-01-30:
-fcoroutines is put into CFG_CXXFLAGS_COROUTINES, in include/verilated.mk when ./configure believes it is needed by running a test.  It's important to run ./configure CXX=<whatever> using the same compiler you are compiling verilated code with.

jordankrim, 2024-01-30:
Right.  On Redhat 8 I am trying to use g++ and it configures and compiles Verilator itself fine, but it gives those warnings and errors that someone thought was something having to do with the include of the PCH file(s) on the Verilated code.  Then, clang++ on Redhat 8 gives atomic errors on the linker, so I can't seem to compile (the latest Verilator) no matter what at this point.  I am not having much luck on Ubuntu either.

wsnyder, 2024-01-30:
Every single commit compiles ok on Ubuntu 22.04 & 20.24, so debug 22.04 first as that not working suggests something in your config etc.   I would avoid Redhat until you have Ubuntu working, especially as Redhat's known to be very old and nearly end-of-life. Did you try a totally clean Ubuntu 22.04 VM without any personal configuration, and running the verilator examples?  See above for the command scripts that we run.  The install documentation was also correct recently to what's needed for 22.04.

I'd suggest ask around for someone C++ knowledgeable in your organization to help you.  (I presume you are in such an org, as no one would be running RedHat 8 unless inside a huge org ;)

jordankrim, 2024-01-30:
I am able to compile with clang++ (was going to paste the entire problems along the way but skipped for now) on Ubuntu 22.04.3, but when I run the make test I get this:

/usr/include/c++/11/mutex: In member function 'bool std::recursive_timed_mutex::_M_clocklock(clockid_t, const __gthread_time_t&)':
/usr/include/c++/11/mutex:338:17: error: 'pthread_mutex_clocklock' was not declared in this scope; did you mean 'pthread_mutex_unlock'?
  338 |       { return !pthread_mutex_clocklock(&_M_mutex, clockid, &__ts); }
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                 pthread_mutex_unlock

wsnyder, 2024-01-31:
Googling that error I see this: https://bugs.archlinux.org/task/66864 which suggest's arch linux not Ubuntu.   Also Ubuntu uses clang 14, so the  Very odd. Not sure where to go from there.   Here's info from my stock 22.04 that runes fine.
```
# dpkg -S /usr/include/c++/11/mutex
libstdc++-11-dev:amd64: /usr/include/c++/11/mutex
# apt list --installed | grep libstdc++-11-dev
libstdc++-11-dev/jammy-updates,jammy-security,now 11.4.0-1ubuntu1~22.04 amd64 [installed,automatic]
# clang++ --version
Ubuntu clang version 14.0.0-1ubuntu1.1

```

jordankrim, 2024-01-31:
Using their example:

g++ xtoff.cpp -o xtoff

In file included from /usr/include/c++/11/mutex:43,
                 from xtoff.cpp:1:
/usr/include/c++/11/bits/std_mutex.h: In member function 'void std::__condvar::wait_until(std::mutex&, clockid_t, timespec&)':
/usr/include/c++/11/bits/std_mutex.h:169:7: error: 'pthread_cond_clockwait' was not declared in this scope; did you mean 'pthread_cond_wait'?
  169 |       pthread_cond_clockwait(&_M_cond, __m.native_handle(), __clock,
      |       ^~~~~~~~~~~~~~~~~~~~~~
      |       pthread_cond_wait
/usr/include/c++/11/mutex: In member function 'bool std::timed_mutex::_M_clocklock(clockid_t, const __gthread_time_t&)':
/usr/include/c++/11/mutex:272:17: error: 'pthread_mutex_clocklock' was not declared in this scope; did you mean 'pthread_mutex_unlock'?
  272 |       { return !pthread_mutex_clocklock(&_M_mutex, clockid, &__ts); }
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                 pthread_mutex_unlock
/usr/include/c++/11/mutex: In member function 'bool std::recursive_timed_mutex::_M_clocklock(clockid_t, const __gthread_time_t&)':
/usr/include/c++/11/mutex:338:17: error: 'pthread_mutex_clocklock' was not declared in this scope; did you mean 'pthread_mutex_unlock'?
  338 |       { return !pthread_mutex_clocklock(&_M_mutex, clockid, &__ts); }
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
      |                 pthread_mutex_unlock

But, using clang++ it compiles fine.   Does that help?

Also, if I use their example and try to compile it with /usr/bin/g++:

/usr/bin/g++ xtoff.cpp -o xtoff
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `_dl_audit_symbind_alt@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `__nptl_change_stack_perm@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `_dl_find_dso_for_object@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `_dl_fatal_printf@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `_dl_exception_create@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `__tunable_get_val@GLIBC_PRIVATE'
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: undefined reference to `_dl_audit_preinit@GLIBC_PRIVATE'
collect2: error: ld returned 1 exit status

wsnyder, 2024-01-31:
Maybe I missed it before but:

/home/jkrim/cep_v4_4_new3_test_verilator/CEP/.conda-env/bin/ld

You are using conda's loader, not ubuntu's, that I suspect is your root problem.  Use the Ubuntu tool chain!!!!

jordankrim, 2024-01-31:
Do you know how to dump conda's and use the Ubuntu tool chain?

wsnyder, 2024-01-31:
Please ask conda team.

jordankrim, 2024-01-31:
I think I undid the conda env, but the results on Ubuntu didn't change.  Is there something wrong with my versions?

g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

c++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0

Ubuntu clang version 14.0.0-1ubuntu1.1
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

The problem is then when I want to create the verilated code I *must* be in the conda env and so that g++ version may be different from the one use to create the Verilator binary.

wsnyder, 2024-02-01:
Please run a verilator example (from the Verilator git) under Ubuntu.  Completely ignore conda in all respects.   Do those pass?

wsnyder, 2024-02-01:
(If not make sure nothing in your PATH/.bashrc, and make a log file of your entire configuration and run of Verilator starting from a completely clean start)