verilator/verilator #4840: Using the two modules generated by verilator, the top level connects the two modules via CPP, and now only the waveform of one of the modules can be dumped at the same time

myloverleaf, 2024-01-19:
Here is a simple example that generates obj_dir_sub_top0 and  obj_dir_sub_top1, respectively.We manually create a Vtop CPP that connects Vsub_top0 and Vsub_top1.
Here are the connections I want to implement by verilog and Details of the implementation using C++ linkage
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wsnyder, 2024-01-20:
It's very hard to read this, in the future please paste only text of your example, using OCR if needed to convert text to ascii.  Also we don't really need to see the generated code.

Anyhow you seem to be using the same contextp for both modules which is good.  I don't see where you are calling eval_end_step.  See this example of how to do this: https://github.com/verilator/verilator/blob/master/test_regress/t/t_trace_two_cc.cpp

Closing as answered but feel free to follow up here.

closed 2024-01-20 by wsnyder

myloverleaf, 2024-01-20:
"I can't see the connection between the two modules in the examples you provided. What I want to express is that the output sum0 and sum1 from my subtop0 module will be connected to the inputs p4 and p5 of subtop1. Currently, my approach is to pass the result of evaluating the first module to the second module, but after both modules produce VCD files, the waveform cannot be displayed correctly."

wsnyder, 2024-01-20:
You need to call separately step and end_step.

```
        ap->eval_step();
        bp->eval_step();
        ap->eval_end_step();
        bp->eval_end_step();
```

Also the C++ interface does not have any way to deal with races, so you should pass all signals across when and only when no clocks are changing, then pass clocks separately.  This can be difficult, SystemC is a much better way to connect models.