verilator/verilator #4841: Wrong runtime value after optimization under specific circumstances

flaviens, 2024-01-19:
Hi there!

My apologies for opening another issue :) this one is super short and easy to reproduce.
[In this very little repository](https://github.com/flaviens/verilator-b11), you will find a small design on which Verilator produces a wrong value.
This bug is corrected by the individual flags:
`-fno-const-bit-op-tree`
`-fno-expand`
`-fno-subst`
`-fno-subst-const`

Please let me know if you require any more information.
Thank you,
Flavien

yTakatsukasa, 2024-01-19:
@wsnyder could you please help me interpreting AST ?

Here is the input of bit-op-tree opt.
```
    1:2:3:1:2:1: AND 0x556bb9df4670 {d13cf} @dt=0x556bb9df8ac0@(G/wu32/2)
    1:2:3:1:2:1:1: CONST 0x556bb9e0e270 {d13bj} @dt=0x556bb9df8f60@(G/wu32/1)  32'h2
    1:2:3:1:2:1:2: AND 0x556bb9e08220 {d13cq} @dt=0x556bb9df8f60@(G/wu32/1) 
 ```
In the 3rd line, type is `(G/wu32/1)` so bit op tree thinks only the LSB matters, then optimize the subtree of the node.
But the result of the 3rd line is AND-ed with constant `32'h2`, whose width is `(G/wu32/1) `.
It means what is used of the 3rd line is `[1]`, not LSB of `[0]`.

I think width of the 3rd line should be `(G/wu32/2)`. What do you think ?

yTakatsukasa, 2024-01-20:
I think I understand what is going on, and tests pass on my branch.
I'll polish it to PR.

yTakatsukasa, 2024-01-20:
#4837 looks to have same root cause.

wsnyder, 2024-01-21:
Thanks @yTakatsukasa !

closed 2024-01-21 by wsnyder