verilator/verilator #4842: Wrong runtime value after optimization corrected by tracing or 3 individual flags

flaviens, 2024-01-19:
Hi there,

Thank you for all your contributions!
[Here](https://github.com/flaviens/verilator-b13) is what looks like an optimization bug that is corrected by tracing or any of:
`-fno-const-bit-op-tree`
`-fno-expand`
`-fno-gate`

Please let me know if you need any more information.

Thanks!
Flavien

yTakatsukasa, 2024-01-19:
Hmm,  result with the latest master(89cfa0737fafb81e91a000a3d328b8f02f2a103e) on my side is as below.
Anyway const-bit-op-tree looks to have a trouble. 
```
-fno-acyc-simp
Output[33] 0
-fno-assemble
Output[33] 0
-fno-case
Output[33] 0
-fno-combine
Output[33] 0
-fno-const
Output[33] 0
-fno-const-bit-op-tree
Output[33] 1
-fno-dedup
Output[33] 0
-fno-dfg
Output[33] 0
-fno-dfg-peephole
Output[33] 0
-fno-dfg-pre-inline
Output[33] 0
-fno-dfg-post-inline
Output[33] 0
-fno-expand
Output[33] 1
-fno-gate
Output[33] 0
-fno-inline
Output[33] 0
-fno-life
Output[33] 0
-fno-life-post
Output[33] 0
-fno-localize
Output[33] 0
-fno-merge-cond
Output[33] 0
-fno-merge-cond-motion
Output[33] 0
-fno-merge-const-pool
Output[33] 0
-fno-reloop
Output[33] 0
-fno-reorder
Output[33] 0
-fno-split
Output[33] 0
-fno-subst
Output[33] 1
-fno-subst-const
Output[33] 1
-fno-table
Output[33] 0
```

flaviens, 2024-01-19:
Thank you for your response! Interesting. Tested with `Verilator 5.021 devel rev v5.020-31-g54922c036`.

yTakatsukasa, 2024-01-20:
This case seems to need both fix for #4832 and #4841 (dup of #4837).

yTakatsukasa, 2024-01-21:
The latest master (6b8531f0a647c14ee7045ce707b3cf1961fc84d2 and newer) shows `Output[32] 0`.

closed 2024-01-21 by yTakatsukasa

flaviens, 2024-01-22:
Very cool thanks!