verilator/verilator #4850: State variable missing; latches state incorrectly, and missing in list of CData struct

paul-demo, 2024-01-22:
I am seeing a somewhat alarming bug which I think is caused by some invalid operation performed by Verilator when it generates 8-bit math in the verilated C++. I'm using Verilator version 5.018 2023-10-30 on an arm64 platform. I looked at the slew of bugs reported recently in this area (Github issues 3824, 4709, 4832, 4837, 4841, 4842) but I did not find that any of the recommended "-fno-*" flags had any effect.

To focus just on the 1 line which I have determined is causing the issue with Verilator, see the following section of code. The calculation of d_bytes_next is more complicated than shown, but it should be clear that changing `d_bytes` to *any width greater than or equal to* the width of `d_bytes_next` should have absolutely no effect, since it will truncate to 4 bits in the first line of the always_comb block (`d_bytes_next = d_bytes;`) and it will width-expand with zero padding on the left when it is assigned to d_bytes in the always_ff block (`d_bytes <= d_bytes_next;`).

```verilog
  logic [3:0] d_bytes = '0;  // THIS DOES NOT WORK: d_bytes takes value 0x0 when it should take 0xA.
  // logic [7:0] d_bytes = '0;  // THIS DOES NOT WORK
  // logic [11:0] d_bytes = '0;  // THIS FIXES THE PROBLEM, EVEN THOUGH ONLY 4 BITS ARE NECESSARY
  // logic [8:0] d_bytes = '0;  // THIS FIXES THE PROBLEM ALSO, SUGGESTING 8-BIT VERILATOR MATH IS WRONG

  logic [3:0] d_bytes_next;
  always_ff @(posedge clk) begin
    if (reset) begin
      d_bytes <= '0;
    end else begin
      d_bytes <= d_bytes_next;
  end

  always_comb begin
    d_bytes_next = d_bytes;
   // additional logic that modifies d_bytes_next
  end
```

However, I found that the simulation shows that `d_bytes` is sometimes taking value `'0` when it should not! Specifically, it seems to do this if the width of `d_bytes` is less than or equal to 8 bits. This suggests there is a bug in the way Verilator performs 8-bit math (and 4-bit math, and any width less than or equal to 8-bit).

A screenshot is below showing that it takes the value zero when `d_bytes` is 4 bits wide, even though reset is not asserted and `d_bytes_next` is `0xA`.

[image]

When I increase the width (unnecessarily) to something wider than 8-bits, then it works. There is no other change in the code; I've found that this is the only line which has any effect.

[image]

My suspicion is that there is some math operation which has been optimized out by Verilator (incorrectly), or else there is some fundamental 8-bit math logic bug where some variable is saturating or rolling over incorrectly in the C++ that Verilator generates.

Grepping around in the build directory, I also noticed that `d_bytes` is not declared as are all the other state variables. The variables `d_free` and `d_free_next` have similar declarations in the code, and yet they appear to have different CData declarations in the C++ that Verilator generates. I'm not sure if this is a valid lead or expected, but it does seem suspicious that Verilator is not declaring the exact state variable which is causing problems!

```c++
// Verilated -*- C++ -*-
// DESCRIPTION: Verilator output: Design internal header
// See Vtop.h for the primary calling header

#ifndef VERILATED_VTOP_AXIS_INSERT_BYTES_H_
#define VERILATED_VTOP_AXIS_INSERT_BYTES_H_  // guard

#include "verilated.h"
class Vtop__Syms;

class alignas(VL_CACHE_LINE_BYTES) Vtop_axis_insert_bytes final : public VerilatedModule {
  public:

    // DESIGN SPECIFIC STATE
    // Anonymous structures to workaround compiler member-count bugs
    struct {
        // ...
        // NOTICE THAT D_BYTES IS NOT DECLARED HERE!
        CData/*3:0*/ d_bytes_next;
        CData/*3:0*/ d_free;
        CData/*3:0*/ d_free_next;
        // ...
    };
    struct {
        // ...
        SData/*11:0*/ d_keep;
        SData/*11:0*/ d_keep_next;
        SData/*11:0*/ d_bytes;
        SData/*11:0*/ __Vdly__d_bytes;
        // ...
    };

    // INTERNAL VARIABLES
   // ...
};

#endif  // guard

```

wsnyder, 2024-01-22:
Most likely Verilator is missing inserting some C++ cast, as for the most part that's the only difference between CData (8 bits) and wider that happens at the 8 vs 9-bit width difference, causing the 8-bit wraparound due to C's type rules.

Can you please make this into a either ideally Verilog-only test that shows the problem, or Verilog+C++ wrapper that does, it should fail on Verilator, and pass on other simulators or with the width changed on Verilator?  Or, debug yourself to follow the .tree logic to see where the signal goes bad (see internals.rst for details on these files).

wsnyder, 2024-01-22:
Also make sure you look at -Wall warnings as Verilator may be giving a WIDTH warning. While WIDTH-flagged problems should still simulate correctly, when suppressed often cause hitting these corner-case sort of bugs.

paul-demo, 2024-01-22:
One question I had is how could `d_bytes_next` show the correct value but `d_bytes` does not? Doesn't that suggest that the problem is not in the complicated expansion of next-state logic in the always_comb? It's something to do with just the final flip-flop assignment to `d_bytes`, it seems.

wsnyder, 2024-01-22:
Optimizations can apply to one usage but not another.  Also don't trust waves has the same value as the logic, waves are designed to help users debug their code, not do debug Verilator, and so are still subject to optimization.  Best bet is to make a self-checking test, as then you're sure waves isn't lying to you.

paul-demo, 2024-01-22:
Declaring the variable in the broken version [3:0] and in the working version [8:0] and looking at all the differences in the build artifacts, here's what I see:

*`Vtop__Syms.cpp`: (seems OK)*
- VLVT_UINT8 -> VLVT_UINT16
- dims 1, 3, 0 -> dims 1, 8, 0

*`Vtop__Trace__0.cpp`: (seems OK)*
- chgSData final parameter 4 -> 9

*`Vtop___024root.h`: (maybe OK?)*
- there are two structs, and the offending variable moves from the top struct to the bottom struct as I work around the problem increasing the width from 4-bits to 9-bits:
- With 4-bit width, the top and bottom structs are:
```c++
struct {
CData/*3:0*/ mymodule__DOT__d_bytes;
CData/*3:0*/ mymodule__DOT__d_bytes_next;
}
struct {
}
```
- With 9-bit width, the top struct is:
```c++
struct {
CData/*3:0*/ mymodule__DOT__d_bytes_next;
}
struct {
SData/*8:0*/ mymodule__DOT__d_bytes;
}
```

*'Vtop___024root__DepSet_*__0.cpp`: (maybe OK?)*
- For the line `d_bytes_next = d_bytes;` in the `always_comb` block, the broken version does not mask the width in the initial assignment nor does it cast to `(IData)` type first:
```c++
vlSelf->mymodule__DOT__d_bytes_next = vlSelf->mymodule__DOT__d_bytes;
```
- The functional version does cast it to `(IData)` type and then masks it with the correct bitmask `0xfU`:
```c++
vlSelf->mymodule__DOT__d_bytes_next = 
        (0xfU & (IData)(vlSelf->mymodule__DOT__d_bytes));
```
- Also in this file, all consumers of d_bytes mask the value before consuming it:
```c++
(0xfU & (IData)(vlSelf->mymodule__DOT__d_bytes))
```

Whereas the broken version it does not mask the value with `0xfU`:
```c++
(IData)(vlSelf->mymodule__DOT__d_bytes)
```

There are no other changes between working and non-working versions. So I think this leaves two possibilities:
1) `mymodule__DOT__d_bytes` is storing some higher-order bits and leaking them into the computations, due to the fact that Verilator does not cast to `(IData)` then mask using the width using `0xfU & ...`. This results in incorrect math because bits 5..8 are nonzero and Verilator did not mask them out.

2) The issue is not visible in this "A vs B" comparison that I've done here because it's some other line which is literally unchanged but only differs semantically due to the fact that the offending variable is a `uint8` in one case and a `uint16` in another case.

paul-demo, 2024-01-22:
I think the problem might be in `_vl_insert_WI()` from `verilated_funcs.h` which could possibly be clobbering values undesirably.

Plotting a few more variables and noting their position in the struct:
```
        VlWide<3>/*95:0*/ mymodule__DOT__d_data;
        VlWide<3>/*95:0*/ mymodule__DOT__d_data_next;
        CData/*3:0*/ mymodule__DOT__d_bytes;
        CData/*3:0*/ mymodule__DOT__d_bytes_next;
        CData/*3:0*/ mymodule__DOT__d_free;
        CData/*3:0*/ mymodule__DOT__d_free_next;
```

I also note that `d_free` is being erroneously driven to zero as well as `d_bytes`. Those two are actually the complements of each other and should always add up to 12, since I have `assign d_free_next   = 12 - d_bytes_next;`. However they're both being driven to zero in the offending cycle of the simulation, which suggests that a call that sets `d_data_next` to zero is running beyond the end of the intended width and clobbering subsequent variables.

[image]

I'm trying to dig through what `_vl_insert_WI` is doing (it's called by `VL_ASSIGNSEL_WI` at the problematic section in the verilated code).

paul-demo, 2024-01-22:
Ok, I've found the source code that is problematic and it does appear to be caused by a bug in `_vl_insert_WI()`.

I have a line of source 4 bytes are written to a variable position in a buffer:
```
logic [SS*WIDTH_BYTES-1:0][7:0] d_data, d_data_next;

always_comb begin
  d_data_next = d_data;

  if (something) begin
      d_data_next[d_bytes+:WIDTH_BYTES] = '0;

      /*
      // VERILATOR GENERATES:
        if ((0x5fU >= (0x7fU & VL_SHIFTL_III(7,32,32, (IData)(vlSelf->mymodule__DOT__d_bytes), 3U)))) {
            VL_ASSIGNSEL_WI(96,32,(0x7fU & VL_SHIFTL_III(7,32,32, (IData)(vlSelf->mymodule__DOT__d_bytes), 3U)), vlSelf->mymodule__DOT__d_data_next, vlSelf->mymodule__DOT____Vlvbound_ha82cf1f6__1);
        }
      */

  end
  
  if (something_else) begin
      d_data_next >>= (WIDTH_BYTES * 8);
      
      /*
      // VERILATOR GENERATES:
        VL_SHIFTR_WWI(96,96,32, __Vtemp_5, vlSelf->mymodule__DOT__d_data_next, 0x20U);
        vlSelf->mymodule__DOT__d_data_next[0U] 
            = __Vtemp_5[0U];
        vlSelf->mymodule__DOT__d_data_next[1U] 
            = __Vtemp_5[1U];
        vlSelf->mymodule__DOT__d_data_next[2U] 
            = __Vtemp_5[2U];
      */
  end
end
```

And Verilator implements this assignment unsafely/incorrectly even though `d_bytes == 7` and so the range of outer indices is `[10:7]` which should be valid since `SS == 3` and `WIDTH_BYTES = 4`, so all indices up to 11 are valid.

I can show that this is the line that Verilator is computing incorrectly simply by changing the `'0` in the above code to `32'h12345678`. From the first to the second screenshot you can see that the erroneous `d_bytes = 0x0` and `d_free = 0x0` values now become two different erroneous values: `d_bytes = 0x6` and `d_free = 0x2`.

**So basically Verilator does not write to the 2-D array correctly (even though the index range is valid) and it overwrites unrelated variables that happen to be adjacent in the C++ struct!**

[image]

paul-demo, 2024-01-23:
Here's a very simple test which shows the Verilator bug. You can see that the *only* difference is the order in which the variables are declared at the top of the module body. Obviously, that should have no effect. Furthermore, in the cycle that the write `d_data_next[d_used+:4] = 32'd0;` is clobbering the `d_used`, that range is actually *valid*. Specifically, it should be writing d_data_next[11:8], which is a legitimate range to be writing since the buffer stores 12 bytes. So, even if you think running over the end of a variable should clobber other variables -- which I would argue it shouldn't -- I think there is still a bug because this is not an index-out-of-bounds issue.

```verilog
// Demonstrates the bug in https://github.com/verilator/verilator/issues/4850

module issue_4850 (
    input logic clk,
    input logic insert,

    output logic [3:0] used,
    output logic [3:0] free
);
  // ========
  // OPTION A: THIS FAILS BECAUSE ASSIGNING 32'd0 CLOBBERS d_used.
  logic [11:0][7:0] d_data, d_data_next;
  logic [3:0] d_used = 4'd1, d_free = 4'd11, d_used_next;

  // OPTION A OUTPUT:
  // d_used =  1, d_free = 11
  // d_used =  1, d_free = 11
  // d_used =  2, d_free = 10
  // d_used =  3, d_free =  9
  // d_used =  4, d_free =  8
  // d_used =  5, d_free =  7
  // d_used =  6, d_free =  6
  // d_used =  7, d_free =  5
  // d_used =  8, d_free =  4  -> HERE, WE INTEND TO BE WRITING d_data_next[8+:4] = d_data_next[11:8] which is a valid range!
  // d_used =  0, d_free =  3  -> INCORRECT. d_used was clobbered by a write to d_data_next[11:8] when it should not be!
  // d_used =  1, d_free = 11

  // ========
  // OPTION B: THIS WORKS BECAUSE d_data IS DECLARED LAST, SO IT DOES NOT CLOBBER d_used.
  // logic [3:0] d_used = 4'd1, d_free = 4'd11, d_used_next;
  // logic [11:0][7:0] d_data, d_data_next;

  // OPTION B OUTPUT:
  // d_used =  1, d_free = 11
  // d_used =  1, d_free = 11
  // d_used =  2, d_free = 10
  // d_used =  3, d_free =  9
  // d_used =  4, d_free =  8
  // d_used =  5, d_free =  7
  // d_used =  6, d_free =  6
  // d_used =  7, d_free =  5
  // d_used =  8, d_free =  4
  // d_used =  9, d_free =  3
  // d_used = 10, d_free =  2
  // d_used = 11, d_free =  1
  // d_used = 12, d_free =  0 -> OK. THIS STAYS THIS WAY INDEFINITELY
  // d_used = 12, d_free =  0
  // ...

  assign used = d_used;
  assign free = d_free;

  always_ff @(posedge clk) begin
    d_data <= d_data_next;
    d_used <= d_used_next;
    d_free <= 12 - d_used_next;
    $display("d_used = %d, d_free = %d", d_used, d_free);
  end

  always_comb begin
    d_data_next = d_data;
    d_used_next = d_used;

    if ((insert == 1'b1) && (d_free >= {3'b0, insert})) begin
      d_data_next[d_used+:4] = 32'd0;
      d_used_next += 4'd1;
    end
  end

endmodule

```

The trace is in agreement with the $display commands too:
[image]

paul-demo, 2024-01-23:
This verilog module shows the problem. Basically we are writing blocks of 4 bytes to d_data starting at an index defined by d_used.  starting with 1 byte stored and moving up slowly you find that the write to `d_data{_next}` clobbers an unrelated variable just by them being declared adjacent to each other. Obviously the declaration order should have no effect on the result, but it does because of this bug.

```verilog
// Demonstrates the bug in https://github.com/verilator/verilator/issues/4850

module issue_4850 (
    input logic clk,
    input logic insert,

    output logic [3:0] used,
    output logic [3:0] free
);
  // ========
  // OPTION A: THIS FAILS BECAUSE ASSIGNING 32'd0 CLOBBERS d_used.
  logic [11:0][7:0] d_data, d_data_next;
  logic [3:0] d_used = 4'd1, d_free = 4'd11, d_used_next;

  // OPTION A OUTPUT:
  // d_used =  1, d_free = 11
  // d_used =  2, d_free = 10
  // d_used =  3, d_free =  9
  // d_used =  4, d_free =  8
  // d_used =  5, d_free =  7
  // d_used =  6, d_free =  6
  // d_used =  7, d_free =  5
  // d_used =  8, d_free =  4  -> HERE, WE INTEND TO BE WRITING d_data_next[8+:4] = d_data_next[11:8] which is a valid range!
  // d_used =  0, d_free =  3  -> INCORRECT. d_used was clobbered by a write to d_data_next[11:8] when it should not be!
  // d_used =  1, d_free = 11

  // ========
  // OPTION B: THIS WORKS BECAUSE d_data IS DECLARED LAST, SO IT DOES NOT CLOBBER d_used.
  // logic [3:0] d_used = 4'd1, d_free = 4'd11, d_used_next;
  // logic [11:0][7:0] d_data, d_data_next;

  // OPTION B OUTPUT:
  // d_used =  1, d_free = 11
  // d_used =  2, d_free = 10
  // d_used =  3, d_free =  9
  // d_used =  4, d_free =  8
  // d_used =  5, d_free =  7
  // d_used =  6, d_free =  6
  // d_used =  7, d_free =  5
  // d_used =  8, d_free =  4
  // d_used =  9, d_free =  3
  // d_used = 10, d_free =  2
  // d_used = 11, d_free =  1
  // d_used = 12, d_free =  0 -> OK. STAYS THIS WAY INDEFINITELY
  // d_used = 12, d_free =  0
  // ...

  assign used = d_used;
  assign free = d_free;

  always_ff @(posedge clk) begin
    d_data <= d_data_next;
    d_used <= d_used_next;
    d_free <= 12 - d_used_next;
    $display("d_used = %d, d_free = %d\n", d_used, d_free);
  end

  always_comb begin
    d_data_next = d_data;
    d_used_next = d_used;

    if ((insert == 1'b1) && (d_free >= {3'b0, insert})) begin
      // Uncomment this $display to see where it all goes wrong in Option A.
      $display("Writing to d_data_next[%d+:4] = d_data_next[%d:%d]", d_used, d_used + 3, d_used);
      d_data_next[d_used+:4] = 32'd0;
      d_used_next += 4'd1;
    end
  end

endmodule
```

The waves agree:
[image]

If you uncomment the lower `$display` you can see that the always-comb block gets executed multiple times by Verilator, and during the problematic cycle, that value changes after the first execution! Whereas it does not for other cycles:

```bash
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
Writing to d_data_next[ 8+:4] = d_data_next[        11: 8]
d_used =  8, d_free =  4

Writing to d_data_next[ 9+:4] = d_data_next[        12: 9]  -> this write is unsafe and clobbers other variables in Option A.
Writing to d_data_next[ 0+:4] = d_data_next[         3: 0]
Writing to d_data_next[ 0+:4] = d_data_next[         3: 0]
Writing to d_data_next[ 0+:4] = d_data_next[         3: 0]
Writing to d_data_next[ 0+:4] = d_data_next[         3: 0]
Writing to d_data_next[ 0+:4] = d_data_next[         3: 0]
d_used =  0, d_free =  3

```

wsnyder, 2024-01-23:
Great debugging, perhaps are you willing to carry it through to a pull that fixes it?

Maybe the width of the expression is mis-passed to the functions (long ago there weren't 2D arrays in Verilog)?    Any "W" function should only be called on wide (>32 bit) vectors.

paul-demo, 2024-01-23:
I'm not sure how to do that, unfortunately.

I think you could add some check in VL_ASIGNSEL_WI, but as far as I can tell that's too late to determine whether `hbit` is too high because `iowp` doesn't have that kind of type information -- it appears to just be a pointer to a uint32.

But conceptually you sort of want this (if max_possible_hbit were a real field):

```c++
static inline void VL_ASSIGNSEL_WI(int rbits, int obits, int lsb, WDataOutP iowp,
                                   IData rhs) VL_MT_SAFE {
    int hbit = lsb + obits - 1;
    if (hbit > iowp.max_possible_hbit) {
        hbit = iowp.max_possible_hbit;
   }
    _vl_insert_WI(iowp, rhs, hbit, lsb, rbits);
}
```

rather than what you have which is this:
```c++
static inline void VL_ASSIGNSEL_WI(int rbits, int obits, int lsb, WDataOutP iowp,
                                   IData rhs) VL_MT_SAFE {
    _vl_insert_WI(iowp, rhs, lsb + obits - 1, lsb, rbits);
}
```

If that's not possible, then you need to change `VL_ASSIGNSEL_WI()` at the point that it is called (ie, in the generated code). But, I don't see anywhere where that happens in Github code search. I'm assuming some perl / lexer sort of thing emits the C++ by reading the verilog source one token at a time.

wsnyder, 2024-01-23:
Might be right, can you please further polish you test to be self-checking?  Look at test_regress/t/t_EXAMPLE.v for a way to do this using randomish stimulus you can then check against another simulator.

paul-demo, 2024-01-23:
First time doing this, but please have a look at the PR here:
https://github.com/verilator/verilator/pull/4855

The test I added fails without my change to `verilated_funcs.h` and succeeds with the change.

I didn't know what you want to name the test, so feel free to edit/rename everything as you see fit.

wsnyder, 2024-01-23:
Thanks for the good debug and pull request, merged.

closed 2024-01-23 by wsnyder