verilator/verilator #4851: How to use Verilator with a UVM/SystemVerilog Testbench

jordankrim, 2024-01-22:
I realize that Verilator for UVM is still under development, but I would like to try it.  I can't find any documentation on how to do this.
Is there any?  If not, how does one run a UVM/SystemVerilog Testbench with Verilator?

wsnyder, 2024-01-23:
See later comments in #1538.

>Also, what is the minimum version of Verilator needed to run UVM/SystemVerilog?

As it's not formally supported yet, some number in the future ;)  But, as you're experimenting, use master as will need to likely make pull requests.

closed 2024-01-23 by wsnyder

jordankrim, 2024-01-23:
I see a UVM repo that has mods for Verilator, but not sure how to incorporate UVM into the Verilator compile.

wsnyder, 2024-01-23:
Use it like other simulators, you point to the library directory with +incdir+ and your test Verilog source does an "`include "uvm_pkg.sv"

jordankrim, 2024-01-23:
I gave it a try.  My command line:  
	verilator -CFLAGS -DDISPLAY -CFLAGS -DVL_DEBUG --cc --clk clk --default-language 1364-2001 -I. -I./rsa_uvm_tb +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/macros +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/base +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/comps +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/seq +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/tlm1 +incdir+/mnt/data/projects/project/shared/accellera_uvm_files_modified_for_verilator/src/tlm2 ./rtl_new_div_sans_secworks/modexp_core.v ./rtl_new_div_sans_secworks/blockmem2rptr1w.v ./rtl_new_div_sans_secworks/blockmem2r1wptr.v ./rtl_new_div_sans_secworks/unsigned_serial_divide.v --top-module modexp_core -f ./rsa_uvm_tb/uvm_file_list --stats 

Got this (an example -- maxed at 50 errors):
%Error: rsa_uvm_tb/rsa_sequence.sv:4:1: syntax error, unexpected IDENTIFIER
    4 | class rsa_sequence extends rsa_sequence_base;
      | ^~~~~
        rsa_uvm_tb/rsa_master_agent.sv:5:1: ... note: In file included from rsa_master_agent.sv
        rsa_uvm_tb/rsa_env.sv:5:1: ... note: In file included from rsa_env.sv
%Error: rsa_uvm_tb/rsa_sequence.sv:14:3: Define or directive not defined: '`uvm_object_utils_begin'
   14 |   `uvm_object_utils_begin(rsa_sequence)
      |   ^~~~~~~~~~~~~~~~~~~~~~~
        rsa_uvm_tb/rsa_master_agent.sv:5:1: ... note: In file included from rsa_master_agent.sv
        rsa_uvm_tb/rsa_env.sv:5:1: ... note: In file included from rsa_env.sv
%Error: rsa_uvm_tb/rsa_sequence.sv:15:5: Define or directive not defined: '`uvm_field_int'
   15 |     `uvm_field_int(item_count, UVM_ALL_ON)
      |     ^~~~~~~~~~~~~~
        rsa_uvm_tb/rsa_master_agent.sv:5:1: ... note: In file included from rsa_master_agent.sv
        rsa_uvm_tb/rsa_env.sv:5:1: ... note: In file included from rsa_env.sv

wsnyder, 2024-01-24:
Are you sure you're using the latest Verilator?

Anyhow as described in #1538 this is all experimental, if you want to play with it and debug in order to contribute that's fine, but we aren't really ready to debug any user issues as there's too much known work still left.

jordankrim, 2024-01-24:
I am not using the latest Verilator but will as soon as issue #4839 is resolved.

jordankrim, 2024-01-24:
I am now using the latest version (5.021 ?) even though it doesn't pass the "make test" yet and there are problems (I think unrelated to the atomic problem) in compiling my UVM/SV testbench.  I need to understand what the mechanism is for reporting these.  I looked at the issues remaining for UVM and I didn't see the issues that I am getting when compiling, so can I report these somehow for help in determining if I am just missing some compile option etc. or if it is truly not implemented yet?

wsnyder, 2024-01-24:
We really don't want any issues filed on UVM, unless you personally/your company/etc are going to immediately start working on pull requests to fix them (and really please consider doing so!).  Otherwise, they would just be a distraction, as we know there's a ton broken.  Once most/all the UVM self-tests pass we'll announce it as supported and start accepting user issues.