verilator/verilator #4853: VerilogAMS Support Features of Verilator

RionelCaldo, 2024-01-23:
Verilator released the new version 5.021 last January 17, 2024. It supports some Verilog-AMS function, for example "wreal". I wanted to investigate the specification if we can we use it for analog model simulation. Can you provide me working example of Verilog-AMS that can run in Verilator including testbench? I also need to know the language limitations. I already read the released document but with minimal discussions about Verilog-AMS.  Thank you.

Verilator has implemented a very small subset of Verilog AMS (Analog and Mixed-Signal Extensions) with the subset corresponding to those VMS keywords with near-equivalents in Verilog 2005 or SystemVerilog 2017 [1]. 

     AMS parsing is enabled with --language VAMS or --language 1800+VAMS.
     Verilator implements ceil, exp, floor, ln, log, pow, sqrt, string, and wreal.

The Verilator documentation also provides information on simulating Verilog models using Verilated-Model Runtime [2]. However, we could not find any information on whether Verilator supports the “wreal” function for analog model simulation.

[1] “Input Languages — Verilator Devel 5.021 documentation,” verilator.org. https://verilator.org/guide/latest/languages.html (accessed Jan. 19, 2024).
[2] “Simulating (Verilated-Model Runtime) — Verilator Devel 5.021 documentation,” verilator.org. https://verilator.org/guide/latest/simulating.html (accessed Jan. 19, 2024).

wsnyder, 2024-01-23:
Verilator supports "wreal" as a real value wire, but nothing else of value in analog simulation. It's unlikely of value in any complex analog modeling.  If you think that's enough, try it.

closed 2024-01-23 by wsnyder

RionelCaldo, 2024-01-23:
Thank you for your kind reply. I've tried simple example that uses the wreal data type to model a simple analog circuit and run both in v2.26 and v5.021 but I've encountered errors (see attached files):

The errors encountered suggest that Verilator is treating the variables voltage_in and current_out as cells instead of variables. This might be due to the fact that Verilog-AMS introduces concepts like wreal and continuous assignments that Verilator, being primarily a digital simulator, may not fully support. Verilator is optimized for synthesizable digital designs in Verilog/SystemVerilog, and it may not handle analog constructs like wreal in the same way as analog simulators. 

Can I get working sample of wreal in Verilator? And is below syntax acceptable: verilator -Wall -cc analog_example.vams --exe tb_analog_example.vams

wsnyder, 2024-01-24:
You didn't provide an attachment.  Anyhow, you can find a wreal example in [test_regress/t/t_vams_wreal.v](https://github.com/verilator/verilator/blob/master/test_regress/t/t_vams_wreal.v).  As above there's no true AMS support, what works works, and not much else ;)

RionelCaldo, 2024-01-24:
[simple_vams.zip](https://github.com/verilator/verilator/files/14031251/simple_vams.zip)

Kindly see attached zipped file. Thank you.

Thank you for sending sample wreal example. By the way, Verilator was very helpful in my Verilator Technical Research last October 2022.

Below are some contents of this material:
1) Background/Purpose
2) Requirements
3) Verilator Capabilities
4) Running Simple ALU in Verilator 
5) Understanding the basics of using Verilator and writing C++ testbenches for Verilog/SystemVerilog modules
6) Simulation Time Comparison of Verilator and Xcelium for Simple ALU
7) How to Install Verilator in Linux CentOS 7
8) How to Install Verilator in Windows 10
9) How to Run System Verilog DPI in Verilator
10) Delivering IP blocks via Verilator’s --protect-lib
11) Steps when protecting and compiling the verilated secret model in an IC Company 
12) Steps when simulating the encrypted RTL design and testbench using Verilator by the Customer
13) Verilator Language Limitations

wsnyder, 2024-01-24:
Great!  As to your example Verilator treats wreal as a "wire real".  It doesn't allow combinational assigns to it as your example does.