verilator/verilator #4857: Bug in Verilator that leads to a wrong runtime value, corrected by tracing

flaviens, 2024-01-24:
Hi there,

Thank you for all your contributions!

I stumbled into an occurrence that looks like an optimization bug that is corrected by tracing but by none of the `-fno*` flags.
[Here is a repository to help you reproduce](https://github.com/flaviens/verilator-b14).

I'm using `Verilator 5.021 devel rev v5.020-47-g6b8531f0a`, which does not show any suspicious warning.

Please let me know if you need any more information.

Thanks!
Flavien

wsnyder, 2024-01-24:
Don't turn off UNDRIVEN and you'll see the problem is your code ;)

%Warning-UNDRIVEN: top.sv:10:17: Bits of signal are not driven: 'out_data'[95:33,31:0]

flaviens, 2024-01-24:
Hi @wsnyder, thank you for having a look! The problem affects `out_data[32]`, which is exactly the one being driven.

wsnyder, 2024-01-24:
Ah, ok, the UNDRIVEN is likely causing the misoptimization will look deeper.

wsnyder, 2024-01-24:
@yTakatsukasa this passes with -fno-const-bit-op-tree, mind taking a look?

@flaviens if you can please try to use test_regress format like t_EXAMPLE.v, or like the below in the future, it'll help debug - and also you can then check against other simulators, thanks.

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Flavien Solt.
// SPDX-License-Identifier: CC0-1.0

module t(/*AUTOARG*/
   // Inputs
   clk
   );
   input clk;
   int   cyc;

   wire celloutsig_12z;
   wire celloutsig_15z;
   wire celloutsig_17z;
   wire celloutsig_4z;
   wire celloutsig_67z;
   wire celloutsig_9z;
   logic [95:0] in_data;

   // verilator lint_off UNDRIVEN
   wire [95:0] out_data;
   // verilator lint_on UNDRIVEN

   assign celloutsig_4z = ~(in_data[72] & in_data[43]);  // 1
   assign celloutsig_67z = | { in_data[64], celloutsig_12z }; // 0
   assign celloutsig_15z = in_data[43] & ~(celloutsig_4z); // 0
   assign celloutsig_9z = celloutsig_17z & ~(in_data[43]); // 00000000
   assign celloutsig_17z = celloutsig_15z & ~(in_data[43]);// 0
   assign celloutsig_12z = celloutsig_4z != celloutsig_9z; // 1
   assign out_data[32] = celloutsig_67z; // 1

   always_comb $display("celloutsig_12z = %x", celloutsig_12z);
   always_comb $display("out_data = %x", out_data);

   always @ (posedge clk) begin
      cyc <= cyc + 1;
      if (cyc == 0) begin
         in_data <= 96'h00000000_00000800_00000000;
      end
      else if (cyc == 1) begin
         if (out_data[32] != 1'b1) $stop;
         $write("*-* All Finished *-*\n");
         $finish;
      end
   end

endmodule
```

yTakatsukasa, 2024-01-25:
@wsnyder Thank you for narrowing down the issue.

Changing the following line
https://github.com/verilator/verilator/blob/3aece189acfe419297ac30237c4685c054682d0a/src/V3Const.cpp#L619

as below fixes the case. test_regress pass with the change, I'd like to check again before sending a PR.
```diff
-                    m_bitPolarities.emplace_back(ref, polarity, bitIdx);
+                    m_bitPolarities.emplace_back(ref, polarity  == ref.polarity(), bitIdx);
```

PR #4863 (2024-01-28): Fix #4857 consider NOT when checking EQ/NEQ under AND/OR tree
Fix #4857 
DFG opt is necessary to trigger the bug. So .pl that enabled DFG opt. is added.

Push a test, then push the fix for it later as usual.

ext tests pass.
commit e4a0e5aa (2024-01-28): Fix #4857 consider NOT when checking EQ/NEQ under AND/OR tree (#4863)
closed 2024-01-28 by yTakatsukasa