verilator/verilator #4859: False positive UNUSEDSIGNAL warning

leonardt, 2024-01-24:
Thanks for taking the time to report this.

Receiving an unused signal warning that appears to be a false positive, although I may be missing something.  The simulation test seems to work with the warning suppressed.  This is the reported error:

```
%Warning-UNUSEDSIGNAL: RegisterFile.v:63:13: Signal is not used: 'Register_inst0'
                                           : ... note: In instance 'RegisterFile'
   63 |   reg [3:0] Register_inst0;
      |             ^~~~~~~~~~~~~~
                       ... For warning description see https://verilator.org/warn/UNUSEDSIGNAL?v=5.021
                       ... Use "/* verilator lint_off UNUSEDSIGNAL */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Can you attach an example that shows the issue? 

```verilog
module RegisterFile(
  input  [1:0] raddr,
  input        wen,
  input  [1:0] waddr,
  input  [3:0] wdata,
  input        ren,
               CLK,
  output [3:0] rdata
);

  reg [3:0] Register_inst4;
  reg [3:0] Register_inst3;
  reg [3:0] Register_inst2;
  reg [3:0] Register_inst1;
  reg [3:0] _GEN;
  always_comb begin
    _GEN = Register_inst1;
    if (waddr == 2'h0 & wen)
      _GEN = wdata;
  end // always_comb
  reg [3:0] _GEN_0;
  always_comb begin
    _GEN_0 = 4'h0;
    if (raddr == 2'h0 & ren)
      _GEN_0 = Register_inst1;
  end // always_comb
  reg [3:0] _GEN_1;
  always_comb begin
    _GEN_1 = Register_inst2;
    if (waddr == 2'h1 & wen)
      _GEN_1 = wdata;
  end // always_comb
  reg [3:0] _GEN_2;
  always_comb begin
    _GEN_2 = _GEN_0;
    if (raddr == 2'h1 & ren)
      _GEN_2 = Register_inst2;
  end // always_comb
  reg [3:0] _GEN_3;
  always_comb begin
    _GEN_3 = Register_inst3;
    if (waddr == 2'h2 & wen)
      _GEN_3 = wdata;
  end // always_comb
  reg [3:0] _GEN_4;
  always_comb begin
    _GEN_4 = _GEN_2;
    if (raddr == 2'h2 & ren)
      _GEN_4 = Register_inst3;
  end // always_comb
  reg [3:0] _GEN_5;
  always_comb begin
    _GEN_5 = Register_inst4;
[... 276 lines cut ...]
    std::cerr << "Port     : " << "RegisterFile.rdata" << std::endl;
    
#if VM_TRACE
    // Dump one more timestep so we see the current values
    tracer->dump(main_time);
    tracer->close();
#endif
    
    exit(1);
}
    

#if VM_TRACE
  tracer->dump(main_time);
  tracer->close();
#endif
  

#if VM_COVERAGE
    write_coverage();
#endif
  top->final();

}
```

wsnyder, 2024-01-24:
I don't see where Register_inst0 is used, why do you think this is an incorrect warning?

leonardt, 2024-01-25:
You're correct, I think my build system may have been caching an older version of the input verilog or something.  Please ignore.

closed 2024-01-25 by leonardt