verilator/verilator #4860: tracep error: expected unqualified-id on VerilatedTraceSigType

ngraybeal, 2024-01-24:
It appears https://github.com/verilator/verilator/commit/95c4ade718a21d51c661cbe37b252924c0a822cd introduced a bug when [emitting](https://github.com/verilator/verilator/blob/23836ee8d00a4da961be6c761f7610a4cee74f36/src/V3EmitCImp.cpp#L743) the `tracep` for cases where tracing is supported by `VerilatedTraceSigType` isn't [set](https://github.com/verilator/verilator/blob/8a7f36c34b2a04a73aa2b472eb502d02d28c9a4a/src/V3Ast.h#L681) (ie `chandle`).

Verilator: 5.020
Compiler: Clang 17.0.4
OS: Rhel 8.7

Error Message
```
WARNING:root:Testbench__Trace__0__Slow.cpp:836:127: error: expected unqualified-id
WARNING:root:  836 |     tracep->declQuad(c+25,0,"foo_if",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);
WARNING:root:      |                                                                                                                               ^
```

As for a fix, I'm unsure if `chandle` needs it's own value in the [enum](https://github.com/verilator/verilator/blob/23836ee8d00a4da961be6c761f7610a4cee74f36/include/verilated_trace.h#L83), if there should be some `NONE` value in the enum, or a change is needed to the [function](https://github.com/verilator/verilator/blob/23836ee8d00a4da961be6c761f7610a4cee74f36/src/V3EmitCImp.cpp#L697).

Just for reference, using an older version of verilator 5.012, the chandle will appear as
```
    tracep->declQuad(c+77,"foo_if", false,-1, 63,0);
```

wsnyder, 2024-01-25:
I'd suggest we dump as a quad.  That way if the user can at least see the pointer propagate to other modules etc.

Could you modify one of the tests, e.g. test_regress/t/t_trace_complex.v to show the issue (trace a chandle), then perhaps look at fixing the code to make this a Quad?

ngraybeal, 2024-01-25:
The issue was easy to replicate, but I'm not exactly sure what you mean by make it a Quad. Are you saying add `Quad` as value to the [enum](https://github.com/verilator/verilator/blob/354a534d68ca6ba065fae7c7b72a49b3e15e7166/include/verilated_trace.h#L83) and then convert that to a `FST_VT_SV_LONGINT` in the if-else [block](https://github.com/verilator/verilator/blob/354a534d68ca6ba065fae7c7b72a49b3e15e7166/include/verilated_fst_c.cpp#L188)?

For what it's worth, i did set it as `/* CHANDLE:                   */ "DOUBLE",` in [traceSigType](https://github.com/verilator/verilator/blob/354a534d68ca6ba065fae7c7b72a49b3e15e7166/src/V3Ast.h#L687) and the test passed.

```
diff --git a/test_regress/t/t_trace_complex.v b/test_regress/t/t_trace_complex.v
index 313440c33..4a36d89bb 100644
--- a/test_regress/t/t_trace_complex.v
+++ b/test_regress/t/t_trace_complex.v
@@ -116,4 +116,5 @@ endmodule
 module p;
    parameter PARAM = 1;
    initial global_bit = 1;
+   chandle foo_if = null;
 endmodule
 ```
 
 Error:
 ```
 In file included from Vt_trace_complex__ALL.cpp:13:
Vt_trace_complex__Trace__0__Slow.cpp: In function ‘void Vt_trace_complex___024root__trace_init_sub__TOP__0(Vt_trace_complex___024root*, VerilatedVcd*)’:
Vt_trace_complex__Trace__0__Slow.cpp:72:127: error: expected unqualified-id before ‘,’ token
     tracep->declQuad(c+39,0,"foo_if",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);
                                                                                                                               ^
Vt_trace_complex__Trace__0__Slow.cpp:76:127: error: expected unqualified-id before ‘,’ token
     tracep->declQuad(c+39,0,"foo_if",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);
                                                                                                                               ^
Vt_trace_complex__Trace__0__Slow.cpp:80:127: error: expected unqualified-id before ‘,’ token
     tracep->declQuad(c+39,0,"foo_if",-1, VerilatedTraceSigDirection::NONE, VerilatedTraceSigKind::VAR, VerilatedTraceSigType::, false,-1, 63,0);
                                                                                                                               ^
```

wsnyder, 2024-01-25:
I meant add in V3Ast.h:   /* CHANDLE:                   */ "LONGINT",

I tried this and it worked, fix pushed, thanks for the good bug report.

closed 2024-01-25 by wsnyder

ngraybeal, 2024-01-25:
Thanks @wsnyder!