verilator/verilator #4861: warning: 'VerilatedScTraceBase::trace' hides overloaded virtual function [-Woverloaded-virtual]

augierg, 2024-01-25:
Environment
```
Ubuntu 22.04 LTS
Verilator 5.020 2024-01-01 rev v5.020 
SystemC 2.3.4
Clang++14 --std=c++17
```
Compilation in systemC model produces numerous warnings of the like below

```
In file included from /usr/local/share/verilator/include/verilated_vcd_sc.h:26:
/usr/local/share/verilator/include/verilated_sc_trace.h:158:10: warning: 'VerilatedScTraceBase::trace' hides overloaded virtual function [-Woverloaded-virtual]
    void trace(const unsigned int&, const std::string&, const char**) override {}
         ^
/usr/include/sysc/kernel/sc_object.h:83:18: note: hidden overloaded virtual function 'sc_core::sc_object::trace' declared here: different number of parameters (1 vs 3)
    virtual void trace( sc_trace_file* tf ) const;
                 ^

In file included from /usr/local/share/verilator/include/verilated_vcd_sc.h:26:
/usr/local/share/verilator/include/verilated_sc_trace.h:200:5: warning: 'VerilatedScTraceBase::trace' hides overloaded virtual function [-Woverloaded-virtual]
    DECL_TRACE_METHOD_A( sc_dt::sc_lv_base )
    ^
/usr/local/share/verilator/include/verilated_sc_trace.h:161:10: note: expanded from macro 'DECL_TRACE_METHOD_A'
    void trace(const tp& object, const std::string& name) override {}
         ^
/usr/include/sysc/kernel/sc_object.h:83:18: note: hidden overloaded virtual function 'sc_core::sc_object::trace' declared here: different number of parameters (1 vs 2)
    virtual void trace( sc_trace_file* tf ) const;
                 ^

```

These warnings did not occur with v5.018, and seem to be introduced by the commit https://github.com/verilator/verilator/commit/ab06ace1f651e23de4929f45ad161b38a2a6a583

wsnyder, 2024-01-25:
Please try the current git head version.

augierg, 2024-01-26:
Upon closer review of the recent commits, I see it was fixed in https://github.com/verilator/verilator/commit/f80c167d65b284ea03835d3a5a0e3584b7c95377
Confirmed locally with current head, these warnings are gone

closed 2024-01-26 by augierg