verilator/verilator #4862: fwrite skipped over loop in initial

jotego, 2024-01-27:
- $fwrite ignored during _for_ loop in an _initial_ statement
- Occurs on `Verilator 5.021 devel rev v5.020-68-gd6f8ccd` also seen on v5.018

```
module test;

integer f,cnt;

initial begin
	f=$fopen("zeros","wb");
	for(cnt=0;cnt<1024;cnt=cnt+4)
		$fwrite(f,"%u",32'd0);
	$fclose(f);
end

endmodule
```

This will create a 1024-byte long file using *iverilog*, but results in a 0-byte file with *verilator*

[test files](https://github.com/verilator/verilator/files/14072983/verilator_fwrite.zip)

wsnyder, 2024-01-27:
This is writing a null. Perhaps you use C/C++ and intended to write using %d instead?

In C, nulls terminate strings, so this prints nothing.  Verilator is using C++ strings now in many places, but not all, and it is still iterating based on C strings looking for nulls.  Perhaps you'd consider making a pull request to fix to use strings uniformly so nulls can be passed?

jotego, 2024-01-27:
Actually, `%u` is the only way to pass binary data to $fwrite and get it dumped as binary data. It should not be parsed as a string. iverilog, and some commercial tools I have tested, interpret `%u` this way. That is why `$fopen` uses `wb`, so the file is opened for binary data.

So maybe the bug is that verilator cannot parse `%u`?

wsnyder, 2024-01-27:
The bug is that FWRITE sees the null and stops writing as that ends the string as it thinks it is.

jotego, 2024-01-27:
> The bug is that FWRITE sees the null and stops writing as that ends the string as it thinks it is.

You are right. If I use `$fwrite(f,"%u",32'h1234);` the binary data $1234 gets written to the file correctly.

wsnyder, 2024-01-27:
All these need to be converted to use C++ std::strings, plus any functions they call:

```
include/verilated_funcs.h:128:extern void VL_WRITEF(const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:129:extern void VL_FWRITEF(IData fpi, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:131:extern IData VL_FSCANF_IX(IData fpi, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:132:extern IData VL_SSCANF_IIX(int lbits, IData ld, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:133:extern IData VL_SSCANF_IQX(int lbits, QData ld, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:134:extern IData VL_SSCANF_IWX(int lbits, WDataInP const lwp, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:136:extern void VL_SFORMAT_X(int obits, CData& destr, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:137:extern void VL_SFORMAT_X(int obits, SData& destr, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:138:extern void VL_SFORMAT_X(int obits, IData& destr, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:139:extern void VL_SFORMAT_X(int obits, QData& destr, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:140:extern void VL_SFORMAT_X(int obits, void* destp, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:2293:extern IData VL_SSCANF_INX(int lbits, const std::string& ld, const char* formatp, ...) VL_MT_SAFE;
include/verilated_funcs.h:2294:extern void VL_SFORMAT_X(int obits_ignored, std::string& output, const char* formatp,
include/verilated_funcs.h:2296:extern std::string VL_SFORMATF_NX(const char* formatp, ...) VL_MT_SAFE;
```

Is this something you'd be able to make a pull for perhaps?

wsnyder, 2024-01-27:
I suppose another uglier alternative is to not constify strings in V3Const if they contain nulls.

wsnyder, 2024-01-27:
I suppose another uglier alternative is to not constify strings in V3Const if they contain nulls.

(Actually that won't work in the general case as a user could have \000 in the string they pass directly to $fwrite.)

jotego, 2024-01-27:
> I suppose another uglier alternative is to not constify strings in V3Const if they contain nulls.

I tried passing a variable like this:

```
integer zero=0;

initial begin
    $fwrite(f,"%u",zero)
...
```

But that did not work either

wsnyder, 2024-01-27:
Let me take a look this weekend, if it doesn't spread to too many functions I'll commit a fix.

closed 2024-01-28 by wsnyder