verilator/verilator #4864: Runtime value issue under some specific conditions

flaviens, 2024-01-28:
Hi there!

Thank you as always for your contributions and for taking the time for reading this issue!
[In this repository, I reduced for you a design in which I had runtime value issues](https://github.com/flaviens/verilator-b15). It is not in test_regress format (I'm not very familiar with it) but it seems to be easily transferable for someone knowledgeable.

The interesting thing is that the bug is extremely unstable: most minor alterations, even outside of the path from inputs to outputs, will clear the bug.

I tested it with `Verilator 5.021 devel rev v5.020-69-ge4a0e5aad`.

A timeline of what happens:
1. The little `always_ff` puts `'0` into `sig_b`.
2. The  input changes (which should not influence the output, but ignoring this step will clear the bug).
3. The bigger `always_ff` transmits the zero-value from `sig_b` through `sig_c` into `sig_e`.
So eventually, the output value should be `0`.
However, with some optimizations enabled (in some non-reduced case, I think it was the opposite), such as `-fno-subst -fno-subst-const`, the output value is `1`.

Please let me know if you have any question or require further info!

Kind regards,
Flavien

wsnyder, 2024-01-28:
@yTakatsukasa this also goes away with only -fno-const-bit-op-tree

closed 2024-01-29 by yTakatsukasa

flaviens, 2024-01-29:
Crazy, thank you for the fix! :heart:

PR #4867 (2024-01-29): Fix #4864 of wrong runtime result
Fix #4864 

EQ/NE under SHIFTR was wrongly target of the optimization.

EQ/NE under AND/OR tree such as `a[0] & (a[2:1] == 2'b10)` can be optimized to  `a[2:0] == 3'b101`.

But `(a[2:1] == 2'b10) >> 1` is not the case, but was wrongly shifted.

Push a test to reproduce the bug, then will push the fix later.

ext tests pass after manuallytweaking C++14 in the following line to be compatible with string literal.
https://github.com/ZipCPU/wbuart32/blob/d8ea86d1b8ef47d0b1c13f291a958b6aac9229f9/bench/cpp/Makefile#L74
commit 999c9ae2 (2024-01-29): Fix #4864 of wrong runtime result (#4867)

* Fix #4864. Ignore if EQ/NE is under SHIFTR.