verilator/verilator #4871: Fatal error on split variable when tracing is enabled

adrienlemasle, 2024-02-02:
Hi there.

The following example code:
```
module test
  (
   input      [10:0] [31:0]     register_in,
    output     [10:0] [31:0]     register
   );
   assign register = register_in;

endmodule

module verilator_split;

   logic [10:0] [31:0] register_in /*verilator split_var*/;

   test test
     (.register_in(register_in),
      .register());
endmodule
```

Leads to the following compilation error when tracing is enabled:
```
verilator --cc -O3 --trace-fst verilator_split.sv --top-module verilator_split --exe verilator_split.cpp
%Error: Internal Error: verilator_split.sv:15:20: ../V3SplitVar.cpp:1124: not enough split variables
                                                : ... note: In instance 'verilator_split'
   15 |      (.register_in(register_in),
      |                    ^~~~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

Without the `--trace-fst` flag it works properly.

Quickly going through the code it seems to be related to the fact that some undriven signals  don't get optimised out when tracing is enabled and I'm therefore wondering if this assert in V3SplitVar.cpp:1124 is correct in this case?

Thanks

Adrien

wsnyder, 2024-02-04:
split_var does not support multiple dimensional arrays at present.  It should give a warning but that seems missing.

Note also doing a split on a primary input won't have any real usefulness, as the input itself cannot be split, and the use of split_var to avoid UNOPTFLAT is not needed because UNOPTFLAT only can occur related to driving outputs.

adrienlemasle, 2024-02-04:
This is an over-simplified example here to trigger the crash but in this
particular case I was using split_var to prevent errors/warnings about
driving the same signal in combinatorial and sequential logic.

wsnyder, 2024-02-14:
If you'd like to contribute a pull request to fix splitting multi-dim arrays it would be appreciated.  Alas our backlog of issues is pretty large, so it's unlikely this is something I myself will be getting to soon.

wsnyder, 2025-08-21:
Works on master, fixed somewhere in the many split improvements over the last year.

closed 2025-08-21 by wsnyder