verilator/verilator #4872: Segmentation Fault trying to load VCD file

jordankrim, 2024-02-02:
I am running 2 Verilated sims under Verilator 5.021 on Ubuntu (22.04.3).   The first one is a fairly small standalone one and the VCD is generated and loads fine into gtkwave.  The second one is more complicated as it is inside a "chip".  This one generates a VCD but there are errors when loading it into gtkwave and it soon crashes with a segmentation fault.  I tried to convert the "bad" VCD to an FST file and it loads fine into gtkwave, but the entire structure seems corrupted.   I cannot find the specific component or even the "system" component that should be there.

When I run the same "chip" on RHEL8 under Verilator 4.226, the VCD is generated and loads into gtkwave fine.  When I drop back to Verilator 4.226 on Ubuntu and create the VCD for the "chip" and load it into gtkwave, it loads fine and has the correct structure.

The only change I made is to add the --timing parameter for 5.021 (and remove it when using 4.226).  Other than that, the verilated compile for both versions should be the same.

jordankrim, 2024-02-04:
I have been on the gtkwave forum and this is what tybell says:

That VCD file looks massively corrupted. It fails on my gtkwave-1.3.119 as well. Something is wrong with your verilator.

504:/tmp> gtkwave jordans_part.vcd

GTKWave Analyzer v3.3.119 (w)1999-2024 BSI

Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire'
$VAR encountered after $ENDDEFINITIONS near byte 2227. VCD is malformed, exiting.

00000000: 24 76 65 72 73 69 6f 6e 20 47 65 6e 65 72 61 74 $version Generat
00000010: 65 64 20 62 79 20 56 65 72 69 6c 61 74 65 64 56 ed by VerilatedV
00000020: 63 64 20 24 65 6e 64 0a 24 74 69 6d 65 73 63 61 cd $end.$timesca
00000030: 6c 65 20 31 70 73 20 24 65 6e 64 0a 20 24 73 63 le 1ps $end. $sc
00000040: 6f 70 65 20 6d 6f 64 75 6c 65 20 54 4f 50 20 24 ope module TOP $
00000050: 65 6e 64 0a 20 20 24 73 63 6f 70 65 20 6d 6f 64 end.  $scope mod
00000060: 75 6c 65 20 6c 6c 6b 69 5f 70 6b 67 20 24 65 6e ule llki_pkg $en
00000070: 64 0a 20 20 20 24 76 61 72 20 77 69 72 65 20 06 d.   $var wire .
00000080: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000090: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
000000a0: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
000000b0: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
000000c0: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
000000d0: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
000000e0: 20 06 20 20 20 24 76 61 72 20 77 69 72 65 20 06  .   $var wire .
000000f0: 20 20 20 24 76 61 72 20 77 69 72 65 20 06 20 20    $var wire .  
00000100: 20 24 76 61 72 20 77 69 72 65 20 06 20 20 20 24  $var wire .   $
00000110: 76 61 72 20 77 69 72 65 20 06 20 20 20 24 76 61 var wire .   $va
00000120: 72 20 77 69 72 65 20 06 20 20 20 24 76 61 72 20 r wire .   $var 
00000130: 77 69 72 65 20 06 20 20 20 24 76 61 72 20 77 69 wire .   $var wi
00000140: 72 65 20 06 20 20 20 24 76 61 72 20 77 69 72 65 re .   $var wire
00000150: 20 38 20 31 7c 25 20 4c 4c 4b 49 5f 4d 49 44 5f  8 1|% LLKI_MID_

So it seems like a verilator problem.   Comments?

wsnyder, 2024-02-04:
Are you sure your simulation is exiting cleanly, that is isn't aborting and writing a partial file?
Does FST dumping work correctly?
Is the VCD file obviously corrupt starting at some point?
What does vcddiff say comparing the VCDs from the two versions?
Are you using --threads?  Try without.

wsnyder, 2024-02-04:
Also make sure you are using the git master version, so we can debug the latest please.

jordankrim, 2024-02-04:
The simulation is exiting cleanly.
I don't know how to do an FST dump.   If I run my verilated executable --help, it only shows an option to dump a VCD.  Do you know of a way to be able to get an FST file?  Also, when I converted the VCD to FST using vcd2fst, gtkwave loaded the FST file with no problem, but when I looked at the structure from 5.021, it didn't match the "good" VCD from 4.226 and worse yet, it didn't have anything useful to look at, specifically, the particular design I am looking at under a "system" tree which is completely missing.
I don't know enough about VCD file structure to know where it is corrupted.  The gtkwave puts this out:
Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire'
$VAR encountered after $ENDDEFINITIONS near byte 2227. VCD is malformed, exiting.
I have not run vcddiff.  I will have to try that.
I don't know if I am using --threads (with either or both 4.226 and 5.021).   Where is it specified and how do I disable --threads?

wsnyder, 2024-02-04:
>I don't know how to do an FST dump. 

    --trace-fst                 Enable FST waveform creation

      https://veripool.org/guide/latest/faq.html?highlight=trace%20fst#how-do-i-generate-fst-waveforms-traces-in-c-or-systemc

>I don't know enough about VCD file structure to know where it is corrupted. The gtkwave puts this out: Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire' $VAR encountered after $ENDDEFINITIONS near byte 2227. VCD is malformed, exiting.

So when you look at the file in an editor around that byte there's nothing unusual?  Is there a $var after $enddefinitions?

>I don't know if I am using --threads (with either or both 4.226 and 5.021). Where is it specified and how do I disable --threads?

If you aren't putting --threads on the verilator command line you are not using it.

Is it possible to post a self-contained simple example showing the problem?

jordankrim, 2024-02-04:
When I tried to do an FST with --trace-fst, I got the following compile error:

home/jkrim/cep_v4_4_new3_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/emulator.cc:295:17: error: cannot initialize a parameter of type 'VerilatedFstC *' with an rvalue of type 'std::unique_ptr<VerilatedVcdC>::pointer' (aka 'VerilatedVcdC *')
    tile->trace(tfp.get(), 99);  // Trace 99 levels of hierarchy
                ^~~~~~~~~
/home/jkrim/cep_v4_4_new3_test_verilator/CEP/sims/verilator/generated-src/chipyard.TestHarness.CEPVerilatorRocketConfig/chipyard.TestHarness.CEPVerilatorRocketConfig/VTestHarness.h:76:31: note: passing argument to parameter 'tfp' here
    void trace(VerilatedFstC* tfp, int levels, int options = 0);                             
1 warning and 1 error generated.
make[1]: *** [VTestHarness.mk:85: emulator.o] Error 1

For the error when loading VCD into gtkwaves:

Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire'

Is that 143 decimal or 143 hex?   Using an editor, I do not see a $var after the enddefinitions in the VCD file.

I posted a partial file (~<25 MB) of the VCD file (which is ~335MB) on the gtkwave forum as that is the largest that can be posted.  Should I post that here?   Otherwise, I will have to investigate how to create a smaller postable VCD.

wsnyder, 2024-02-04:
You need to change the include and class name as described in the mentioned link: https://veripool.org/guide/latest/faq.html?highlight=trace%20fst#how-do-i-generate-fst-waveforms-traces-in-c-or-systemc

What we really want is an example that has sources, a file that's corrupted might give a  hint but is unlikely by itself to be enough to debug the problem.

jordankrim, 2024-02-04:
I did find the magic to switch to the FST file.  It is built-in to my make files via a parm (VERILATOR_FST_MODE) and the FST built and run under 5.021 loads and has the correct structure (just like the VCD file that works under 4.226).  So, somewhat of a workaround to the VCD problem?

If you haven't figured it out by now, this is the CEP (Common Evaluation Platform) which is very large, so coming up with an "example" seems not possible. I attempted to run vcddiff on the bad vs. good VCD.   It was taking too long so killed it.  I head'ed some of the lines of each file and then did a tkdiff.   They are vastly different from the get go.   

The bad one starts this way:

$version Generated by VerilatedVcd $end
$timescale 1ps $end
 $scope module TOP $end
  $scope module llki_pkg $end
   $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire 8 1|% LLKI_MID_C2LOADKEYREQ [7:0] $end
   $var wire 8 nz% LLKI_MID_C2CLEARKEYREQ [7:0] $end
   $var wire 8 pz% LLKI_MID_C2KEYSTATUSREQ [7:0] $end
   $var wire 8 U.& LLKI_MID_C2LOADKEYACK [7:0] $end
   $var wire 8 qz% LLKI_MID_C2CLEARKEYACK [7:0] $end
   $var wire 8 V.& LLKI_MID_C2KEYSTATUSRESP [7:0] $end
   $var wire 8 W.& LLKI_MID_C2ERRORRESP [7:0] $end
   $var wire 8 X.& LLKI_MID_KLLOADKEYREQ [7:0] $end
   $var wire 8 rz% LLKI_MID_KLCLEARKEYREQ [7:0] $end
   $var wire 8 Y.& LLKI_MID_KLKEYSTATUSREQ [7:0] $end
   $var wire 8 Z.& LLKI_MID_KLLOADKEYACK [7:0] $end
   $var wire 8 [.& LLKI_MID_KLCLEARKEYACK [7:0] $end
   $var wire 8 \.& LLKI_MID_KLKEYSTATUSRESP [7:0] $end
   $var wire 8 W.& LLKI_MID_KLERRORRESP [7:0] $end
   $var wire 8 1|% LLKI_STATUS_GOOD [7:0] $end
   $var wire 8 nz% LLKI_STATUS_KEY_PRESENT [7:0] $end
   $var wire 8 pz% LLKI_STATUS_KEY_NOT_PRESENT [7:0] $end
   $var wire 8 tz% LLKI_STATUS_BAD_MSG_ID [7:0] $end
   $var wire 8 ].& LLKI_STATUS_BAD_MSG_LEN [7:0] $end
   $var wire 8 ^.& LLKI_STATUS_KEY_INDEX_EXCEED [7:0] $end
   $var wire 8 _.& LLKI_STATUS_KEY_INDEX_INVALID [7:0] $end
   $var wire 8 `.& LLKI_STATUS_BAD_POINTER_PAIR [7:0] $end
   $var wire 8 a.& LLKI_STATUS_BAD_CORE_INDEX [7:0] $end
   $var wire 8 b.& LLKI_STATUS_KL_REQ_BAD_MSG_ID [7:0] $end
   $var wire 8 c.& LLKI_STATUS_KL_REQ_BAD_MSG_LEN [7:0] $end
   $var wire 8 d.& LLKI_STATUS_KL_RESP_BAD_MSG_ID [7:0] $end
   $var wire 8 e.& LLKI_STATUS_KL_TILELINK_ERROR [7:0] $end
   $var wire 8 f.& LLKI_STATUS_KL_LOSS_OF_SYNC [7:0] $end
   $var wire 8 g.& LLKI_STATUS_KL_BAD_KEY_LEN [7:0] $end
   $var wire 8 h.& LLKI_STATUS_KL_KEY_OVERWRITE [7:0] $end
   $var wire 8 {|% LLKI_STATUS_UNKNOWN_ERROR [7:0] $end
   $var wire 8 Z.& SROT_WAIT_STATE_COUNTER_INIT [7:0] $end
   $var wire 8 Z.& MOCKTSS_WAIT_STATE_COUNTER_INIT [7:0] $end
   $var wire    $var wire ﾨ i.& AES_MOCK_TSS_KEY_WORDS[0] [   $var wire ﾨ k.& AES_MOCK_TSS_KEY_WORDS[1] [   $var wire    $var wire ﾨ m.& DES3_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ o.& SHA256_MOCK_TSS_KEY_WORDS[0] [   $var wire ﾨ q.& SHA256_MOCK_TSS_KEY_WORDS[1] [   $var wire ﾨ s.& SHA256_MOCK_TSS_KEY_WORDS[2] [   $var wire ﾨ u.& SHA256_MOCK_TSS_KEY_WORDS[3] [   $var wire ﾨ w.& SHA256_MOCK_TSS_KEY_WORDS[4] [   $var wire ﾨ y.& SHA256_MOCK_TSS_KEY_WORDS[5] [   $var wire ﾨ {.& SHA256_MOCK_TSS_KEY_WORDS[6] [   $var wire ﾨ }.& SHA256_MOCK_TSS_KEY_WORDS[7] [   $var wire    $var wire ﾨ !/& MD5_MOCK_TSS_KEY_WORDS[0] [   $var wire ﾨ #/& MD5_MOCK_TSS_KEY_WORDS[1] [   $var wire ﾨ %/& MD5_MOCK_TSS_KEY_WORDS[2] [   $var wire ﾨ '/& MD5_MOCK_TSS_KEY_WORDS[3] [   $var wire ﾨ )/& MD5_MOCK_TSS_KEY_WORDS[4] [   $var wire ﾨ +/& MD5_MOCK_TSS_KEY_WORDS[5] [   $var wire ﾨ -/& MD5_MOCK_TSS_KEY_WORDS[6] [   $var wire ﾨ //& MD5_MOCK_TSS_KEY_WORDS[7] [   $var wire    $var wire ﾨ 1/& RSA_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ 3/& IIR_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ 5/& FIR_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ 7/& DFT_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ 9/& IDFT_MOCK_TSS_KEY_WORDS[0] [   $var wire    $var wire ﾨ ;/& GPS_MOCK_TSS_KEY_WORDS[0] [   $var wire ﾨ =/& GPS_MOCK_TSS_KEY_WORDS[1] [   $var wire ﾨ ?/& GPS_MOCK_TSS_KEY_WORDS[2] [   $var wire ﾨ A/& GPS_MOCK_TSS_KEY_WORDS[3] [   $var wire ﾨ C/& GPS_MOCK_TSS_KEY_WORDS[4] [  $upscope $end
  $scope module top_pkg $end
   $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire    $var wire   $upscope $end
  $scope module tlul_pkg $end
   $var wire    $var wire 1   $var wire 1  $upscope $end
  $var wire 1 zf% clock $end

The good one starts this way:

$version Generated by VerilatedVcd $end
$date Sun Feb  4 09:12:17 2024 $end
$timescale 1ps $end

 $scope module TOP $end
  $var wire  1 >6& clock $end
  $var wire  1 @6& io_success $end
  $var wire  1 ?6& reset $end
  $scope module TestHarness $end
   $var wire  1 >6& SimJTAG_clock $end
   $var wire  1 $ SimJTAG_enable $end
   $var wire 32 K! SimJTAG_exit [31:0] $end
   $var wire  1 C6& SimJTAG_init_done $end
   $var wire  1 G! SimJTAG_jtag_TCK $end
   $var wire  1 I! SimJTAG_jtag_TDI $end
   $var wire  1 A6& SimJTAG_jtag_TDO_data $end
   $var wire  1 <7& SimJTAG_jtag_TDO_driven $end
   $var wire  1 H! SimJTAG_jtag_TMS $end
[... cut ...]
   $var wire  1 S{% uart_sim_0_io_uart_rxd $end
   $var wire  1 4) uart_sim_0_io_uart_txd $end
   $var wire  1 ?6& uart_sim_0_reset $end
   $scope module SimJTAG $end

If the bad VCD is converted to an FST (vcd2fst), it loads without errors and crashing (seg fault), but I cannot find under TOP the TestHarness tree (see line 9 in above good VCD) which then leads to the piece I am trying to look at in the waves.  So, basically the bad VCD and bad FST do not contain the correct structure of my design.

wsnyder, 2024-02-04:
>$var wire � $var wire � $var wire � $var wire

Literally like that with a $var wire some control characters than another $var?  Can you check that the control characters aren't messing up the formatting?  The code that prints that cannot do that, so the buffer or file is getting corrupted otherwise.

Double check you using the git master version. Then Verilate with "--runtime-debug" and see if you get errors.

jordankrim, 2024-02-05:
I did this:
cd ~/
git clone https://github.com/verilator/verilator.git verilator-master
cd verialtor-master
git pull
autoconf
configure CXX=clang++
make 
make test
Seemed to compile OK and the Tests passed.
Then I added the --runtime-debug parameter to the invocation of verilator to create the verilated code. If took many hours to compile, but seemed to completely successfully.   When I run my sim, however, I get this:

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 46031
==44662==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
/usr/bin/../lib/gcc/x86_64-linux-gnu/12/../../../../include/c++/12/debug/deque:364:
In function:
    std::deque::reference std::deque<unsigned int>::front() [_Tp = unsigned 
    int, _Allocator = std::allocator<unsigned int>]

Error: attempt to access an element in an empty container.

Objects involved in the operation:
    sequence "this" @ 0x61b000000bf8 {
      type = std::debug::deque<unsigned int, std::allocator<unsigned int> >;
    }
[UART] UART0 is here (stdin/stdout).

Normally, without the --runtime-debug parm, it says "Test Passed" at the end.   When I load the VCD into gtkwave it is the same result as before:

Near byte 143, $VAR parse error encountered with 'TOP.llki_pkg.wire'
$VAR encountered after $ENDDEFINITIONS near byte 2227.  VCD is malformed, exiting.

Again, looking at the VCD in vi, line 5 in the file is:
$var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire ^F   $var wire 8 1|% LLKI_MID_C2LOADKEYREQ [7:0] $end

wsnyder, 2024-02-05:
I don't think the std::deque warning is related.

You'll need to debug the details or provide a self-contained example code.  To debug, add prints to e.g. here https://github.com/verilator/verilator/blob/7ada5195d8beae1cbf24e3bd5ae3bfc5d389ce7a/include/verilated_vcd_c.cpp#L414
where it creates $var, and inside printStr around where the buffer is being built, and where written.  I speculate the buffer is overwriting but cannot explain how/why as the code looks ok.

jordankrim, 2024-02-05:
I see that line in the code, but since it takes several hours to compile and I am not sure exactly what to print there, can you please be more specific on what needs to be printed and where.

wsnyder, 2024-02-05:
It's hard to know for sure, it'll take many passes, and debugging by text is crazy slow. Can you look at the code yourself and debug, or find someone at your org/school that knows how to debug C code that can help you?  Alternatively please figure out how to send a test case.

jordankrim, 2024-02-05:
I added a couple print statements and they seemed to work in the make test, but when I tried to verilate my code for some strange reason, now my VM is crashing and becoming read-only, so I need to find out what I messed up, but in the meantime, I am wondering if I still need the --runtime-debug.

wsnyder, 2024-02-05:
You can remove --runtime-debug as it didn't point to the problem.  Note it does also turn on gdb debug, so if you later want to run a debugger you'll need --runtime-debug or at least the '-CFLAGS -ggdb -LDFLAGS --ggdb" flags that --runtme-debug sets (see docs).

jordankrim, 2024-02-05:
Here is my first attempt at debug (I had to truncate it at 25 MB to upload because it is really 41 MB):
[rmbt.log.tar.gz](https://github.com/verilator/verilator/files/14169285/rmbt.log.tar.gz)

wsnyder, 2024-02-06:
That file can't be decompressed.

jordankrim, 2024-02-06:
I was able to untar this using 7-zip manager on Windows after pasting this URL into my browser:
[rmbt_log.tar.gz](https://github.com/verilator/verilator/files/14181883/rmbt_log.tar.gz)

wsnyder, 2024-02-06:
The file was misnamed, you didn't compress it so it shouldn't be named .gz, hence I was using the wrong command.

Anyhow,

```
wirep = 'wire', bits = '32', array = '0', arraynum = '-1', bussed = '1', msb = '31', lsb = '0', hierarchicalName = 'TOP llki_pkg SROT_CTRLSTS_ADDR' vcdCode[0] = 'L' vcdCode[1] = '.' vcdCode[2] = '&$
ASSEMBLED decl = '$var wire ^F'
```

Please add these additional prints

```
    std::string decl = "$var ";
    decl += wirep;
    decl += ' ';
    std::cout<<"Pre to_string(bits) " << decl.c_str() << "\n";
    decl += std::to_string(bits);
    decl += ' ';
    std::cout<<"Post to_string(bits) " << decl.c_str() << "\n";
```

If the output is other than that I expect e.g. roughly

```
Pre to_string)bits) = '$var wire '
Post to_string(bits) = '$var wire ^F'

```
than please add more prints after each string addition to see where it goes bad.

jordankrim, 2024-02-07:
It is unclear to me what you mean by "go bad".  Here are the first few lines of output (note the little box below in the editor shows as <ack>):

Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire '
ASSEMBLED decl = '$var wire '
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire 8 '
ASSEMBLED decl = '$var wire 8  1|% LLKI_MID_C2LOADKEYREQ [7:0] $end
'
Pre to_string(bits) = '$var wire '
Post to_string(bits) = '$var wire 8 '
ASSEMBLED decl = '$var wire 8  nz% LLKI_MID_C2CLEARKEYREQ [7:0] $end
'

wsnyder, 2024-02-07:
Thanks, this matches what I expected to see.

Post to_string(bits) = '$var wire �'

The C++ code printing that control character followed by null is:

    decl += std::to_string(bits);

Your compiler is not correctly making this code, it must add a string representation of a number here e.g. "4".  Unfortunately, with a broken compiler there's not really anything else I can do here.  You'll need to switch to Ubuntu, or otherwise use a working compiler.

closed 2024-02-07 by wsnyder

jordankrim, 2024-02-07:
I am on Ubuntu and I thought I had a working compiler, as Verilator (5.021) compiles and passes the tests.

wsnyder, 2024-02-07:
>I thought I had a working compiler, as Verilator (5.021) compiles and passes the tests.

Are you running Verilator completely standalone, and likewise compiling completely standalone, that is with no other scripts/frameworks?  And your environment has no strange PATH or binaries, etc?  Perhaps you are running it under some framework that is breaking the compiler?

Or there's some complicated corruption issue in your code that is corrupting the std::to_string - which is very very unlikely.

Or, your system is physically broken (hardware or software OS image wise)  - did you try on other machines?

Anyhow feel free to poke at it or call in experts in your org to debug further if you want. But anyhow I don't see how I can offer any additional assistance - my conclusion is the standard library is not working correctly for only you, and so it's not a verilator bug.

jordankrim, 2024-02-07:
As you pointed out several posts ago, I had a conda environment that was apparently interfering with the Verilator stand-alone compile.  I seemed to have that resolved because when I dumped that env and compiled Verilator stand-alone, it passed all the tests (make test).  The verilated code needs to be done in the cond env supplied with the CEP.  With the exact same input as far as I can tell, Verilator 4.226 works (good VCD) and Verilator 5.021 does not (bad VCD).  The Verilator 4.226 was supplied with this conda env, so I have no idea if files were changed or not.  I think it will be difficult if not impossible to tell if something was changed in the Verilator 4.226 files in that conda env since diffing them against the Verilator 5.021 will probably yield a lot of changes as I know the code has moved on since 4.226.   All I can really see is that when I run the 4.226, the first part of the log does not contain any of these \<ack\> chars that I recall.

wsnyder, 2024-02-07:
The VCD code had many many changes between those versions, including adding use of the function that appears broken, so it's not surprising it behaves differently.  You could change the code that is showing the problem, but I wouldn't advise that the base problem whatever it is, will likely show up in other places that will be even harder to debug.  It's quite possible that 4.226 is also broken in some way you just haven't come across yet. For example, if it's memory corruption due to your wrapper code or something in your model, that explains the Verilator tests passing, and all your other symptoms, and is also something I can't really help you debug.  Again I'd advise asking for an expert at your end to help you if you want to proceed further.

jordankrim, 2024-02-07:
I know this isn't maybe the "real" fix, but I changed the verialted_vcd_c.cpp code and got it to work with no errors loading the VCD and the waveforms are as expected.  I replaced all the std::to_string with a call to my_to_string which is this:

std::string my_to_string(int abc) {
    std::stringstream stream;
    std::string str;    

    stream << abc;
    stream >> str;
    return(str);
}