verilator/verilator #4877: Allow setting top level parameters at runtime

Timmmm, 2024-02-05:
Questa has a feature where at compile time you can "float" module parameters, which then allows you to set them at runtime. This is useful for verifying parameters since you can run a load of tests with different values without having to recompile the simulator. The Questa options are `-floatparams` for `vopt` and then `-Gparam_name=param_value` for `vlog`.

It would be great if Verilator could support this too. Actually I found issue #1045 which sounds like exactly this, however all the links are dead and I can't find any reference to the code snippets (e.g. `+parameter+`) anywhere in the Verilator source, even with `git log -S '+parameter+' --oneline`. Does this feature exist?

Guiltybyte, 2024-02-05:
This would be very cool.

I decided to run the following simple example through verilator and inspect the outputted V*.cpp file to see if any concept of the top level parameters survive the verilation process.

```
module example_model #(
  parameter int WIDTH_PARAM = 8, 
  parameter bit CONDITIONAL_CODE_GEN_PARAM = 0
  )(
  input  logic clk,
  input  logic[WIDTH_PARAM-1:0] in1,
  input  logic[WIDTH_PARAM-1:0] in2,
  output logic[WIDTH_PARAM:0] out
  );

  generate
  if(CONDITIONAL_CODE_GEN_PARAM) begin: registered_case
    always_ff @(posedge clk) begin
      out <= in1 + in2;
    end
  end: registered_case
  else begin: combinatorial_case
    assign out = in1 + in2;
  end: combinatorial_case
  endgenerate

endmodule: example_model
```

Running, `verilator --binary *sv` produces the obj_dir, and a quick `grep PARAM obj_dir/` shows that at the very least the names are not preserved.

Inspecting `obj_dir/Vexample_model.h`  you can find the port definitions:

[image]

Which are using literal values for the port width rather than some variable.

TLDR; Pretty much confirmed that in a default verilator run, params are sorted during verilation, so no "runtime"
config would be possible.

I have a hunch that supporting these floating params for width params will be a lot easier than params which result in conditional elaboration of code.

wallento, 2024-02-05:
The linked issue found its way into the code base as 482bdab0e0abdf9724e0558c99f6c242eea24381 as `-pvalue` or `-G`.
It is indeed for the verilation phase and neither compilation time nor runtime (which both could be options).
The feature seems useful, but not trivial to implement. It would probably require a more in-depth analysis of the different paths that parameters manifest in the emitted code.

wsnyder, 2024-02-06:
Parameters must be done at elaboration time.  Perhaps Questa delays part of that process so it can do this feature closer to runtime.  Regardless this process is already extremely complex and I don't see any realistic way to consider Verilator taking on this feature.

Note if you have e.g. numbers that you want at runtime I'd encourage having "tie" input wires to the model instead of parameters. If you make a wrapper around your model and set the inputs to constants, good synthesis tools should flatten such inputs out when synthesizing the wrapper.

closed 2024-02-06 by wsnyder

Timmmm, 2024-02-06:
Sorry I thought I put a comment here... But anyway could Verilator do this by allowing you to automatically convert a parameter to a port?

I agree with what you said about using ports instead of parameters but the design guys didn't want to do that. I think out of fear that the synthesis tools weren't actually that smart but I might have misremembered. I'll double check.