verilator/verilator #4882: adding support for different VCD format

franout, 2024-02-06:
Hello guys,

i was using verilator for dumping some traces (VCDs).

i found out that is does not support the following VCD format 

```
....
$var port [31:0] <0  a $end
$var port 1 <1  b $end
....

#1
0<1
00000000000000000000000000<0
```

I don't know if it is an accepted widely used standards, we (at University) are using this format (from another logic simulator) and we developed a toolchain for evaluating the toggle activity/coverage of signals.

Moreover, another format used is the following (mainly used from fault simulators, as far as i knwo) is the eVCD (extended VCD):
Signals in the upscopes are declared as the previous format.
```
pD 6 0 <0
pDDDNNDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDNDDDDDDDDDDDDDDDDDDDDDDDDDDDD 6666666666666666666666666666666666666666666666666666666666666666 0006600000000000000000000000000000060000000000000000000000000000 <4
pFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFDDDUDDDDDDDDDDDDDDDDDDDDDDDDDDDD 0000000000000000000000000000000066606666666666666666666666666666 0000000000000000000000000000000000060000000000000000000000000000 <5
pD 6 0 <6
pLLLLLLLH 66666660 00000006 <9
```
Where, for example:
- **p** represents port change 
- **H** output high 
- **0** zero drive strength (no drive)  to **7** drive strength (pull) as verilog implements
- **U**  Input high(up)

If i am not wrong, verilator supports signals strength. I do not know if it could be interesting to implement a dump of an eVCD.

I already implemented a custom ```verilated_vcd_c.cpp``` for dumping the first format, if you need it.

Cheers 😄 😈

wsnyder, 2024-02-06:
"$var port" is also part of the eVCD format you later describe.

If you'd like to make a pull to implement e.g. --trace-evcd, and $dumports along with tests that would be likely accepted.  I'd suggest modify verilated_vcd_c to make this an option, as opposed to copying the code as there will otherwise be a lot of duplication.

Note Verilator determines all strengths at compile time.

wsnyder, 2024-02-06:
Also note we'll need a pull to make "vcddiff" support eVCD format, not sure if it does already or not.

franout, 2024-02-08:
I am not a good C++ programmer but i wil try my best to follow the best practices and verilator's code style.

Regarding "vcddiff" I would say i can start by making a pull request to that repo.

for recap:
- [ ] support of "--trace-evcd" as option 
- [ ] support of $dumpports 
- [ ] related tests

wsnyder, 2024-02-08:
Yes, a pull for vcddiff would be a great first step.  I see it does have "port" in it's sources - so maybe it'll just work!

wsnyder, 2024-09-08:
@franout you fixed vcddiff, thanks, but I'm wondering if you plan on finishing the Verilator EVCD support, or if we can close this issue?

wsnyder, 2024-09-10:
Thanks for the update, no rush, just wanted to know if this should stay open, which it will.

franout, 2024-10-27:
@wsnyder, a couple of questions:
- all strengths are determined at compile time; there is no way to access their value from verilator_vcd_c.cpp. I did not check very deeply.
- do you prefer a class extending the VerilatorVcd/C (i.e., VerilatorEvcd/C) or a boolean value in the class VerilatorVcd? I think the second would make the code "cleaner".
- I am a little bit lost for introducing the dumpports (when parsing sv or v files) support, as far I understand, now it is treated like it would be dumpvars, but it should activate the related boolean value of -trace-evcd when it is parsed? Am I getting it correctly? Could you point me to the right file?
thanks

wsnyder, 2024-10-27:
Correct, strengths are static and not accessible currently.

I'd prefer a Boolean in VerilatorVcd.

Perhaps we have --trace-evcd which sets a bool to later modify --trace-vcd to mean evcd, but also if there's a $dumpports we set that same flag (so a --trace and $dumpports will do eVCD, or a --trace-evcd and $dumpvars will do eVCD), and otherwise $dumpports is treated as $dumpvars?