verilator/verilator #4884: coverage question

develone, 2024-02-06:
Testing on a 8Gb raspberry pi 4. 2048 swap space my repo "https://github.com/develone/verilator-tests.git"
The Makefile from the verilator/examples/make_trace_c started with the following option enable: VERILATOR_FLAGS += --coverage 
I tested adding several Verilog files (counter.v  myreset.v   sdram_cntl.v) in hopes of learning how to use verilator instead of just iverilog.  This worked ok up to point where I added these Verilog files (inc_rom_size.v  jpeg.v rom.v ). Now I have to commit VERILATOR_FLAGS += --coverage or else I take a very long time and crash my system.  I have seen a ulimit error.

wsnyder, 2024-02-07:
Is your system paging?  I don't know what inc_rom_size etc are, but 8 GB is a relatively small amount of memory if this is a large design.

If you can run it to completion with the "--stats" option on e.g. another larger system (if you don't have one, rent an hour on a VM), and post the resulting obj_dir/*_stats.txt file, I can determine if this is a normally large design, or something about coverage is using more memory than expected.

wsnyder, 2024-02-07:
Note running on a medium size design --coverage added about 20% memory (increased from 4.3 GB to 5.1 GB), so 20% is about what to expect in additional memory if it's working correctly and the design is "average".

develone, 2024-02-07:
make

-- Verilator tracing example

-- VERILATE ----------------
verilator -cc --exe -x-assign fast -Wall --stats --trace --assert --coverage -f input.vc top.v sim_main.cpp
%Error: Verilator threw signal 9. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/bin/verilator_bin -cc --exe -x-assign fast -Wall --stats --trace --assert --coverage -f input.vc top.v sim_main.cpp
make: *** [Makefile:65: run] Error 255

This is the Vtop__stats.txt without --coverage.

Verilator Statistics Report

Information:
  Verilator 5.017 devel rev UNKNOWN.REV
  Arguments: -cc --exe -x-assign fast -Wall --stats --trace --assert -f input.vc top.v sim_main.cpp +librescan +libext+.v+.sv+.vh+.svh -y .
  Build jobs: 1
  Verilate jobs: 1

Global Statistics:

  Assertions, assert immediate statements                                                             1
  Assertions, assert non-immediate statements                                                         0
  Assertions, cover statements                                                                        1
  Assertions, full/parallel case                                                                      0
  ConstPool, Constants emitted                                                                        0
  ConstPool, Tables emitted                                                                           1
  Optimizations, Cases complex                                                                        1
  Optimizations, Cases parallelized                                                                   1
  Optimizations, Clocker decomposed vectors                                                           0
  Optimizations, Clocker seen vectors                                                                 0
  Optimizations, Combined CFuncs                                                                      0
  Optimizations, Const bit op reduction                                                              47
  Optimizations, Const bit op reduction                                                               0
  Optimizations, DFG  pre inline 1st CSE, expressions eliminated                                      1
  Optimizations, DFG  pre inline 2nd CSE, expressions eliminated                                      0
  Optimizations, DFG  pre inline Ast2Dfg, coalesced assignments                                       0
  Optimizations, DFG  pre inline Ast2Dfg, input equations                                            28
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (dtype)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (impure)                                  0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (lhs)                                     0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (node)                                    0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (timing)                                  0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (unknown)                                 0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (var ref)                                 0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (width)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, representable                                              28
  Optimizations, DFG  pre inline Dfg2Ast, intermediate variables                                      0
  Optimizations, DFG  pre inline Dfg2Ast, replaced variables                                          0
  Optimizations, DFG  pre inline Dfg2Ast, result equations                                           16
  Optimizations, DFG  pre inline General, modules                                                     9
  Optimizations, DFG  pre inline Peephole, fold assoc binary                                          1
  Optimizations, DFG  pre inline Peephole, fold assoc binary lhs of rhs                               0
  Optimizations, DFG  pre inline Peephole, fold assoc binary rhs of lhs                               0
  Optimizations, DFG  pre inline Peephole, fold binary                                                0
  Optimizations, DFG  pre inline Peephole, fold sel                                                   0
  Optimizations, DFG  pre inline Peephole, fold unary                                                 0
  Optimizations, DFG  pre inline Peephole, inline arraysel                                            0
  Optimizations, DFG  pre inline Peephole, pull nots through cond                                     0
  Optimizations, DFG  pre inline Peephole, push bitwise op through concat                             0
  Optimizations, DFG  pre inline Peephole, push bitwise through reduction                             0
  Optimizations, DFG  pre inline Peephole, push compare op through concat                             0
  Optimizations, DFG  pre inline Peephole, push concat through nots                                   0
  Optimizations, DFG  pre inline Peephole, push not through cond                                      0
  Optimizations, DFG  pre inline Peephole, push reduction through concat                              0
  Optimizations, DFG  pre inline Peephole, push reduction through cond with const branch              0
[... cut ...]
  Vars, width    24                                           3          2          2
  Vars, width    32                                           4          8          8          2
  Vars, width    40                                           2          2          2
  Vars, width    70                                           2          2          2
  Vars, width    96                                                                 4          2
  Vars, width   160                                                                 1

wsnyder, 2024-02-08:
I ran nearly all the tests I have adding coverage, and found a hang, let me debug and fix that first, then we'll see if that resolves your problem.

wsnyder, 2024-02-09:
I fixed some other issues, but don't think any of them are related to this. Regardless please build the latest git master.  Assuming if fails please see if you can post an example that shows the issue, if not run with "--debug --gdbbt", run for as long as you can stand, then hit Ctrl-C to see where it is executing, and attach that output.

develone, 2024-02-09:
Cloned https://github.com/verilator/verilator.git
commit 248de2be2e223955a7a6533842e83dde4adeb46b (HEAD -> master, origin/master, origin/HEAD)
Author: Wilson Snyder <wsnyder@wsnyder.org>
Date:   Thu Feb 8 21:51:36 2024 -0500

    Fix coverage with  member
autoconf
 ./configure
make 
sudo make install
verilator -V
Verilator 5.021 devel rev v5.020-99-g248de2be2
 make

-- Verilator tracing example

-- VERILATE ----------------
verilator -cc --exe -x-assign fast -Wall --trace --assert --coverage --stats -f input.vc top.v sim_main.cpp

This now run ok and creates obj-dir/Vtop__stats.txt

Verilator Statistics Report

Information:
  Verilator 5.021 devel rev v5.020-99-g248de2be2
  Arguments: -cc --exe -x-assign fast -Wall --trace --assert --coverage --stats -f input.vc top.v sim_main.cpp +librescan +libext+.v+.sv+.vh+.svh -y .
  Build jobs: 1
  Verilate jobs: 1

Global Statistics:

  Assertions, assert immediate statements                                                             1
  Assertions, assert non-immediate statements                                                         0
  Assertions, cover statements                                                                        1
  Assertions, full/parallel case                                                                      0
  ConstPool, Constants emitted                                                                        0
  ConstPool, Tables emitted                                                                           1
  Coverage, Toggle points joined                                                                    386
  Optimizations, Cases complex                                                                        1
  Optimizations, Cases parallelized                                                                   1
  Optimizations, Clocker decomposed vectors                                                           0
  Optimizations, Clocker seen vectors                                                                 0
  Optimizations, Combined CFuncs                                                                      0
  Optimizations, Const bit op reduction                                                               6
  Optimizations, Const bit op reduction                                                               0
  Optimizations, DFG  pre inline 1st CSE, expressions eliminated                                      1
  Optimizations, DFG  pre inline 2nd CSE, expressions eliminated                                      0
  Optimizations, DFG  pre inline Ast2Dfg, coalesced assignments                                       0
  Optimizations, DFG  pre inline Ast2Dfg, input equations                                            28
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (dtype)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (impure)                                  0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (lhs)                                     0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (node)                                    0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (timing)                                  0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (unknown)                                 0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (var ref)                                 0
  Optimizations, DFG  pre inline Ast2Dfg, non-representable (width)                                   0
  Optimizations, DFG  pre inline Ast2Dfg, representable                                              28
  Optimizations, DFG  pre inline Dfg2Ast, intermediate variables                                      0
  Optimizations, DFG  pre inline Dfg2Ast, replaced variables                                          0
  Optimizations, DFG  pre inline Dfg2Ast, result equations                                           16
  Optimizations, DFG  pre inline General, modules                                                     9
  Optimizations, DFG  pre inline Peephole, fold assoc binary                                          1
  Optimizations, DFG  pre inline Peephole, fold assoc binary lhs of rhs                               0
  Optimizations, DFG  pre inline Peephole, fold assoc binary rhs of lhs                               0
  Optimizations, DFG  pre inline Peephole, fold binary                                                0
  Optimizations, DFG  pre inline Peephole, fold sel                                                   0
  Optimizations, DFG  pre inline Peephole, fold unary                                                 0
  Optimizations, DFG  pre inline Peephole, inline arraysel                                            0
  Optimizations, DFG  pre inline Peephole, pull nots through cond                                     0
[... 770 lines cut ...]
  Node memory share (%), VAR                      1.04       1.07       0.51       0.55       0.20
  Node memory share (%), VARREF                  21.97       9.83      15.70      18.05      25.40
  Node memory share (%), VARSCOPE                            0.58       0.27
  Node memory share (%), VOIDDTYPE                0.00       0.00       0.00       0.00
  Node memory share (%), WHILE                    0.01                  0.01       0.01       0.01
[... cut ...]
  Vars, width    24                                             5          4          4          4
  Vars, width    32                                             5          9          9          9
  Vars, width    40                                             4          4          4          4
  Vars, width    70                                             4          4          4          4
  Vars, width    96                                                                   4          4
  Vars, width   160                                                                   1          1

wsnyder, 2024-02-09:
>This now run ok and creates obj-dir/Vtop__stats.txt

So you're all set?

develone, 2024-02-10:
Yes thanks for your help.

closed 2024-02-10 by wsnyder