verilator/verilator #4888: reinitializing the verilog model

vanjoe, 2024-02-08:
Is it possible to reinitialize the model after running the simulation for a while? 

I want to simulate a powercycle of a FPGA. Basically all the flip flops in the design should go back to their "configuration" defaults.

I know in an ordinary design a reset signal would be appropriate, but for various reasons that doesn't do what we want.

wsnyder, 2024-02-08:
Yes, you can call new() to make a new context, and also new() the model, see e.g. test_regress/t/t_leak.cpp.

closed 2024-02-08 by wsnyder

vanjoe, 2024-02-14:
It took me a bit to get to this, but I'm running into an issue with tracing. The first time I run VerilatedFSTC::dump() after reinitializing the model a segfault is encountered. Is that expected? can I get around that somehow?

wsnyder, 2024-02-14:
Tracing is tested to work ok (not that there couldn't be a bug of course). You'll need to debug why, maybe you have some places not using the VerilatedContext or something.

vanjoe, 2024-02-14:
Thanks for all your help, I really appreciate it.

```c++
VerilatedFstC trace = new VerilatedFstC();
Vtestbench *tb = new Vtestbench()
tb->trace( trace,99)
trace->open("wave.fst")

/**
* Do some stuff, save some waveforms
*/
//reinitialize
trace->flush();
detete tb;
tb = new Vtestbench
tb ->trace(trace,99);
```

Is this sort of what you were envisioning? When I call the last `tb->trace(trace,99)` I get an error: `Vtestbench::trace()' shall not be called after 'VerilatedFstC::open()'`
Do I need to recreate the VerilatedFstC object as well with a delete/new pair? How do I make it append to a file rather than recreate it?

wsnyder, 2024-02-14:
You need to be creating the model using a VerilatedContext. See the example mentioned above, test_regress/t/t_leak.cpp.

fredrequin, 2024-02-19:
Hello,
I had to solve the same issue few months ago.
The solution was to do a `delete()` and `new()` of the Verilated model, same stuff for the trace object.
All the peripheral C++ models had to be "re-connected" to the new Verilated model.
One tricky part was restarting the coverage. BTW, the coverage is not explicitely linked to a VerilatedContext, why is that ?
At the end, everything worked as expected.
I am wondering if it could be easier to have the call `_eval_initial_loop(vlSymsp);` in a separate method that would be called after the `new()` or when we want to re-intialize the model.

wsnyder, 2024-02-19:
Yes, the model must always be destroyed and recreated.  It would not be sufficient to redo eval_initial_loop even if that was supported. as many variables etc would not be reset to their proper values. Coverage is a pointer via VerilatedContext so that coverage code doesn't always need to be compiled.