verilator/verilator #4890: Getting a module parameter using scope resolution operator

ba-sc, 2024-02-11:
Hello!

By using the scope resolution operator (`::`) we can access the default parameters from the module definition, which, as far as I understand, is not allowed.

**IEEE 1800-2017:**

[image]

Run example on `VCS`:

```text
Error-[SV-LCM-PND] Package not defined
tb.sv, 9
tb, "my_module"
  Package scope resolution failed. Token 'my_module' is not a package. 
  Originating module 'tb'.
  Move package definition before the use of the package.
```

Run example on `Xcelium`:

```text
  localparam integer NUMBERS = my_module::PARAM;
                                       |
xmvlog: *E,NOPBIND (tb.sv,9|39): Package my_module could not be bound.
```

> Can you attach an example that shows the issue?

```systemverilog
/* verilator lint_off MULTITOP */

module my_module #(
  integer PARAM = 123
);
endmodule

module tb;
  localparam integer NUMBERS = my_module::PARAM;

  initial begin
    $display(NUMBERS);
    $finish();
  end
endmodule
```

Output:

```text
        123
- tb.sv:13: Verilog $finish
```

> What 'verilator' command line do we use to run your example?

```text
verilator --binary -j 0 tb.sv
obj_dir/Vtb
```

> What 'verilator --version' are you using?  Did you try it with the git master version?

`Verilator 5.021 devel rev v5.020-101-gcbc76a781`

Yes, this is git master.

> What OS and distribution are you using?

Ubuntu 22.04.3 (GitHub Codespaces: Verilator Devcontainer)

wsnyder, 2024-02-11:
Makes sense, the code isn't checking it's a parameter when resolving.

Might you be willing to make a pull request with the fix, test case, and .out file to check the new error message?

ba-sc, 2024-02-13:
> Might you be willing to make a pull request with the fix, test case, and .out file to check the new error message?

Yes, I would like to do this. Are there any deadlines?

wsnyder, 2024-02-13:
When you get a chance.  We release every 2 months or so.

wsnyder, 2024-02-13:
(We won't wait for it, just saying in case you want it out there.)

ba-sc, 2024-02-13:
Ok, thanks! I'll take it on and open a PR when I have something to show.

ba-sc, 2024-02-19:
Something else. If you specify the top module

```bash
verilator --binary -j 0 -top tb tb.sv
```

, then we get an internal error

```systemverilog
%Error: Internal Error: tb.sv:3:8: ../V3LinkDot.cpp:422: Module/etc never assigned a symbol entry?
    3 | module my_module #(
      |        ^~~~~~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

wsnyder, 2025-09-08:
In master this now gives:

```
%Error: t/t_0_bad.v:13:32: Package/class for ':: reference' not found: 'sub'
   13 |   localparam integer NUMBERS = my_module::PARAM;
```

So assuming resolved, if not please reopen, thanks.

closed 2025-09-08 by wsnyder