verilator/verilator #4893: VERIALTOR_ROOT breaks relocated installations

gezalore, 2024-02-13:
Whether relocating an installation should be supported is up for discussion, but I would like to report this anyway, at least to record the problem for others.

The issue is that bin/verilator (the perl script) uses VERILATOR_ROOT to find verilator_bin, so expects it to be set to the installation prefix, but the generated makefile uses VERILATOR_ROOT to find the run-time lib, so expects it to be set to the pkdata dir (which is <prefix>/share/verilator common Linux hosts), which are different when using an installation instead of a source tree.

Now imagine you installed into a prefix 'foo', then `mv foo bar`

Now you run `./bar/bin/verilator --cc a.sv`
you get this, as it can't find the old pkdatadir, which is the default for VERILATOR_ROOT when it's not in the environment
```
%Error: Cannot find verilated_std.sv containing built-in std:: definitions: /home/gl/work/verilator/repo/../install/share/verilator/include/verilated_std.sv
%Error: This may be because there's no search path specified with -I<dir>.
        ... Looked in:
             foo/share/verilator/include/verilated_std.sv
             foo/share/verilator/include/verilated_std.sv.v
             foo/include/verilated_std.sv.sv
...
%Error: Exiting due to 2 error(s)
```

Ok, fine, I know where it is:
`env VERILATOR_ROOT=bar ./bar/bin/verilator --cc a.sv`
Now you get this, as bin/verilator fails to find verilator_bin:
```
sh: 1: exec: var/share/verilator/verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec bar/share/verilator/verilator_bin --cc a.sv
```

I suspect this (setting VERILATOR_ROOT) was the real cause of the failure described in #3933 (and was due to https://github.com/verilator/verilator/blob/65b632a7dd8927930c85c8236e73d0fb1d9a31d5/ci/ci-win-test.ps1#L2), but was worked around by installing verilated_std.sv in the required place.

I appreciate if you don't care about having the installation relocatable, and there are various ways to work around this (though none that works with --build as well I think). Just reporting for the record, in case you want to think of this or improve the experience.

wsnyder, 2024-02-13:
@gezalore knows this, but for others that land on this thread note the currently supported install options are described here https://veripool.org/guide/latest/install.html#eventual-installation-options

Namely, the two first described:
1. not installing and just pointing VERILATOR_ROOT at the e.g. directory with the github tree which was compiled
2. fixed location prefix install without VERILATOR_ROOT (compiled into the program instead)

We can clean up this issue if it's very obvious the implications of such a change.  I'd prefer not to add too much complexity here, as they are already several options, those existing are hard to well test, and installation changes tend to result in problems that take many versions to settle out (because we don't know what everyone is doing in their local or downstream install scripts.)

gezalore, 2024-02-14:
I appreciate your comments about the fallout and pain from changing any of this.

What I don't quite understand is why does `bin/verilator` need to use `VERILATOR_ROOT`? Surely the right binary to pick up is the one next to `bin/verilator` (which works both in the source tree and the install tree)? We already `use FindBin` at the top to get the path of the executing script, so could we just use the following in `bin/verilator`? What am I missing?

```perl
   sub verilator_bin {
       my $basename = $Debug ? "verilator_bin_dbg" : "verilator_bin";
       my $bin = "";
       if (-x "$RealBin/$basename") {
           return "$RealBin/$basename";
       } elsif (-x "$RealBin/$basename.exe") {
           return "$RealBin/$basename.exe";
       } else {
           $bin = $basename;  # Find in PATH
       }
       return $bin;
   }
```

instead of this:
https://github.com/verilator/verilator/blob/a187a16e56e9d4f4435bcdc62fdf77f0600cd2d7/bin/verilator#L153-L178

That would make VERILATOR_ROOT only be used for finding the runtime lib parts, which would make it unambiguous and fix this issue.

You could even set VERIALTOR_ROOT in bin/verilator to `$RealBin/$relaivePathToPackageDataDirAsReportedByConfigure` by default (if unset), which would also make the installation relocatable, though I don't insist on that unless that is considered a good idea.

(I also don't quite get the use case for VERILATOR_BIN, but we can leave that alive if necessary)

wsnyder, 2024-02-14:
>What I don't quite understand is why does bin/verilator need to use VERILATOR_ROOT?

It doesn't for verilator_bin directly, but the tools verilator invokes later from make (including bin/verilator itself and so then bin/verilator_bin for --hier mode) will expect to find the tools there.  You can counterargue that verilator will set VERILATOR_ROOT, which is true, but if it does so it must be using the hardcoded version and so you shouldn't have needed to set VERILATOR_ROOT at all.

I'm open to a pull that can also locate the verilator binary your suggested way, but suspect it won't help much the later stages as above.  Note there's a nodist/install_test that needs to be run manually that should be improved to also test any new cases; I haven't run this manually recently so might need a tweak or two.

VERILATOR_BIN is there to allow at least my site-specific selections of multiple binaries (I have more than just _dbg version) to take place, I don't know how many others use it, but I need to keep it.

wsnyder, 2024-02-14:
[Edit the install_test is passing]

gezalore, 2024-02-15:
One way that I think is relatively clean and painless is to install all executables into prefix/pkgdatadir/bin (currently only private executables are there), and then add relative symlinks from current prefix/bin/* to prefix/pkgdatadir/bin.

Doing that would fix the issue of an installation not finding pieces, and it would require very small amount of changes to the current code. The `verialtor` wrapper script could then set VERIALTOR_ROOT to it's parent directory, or error if already set and it's not it's parent directory.

Would also make the tests runnable against the installation, not just the source tree, which we probably should do in the CI.

If you can live with that amount of hackery (putting all exes in pkgdata/bin) I can attempt a PR

wsnyder, 2024-02-15:
Not all distros support symlinks in installation directories, and such symlinks are also not always relocatable.  I think the alternative of copying the binary is also too confusing as too likely they diverge.

gezalore, 2024-02-15:
> Not all distros support symlinks in installation directories

I'm somewhat curious which ones.

Well if that's the only objection, and you want to be hacking, could emit the perl (which we depend on anyway) equivalent of

```
#!/bin/bash
exec ThisScriptDir/RelativePathToRealThing "$@"
```

as the "binaries" in the install/bin

gezalore, 2024-02-16:
1. I want to be able to run the tests against the installation (otherwise we will surely break changes we make here later)
2. I would like the installation to be relocatable, but if we don't test this it's kind of pointless. I think point 1 on it's own is valuable enough.

Currently a bunch of tests use `$ENV{VERILATOR_ROOT}/bin/verilator` or `$ENV{VERILATOR_ROOT}/bin/verilator_coverage`. In the installation, those binaries don't live under `$VERILATOR_ROOT/bin`, so to make the tests runnable against the installation, you have 2 options:
a. Make all binaries available under `$VERILATOR_ROOT/bin/` in the installation, just like in the source tree. This requires the symlink/redirect script.
b. Split VERILATOR_ROOT into 2 variables VERILATOR_BINDIR (where the public binaries live, typically: prefix/bin/), and VERILATOR_GUTS (where the include/* and the private binaries live, typically: prefix/share/verilator/). In a source tree, both VERILATOR_BINDIR and VERILATOR_GUTS point to the same path (current VERILATOR_ROOT). In the installation, they are different. Then fix up all the tests to use the right thing instead of VERILATOR_ROOT. 

Of those two, I would argue option 'a' is the less 'ugly', and also the least divergence from the current state of things. Putting all the binaries in one place, and using a relative symlink/redirect script also happens to give you relocatable installations at the same time.

wsnyder, 2024-02-16:
Ok, if you'd like to make a pull let's see what (a) would look like.  Ideally if you can make nodist/install_test pass without changes (until you add whatever new cases you want to test), that will be at least a reasonable sign we won't cause too much package maintainer havoc.

commit f56f3182 (2024-03-01): Make installation relocatable, and the installation testable (#4927)

Fixes #4893
closed 2024-03-01 by gezalore