verilator/verilator #4897: Wire not as expected

scorbetta, 2024-02-15:
Hello all,

I'm using `Verilator 5.014 2023-08-06 rev v5.014` under `Linux scorbetta-linux 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux` through COCOTB version `1.8.1`.

I'm having issues with the following Verilog snippet:

```
assign ACK = (~WNR & sci_ack_fall) | (WNR & sci_ack_rise);
```

where `ACK` is declared as `input wire`, `sci_ack_fall` and `sci_ack_rise` are declared as `wire` and are an output of a module instantiated in the top module.

The output of the simulation is shown in the figure. It is clear that the `ACK` wire shall be asserted, while it is not.

[image]

Stranger also that promoting `ACK` to reg the simulation gives proper result (as can be seen with the `test_reg` signal).

```
reg test_reg;
always @(posedge CLK) begin
    test_reg <= (~WNR & sci_ack_fall) | (WNR & sci_ack_rise);
end
```

Any idea?

Thanks a lot
S

Guiltybyte, 2024-02-15:
Looking at sci_ack_rise, it rises on a positive clockedge and it falls on a positive clock edge, quite possible that this is the root of your problem, try holding the sci_ack_rise signal high for multiple positive clock edges and see what happens.

scorbetta, 2024-02-15:
@Guiltybyte Why would be that a problem? That is the output of an edge detector, hence it should not last more than one cycle. As a matter of fact I am using it everywhere in every design, but the problem arises only here.

Guiltybyte, 2024-02-15:
be that as it may we are debugging an apparent discrepancy in the _simulation_ not the design,

so even though it doesn't make sense to hold this signal for the _design_, the results of this experiment will help me help you out in regards the simulation.

scorbetta, 2024-02-15:
As you expected

[image]

Then, what's the problem?

Guiltybyte, 2024-02-15:
It is to do with how value updates are scheduled/ordered in a simulation for example, in a delta cycle simulation environment, where time is divided into multiple segments maybe updating the value of sci_ack_rise happens after updating the value of ACK etc.

You can visualize it something like this:
[image]

If you test bench is in system verilog the usual easy fix is to add some delay to your stimulus (for example whatever in your test bench is driving the s_ack_rise signal, just add a `#0` or `#1` before the assignment) another common option is to have your stimulus update on the negative edge of the clock rather than the posedge.

I'm not familiar with cocoTB but you can surely do something similar there.

wsnyder, 2024-02-15:
From a Verilator API standpoint doing this works fine and will trace fine, see the example below. How to get this from cocoTB would be a question for the coco folks.   I would generally not suggest adding #delays unless required for other reasons as slows simulation.

```
top->sci_ack_rise = 1
contextp->timeInc(1);
top->eval();
top->sci_ack_rise = 0
contextp->timeInc(1);
top->eval();
```

closed 2024-02-15 by wsnyder

scorbetta, 2024-02-15:
@Guiltybyte Thanks, I keep forgetting about the delta cycles issue... However, is there any way to avoid such issues "by design"? I really don't like the delayed assign, nor the use of negedge... still I understand their use. It would be nice to have a sort of methodology to avoid these situations. For instance why is this happening with these particular stimuli? Is really the update of simulated signals not so controllable?

wsnyder, 2024-02-15:
The issue is because there is a primary input that affects a primary output with no state element (flop) between, also known a s "feed through".  Most teams have a design rule that feed thrus are illegal, with the typical recommendation that all outputs must come only from posedge flops. That rule is more for synthesis timing closure than for simulation FWIW.