verilator/verilator #4900: Encountered process that should've been resumed at an earlier simulation time. Missed a time slot?

muzafferkal, 2024-02-16:
I have a model which started in cocotb + verilator and simulates now well in VCS. I wanted to go back but I'm getting the above error. Any idea what this means as I couldn't find anything online about it. 
I'm running with the following options:
EXTRA_ARGS += --timescale 1ns/1ps --timing -Wno-lint

and my version is as follows:

Verilator 5.013 devel rev v5.012-63-g3454641

wsnyder, 2024-02-16:
It means that Verilator said via `nextTimeSlot()` that it needed a wakeup at e.g. time 100, but the upper main() (e.g. Cocotb) woke it next up too late at e.g. time 110.  I'd debug with print statements to see what nextTimeSlot() says each time eval() is called.

Or there's a Verilator bug.  If this is suspected we'll need a standalone test case that doesn't use cocotb.  I suspect that's unlikely, so closing - feel free to reopen otherwise.

closed 2024-02-16 by wsnyder

muzafferkal, 2024-02-16:
@wsnyder in "debug with print statements" do you mean recompile Verilator with a print statement added to nextTimeSlot() function? 

Also when you say "standalone test case" do you mean without cocotb and a test case available to you in source?

wsnyder, 2024-02-16:
You could modify include/verilated.h to print, or modify the code that calls eval() in cocotb or whatever to print.

If that shows the calls are correct into Verilator and the problem is inside Verilator, yes, if we need to debug we'll need sources that don't use cocotb.  Or of course, debug yourself, which is always appreciated!