verilator/verilator #4901: changing a single constant causes Verilator to crash

muzafferkal, 2024-02-16:
I'm trying to generate SC output from Verilator (Verilator 5.013 devel rev v5.012-63-g3454641) 
with 
verilator --debug --sc --timing -Wno-lint --top ce $^

I have a generate for loop ala:

    generate
        for (g = 0; g < 1; g = g+1) begin: gen_mul_acc

My problem is that when the loop check is 1 as above I get 

- V3Ast.cpp:1217:     Dumping obj_dir/Vce_990_final.tree
- Verilator.cpp:779:  Done, Exiting...

But when the same line reads "g < 128" 

I get 

- V3Ast.cpp:1217:     Dumping obj_dir/Vce_035_const.tree
- V3Scope.cpp:421:    scopeAll:
%Error: export VERILATOR_ROOT=
%Error: ulimit -s unlimited 2>/dev/null; exec /remote/global/freeware/verilator/5.013/bin/verilator_bin_dbg --debug --sc --timing -Wno-lint --top ce /projects/kal/projects/hw/ip/cim/icg.sv /projects/kal/projects/hw/ip/cim/ce.sv /projects/kal/projects/hw/ip/cim/adder_tree.sv /projects/kal/projects/hw/ip/cim/accum.sv /projects/kal/projects/hw/ip/cim/DCIM_TOP.sv
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /remote/global/freeware/verilator/5.013/bin/verilator_bin_dbg --debug --sc --timing -Wno-lint --top ce /projects/kal/projects/hw/ip/cim/icg.sv /projects/kal/projects/hw/ip/cim/ce.sv /projects/kal/projects/hw/ip/cim/adder_tree.sv /projects/kal/projects/hw/ip/cim/accum.sv /projects/kal/projects/hw/ip/cim/DCIM_TOP.sv

muzafferkal, 2024-02-16:
One question here:
 is the generated C++ code flexible enough for me to put the 128 number back in it or does it generate 1 or 128 copies to the loop completely disappears after generation?

Guiltybyte, 2024-02-16:
if you check ./docs/guide/exe_verilator.exe, there is a command option `--no-unlimited-stack`
which stops verilator from trying to run `ulimit -s unlimited`
as it appears that your error is caused by this ulimit command failing, try and run verilator with this  `--no-unlimited-stack` option.

muzafferkal, 2024-02-17:
Thanks @Guiltybyte for the suggestion. Alas here is what I get when I do that. Is this a capacity issue? Is there a number which I can increase to get around this issue?

- V3Ast.cpp:1217:     Dumping obj_dir/Vce_034_inst.tree
- V3Const.cpp:3831:   constifyAll: 
- V3Ast.cpp:1217:     Dumping obj_dir/Vce_035_const.tree
- V3Scope.cpp:421:    scopeAll: 
%Error: export VERILATOR_ROOT=
%Error: /remote/global/freeware/verilator/5.013/bin/verilator_bin_dbg --no-unlimited-stack --debug --sc --timing -Wno-lint --top ce /projects/kal/projects/hw/ip/cim/icg.sv /projects/kal/projects/hw/ip/cim/ce.sv /projects/kal/projects/hw/ip/cim/adder_tree.sv /projects/kal/projects/hw/ip/cim/accum.sv /projects/kal/projects/hw/ip/cim/DCIM_TOP.sv
%Error: Command Failed /remote/global/freeware/verilator/5.013/bin/verilator_bin_dbg --no-unlimited-stack --debug --sc --timing -Wno-lint --top ce /projects/kal/projects/hw/ip/cim/icg.sv /projects/kal/projects/hw/ip/cim/ce.sv /projects/kal/projects/hw/ip/cim/adder_tree.sv /projects/kal/projects/hw/ip/cim/accum.sv /projects/kal/projects/hw/ip/cim/DCIM_TOP.sv

Guiltybyte, 2024-02-17:
The error isn't too descriptive but my guess is that the `ulimit -s unlimited` command verilator is trying to run is required for you to run verilator for this particular design configuration.

`ulimit` is a unix command for setting resource limits in a shell.

`ulimit -s` is for setting the max stack size in a shell session.

However to run ulimit you need admin privileges, i.e. try running: `sudo sh -c "ulimit -s unlimited"`
before running verilator with the `--no-unlimited-stack` option enabled.

you might also try running verilator without the `--no-unlimited-stack` option and running it with admin priv (i.e. via sudo)

There is also the possibility that your design results in a ridiculously massive verilated model but I have no way of telling that. Try the above two first anyway.

muzafferkal, 2024-02-17:
@Guiltybyte thanks for the info. I have read up on ulimit and it seems my shell (csh) uses limit for the same purpose. Setting unlimit and trying my Verilator command runs much longer but still fails with the same error output.
My design actually is really not that big. With the generate set to 128, VCS run takes only a couple minutes. I'm only modeling an adder tree and using 128 copies of it. In physical hardware this logic fits into 90000 square micron :-)

I'm also seeing many of the following in the working case and I'm curious if there is a different coding style which might help:

- V3Unroll.cpp:516:   unrollAll:
- V3Unroll.cpp:67:       Can't Unroll: Unable to simulate loop :WHILE 0x51d7b30 <e9921> {h122ar}

- V3Timing.cpp:235:   Always does NOT need process (1534 times)

wsnyder, 2024-02-17:
Those messages should be harmless. Run with --debug --gdbbt and please post the backtrack that is printed (if any).

You can also run e.g. "top" simultaniously In another window to see how much memory is used.

kal-rain, 2024-03-07:
Hi @wsnyder ,
Sorry for the delay and for a full paper reference but here is something like what I'm implementing:

https://semiwiki.com/semiconductor-manufacturers/tsmc/296278-all-digital-in-memory-computing/
This paper implements 256 rows of adder trees, I'm implementing only 128 rows. A single instance compiles and simulates very fast in verilator but 128 doesn't compile. I'm afraid I can't share the design but it's mostly structural and there are a large number of full adders in the full design. I hope this helps.

wsnyder, 2024-03-09:
Believe it's out of memory, either correctly or due to a mis-optimization.  There's not much we can do without a test.  If you debug the problem and can provide a pull request to fix it that's another option.  Closing assuming that won't be soon.

closed 2024-03-09 by wsnyder