verilator/verilator #4902: Too strict with whitespace in pragma protect

paul-demo, 2024-02-16:
Verilator linting rejects this:
```
`pragma protect version = 1
```

But the linter is ok with it if the spaces are removed around the equals sign:
```
`pragma protect version=1
```

I have a Xilinx IP which includes the spaces and would like to not modify it. Regardless of what the strict Verilog spec says, it seems like whitespace like this should be acceptable to Verilator.

wsnyder, 2024-02-16:
IEEE says whitespace are legal around the =.  Perhaps you'd be willing to submit a pull request to fix this, and test it (modify t_preproc.v and V3PreLex.l)?

paul-demo, 2024-02-17:
Sure, I'll get to it some time next week.

paul-demo, 2024-02-22:
Please see PR here:
https://github.com/verilator/verilator/pull/4914

wsnyder, 2024-02-22:
Great! Thanks for your pull request, merged.

closed 2024-02-22 by wsnyder