verilator/verilator #4908: SIGSEGV, Segmentation fault when running "make test"

Xocket, 2024-02-20:
Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)
`Xocket@Xocket-PC ~/verilator
$ make test
------------------------------------------------------------
making verilator in src
make -C src
make[1]: Entering directory '/home/Xocket/verilator/src'
make -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj serial
make[2]: Entering directory '/home/Xocket/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_bin_dbg.exe VL_DEBUG=1 -f ../Makefile_obj
make[2]: Entering directory '/home/Xocket/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/Xocket/verilator"
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg.exe VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
make[2]: Entering directory '/home/Xocket/verilator/src/obj_dbg'
make[2]: Nothing to be done for 'serial_vlcov'.
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_dbg'
make -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg.exe VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
make[2]: Entering directory '/home/Xocket/verilator/src/obj_dbg'
      Compile flags:  g++ -Og -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/Xocket/verilator"
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_dbg'
make -C obj_opt -j 1  TGT=../../bin/verilator_bin.exe -f ../Makefile_obj serial
make[2]: Entering directory '/home/Xocket/verilator/src/obj_opt'
make[2]: Nothing to be done for 'serial'.
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_opt'
make -C obj_opt       TGT=../../bin/verilator_bin.exe -f ../Makefile_obj
make[2]: Entering directory '/home/Xocket/verilator/src/obj_opt'
      Compile flags:  g++ -O3 -DVERILATOR_INTERNAL_ -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/home/Xocket/verilator"
make[2]: Leaving directory '/home/Xocket/verilator/src/obj_opt'
make[1]: Leaving directory '/home/Xocket/verilator/src'
test_regress/t/t_a1_first_cc.pl
======================================================================
dist/t_a1_first_cc: ==================================================
-Skip: dist/t_a1_first_cc: scenario 'dist' not enabled for test
dist/t_a1_first_cc: -Skip: Skip: scenario 'dist' not enabled for test
==SUMMARY: Passed 0  Failed 0  Time 0:00
======================================================================
vlt/t_a1_first_cc: ==================================================
        perl ../bin/verilator --debug --debugi 0 --gdbbt --no-dump-tree -V
warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
No stack.
warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
        add-auto-load-safe-path /home/Xocket/verilator/test_regress/.gdbinit
line to your configuration file "/home/Xocket/.config/gdb/gdbinit".
To completely disable this security protection add
        set auto-load safe-path /
line to your configuration file "/home/Xocket/.config/gdb/gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
        info "(gdb)Auto-loading safe path"
[New Thread 33516.0x8874]
[New Thread 33516.0x7438]
[New Thread 33516.0x4f4c]
Starting Verilator 5.021 devel rev v5.020-106-ga951446f9
Starting Verilator 5.021 devel rev v5.020-106-ga951446f9
Verilator 5.021 devel rev v5.020-106-ga951446f9

Copyright 2003-2024 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            =
[... 54 lines cut ...]

Quit anyway? (y or n) [answered Y; input not from terminal]
%Error: export VERILATOR_ROOT=/home/Xocket/verilator
%Error: ulimit -s unlimited 2>/dev/null; exec gdb /home/Xocket/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc --fdedup --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc +define+TEST_DUMPFILE=obj_vlt/t_a1_first_cc/simx.vcd t/t_a1_first_cc.v" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec gdb /home/Xocket/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run --prefix Vt_a1_first_cc ../obj_vlt/t_a1_first_cc/Vt_a1_first_cc__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_a1_first_cc --fdedup --debug-check --comp-limit-members 10 --debug --debugi 0 --gdbbt --no-dump-tree --trace --clk clk -f input.vc +define+TEST_OBJ_DIR=obj_vlt/t_a1_first_cc +define+TEST_DUMPFILE=obj_vlt/t_a1_first_cc/simx.vcd t/t_a1_first_cc.v" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Warning: vlt/t_a1_first_cc: Exec of perl failed: warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".

vlt/t_a1_first_cc: %Error: Exec of perl failed: warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
vlt/t_a1_first_cc: FAILED: Exec of perl failed: warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
==SUMMARY: Passed 0  Failed 1  Time 0:07
==SUMMARY: Passed 0  Failed 1  Time 0:07

======================================================================
        #vlt/t_a1_first_cc: %Error: Exec of perl failed: warning: File "/home/Xocket/verilator/test_regress/.gdbinit" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
                make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:07
make: *** [Makefile:170: smoke-test] Error 10
`
What 'verilator' command line do we use to run your example?

What 'verilator --version' are you using?  Did you try it with the git master version?
None. It's "make test" after compiling and linking.

What OS and distribution are you using?
cygwin64 under Windows 11

wsnyder, 2024-02-21:
```
#0  ___chkstk_ms () at /usr/src/debug/gcc-11.4.0-1/libgcc/config/i386/cygwin.S:126
#1  0x00000000002af7a8 in ?? ()
#2  0x0000000a00015980 in ?? ()
#3  0x0000000100563070 in V3Stats::statsStageAll (nodep=0x7ffe036e8, stage=..., fastOnly=fastOnly@entry=false) at ../V3Stats.cpp:168
```

Seems like something went bad in the stack.  Unless you can debug further I'll need to install cygwin and debug, this may take a while.  Can you use WSL instead of Cygwin?

Xocket, 2024-02-22:
> ```
> #0  ___chkstk_ms () at /usr/src/debug/gcc-11.4.0-1/libgcc/config/i386/cygwin.S:126
> #1  0x00000000002af7a8 in ?? ()
> #2  0x0000000a00015980 in ?? ()
> #3  0x0000000100563070 in [...]

Works with WSL. Works also under Cygwin with 5.016, it's 5.18/stable/master that have problems.

wsnyder, 2024-02-25:
Ok, I'm unlikely to be able to debug this as don't have a Windows 11 system, if you can that would be great.

wsnyder, 2024-03-26:
Closing as unlikely to get a fix without a pull request, as none of the maintainers AFAIK have Windows 11 + Cygwin.

closed 2024-03-26 by wsnyder